raspberrypi/rpicam-apps #741: [BUG] symbol lookup error _ZN9libcamera8controls3rpi11ScalerCropsE again

arsatyants, 2024-11-03:
-------------------------------------------------------------------------------
rpicam-apps Bug Report
--------------------------------------------------------------------------------
Date: 04-11-2024 (01:11:15)
Command: ./camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
pi5

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

Revision	: d04170
Serial		: eecd602356a24b7c
Model		: Raspberry Pi 5 Model B Rev 1.0

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/firmware/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=458b983b-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=UA
** cat /boot/firmware/config.txt **
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
[... 1007 lines cut ...]

--------------------------------------------------------------------------------
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux pi5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
2024/10/21 15:27:49 
Copyright (c) 2012 Broadcom
version 951e1cc9 (release) (embedded)

** rpicam-hello --version **
rpicam-hello: symbol lookup error: /lib/aarch64-linux-gnu/rpicam_app.so.1.5.2: undefined symbol: _ZN9libcamera8controls3rpi11ScalerCropsE

arsatyants, 2024-11-03:
**it does not** work again after removing all libcamera-related files in /usr/local/lib/aarch64-linux-gnu/ (but not the folder libcamera)

take in account /lib/aarch64-linux-gnu/rpicam_app.so.1.5.2   the last #723  bug was with /lib/aarch64-linux-gnu/rpicam_app.so.1.5.1:

naushir, 2024-11-04:
You need to build and install libcamera from this repo: https://github.com/raspberrypi/libcamera.  Removing the system install of libcamera is not enough.

arsatyants, 2024-11-05:
libcamera is not buildable code

../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘int libcamera::PiSPCameraData::configureCfe()’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1868:13: error: ‘PISP_IMAGE_FORMAT_COMPRESSED’ was not declared in this scope; did you mean ‘PISP_IMAGE_FORMAT_UNCOMPRESSED’?
 1868 |         if (PISP_IMAGE_FORMAT_COMPRESSED(image.format)) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             PISP_IMAGE_FORMAT_UNCOMPRESSED

naushir, 2024-11-05:
That's because you have an outdated version of [libpisp](https://github.com/raspberrypi/libpisp).  Please remove your current installation and either use ``sudo apt install libpisp-dev`` or build/install this manually.

arsatyants, 2024-11-05:
**updated:**

sudo apt install libpisp-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libpisp-dev is already the newest version (1.0.7-1).

**libpisp is not buildable:**
sudo meson install -C build

ERROR: Build data file '/home/admin/libpisp/build/meson-private/build.dat' references functions or classes that don't exist. This probably means that it was generated with an old version of meson. Try running from the source directory meson setup /home/admin/libpisp/build --wipe

**sudo apt install meson**
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
**meson is already the newest version (1.0.1-5).**

naushir, 2024-11-06:
Can you delete the ``build`` directory and start again?  ``libpisp`` builds correctly when I try it locally and in our CI system.

arsatyants, 2024-11-06:
The workaround is to update meson to 1.6 manually using source code but not apt install. Fixed now. Thanks.

closed 2024-11-06 by naushir