raspberrypi/rpicam-apps #742: [BUG] Pi Camera not detecting

subhodeep-dey, 2024-11-04:
Pi Camera not detecting in raspi-config interface options
But when i am trying for online webcam test (https://webcammictest.com) - the camera is working.

Added bug-report.txt in attachment
[bug-report.txt](https://github.com/user-attachments/files/17613578/bug-report.txt)

subhodeep-dey, 2024-11-04:
--------------------------------------------------------------------------------
rpicam-apps Bug Report
--------------------------------------------------------------------------------
Date: 04-11-2024 (09:12:28)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Revision	: c03115
Serial		: 100000004290d15c
Model		: Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/firmware/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=8879e2a0-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=GB
** cat /boot/firmware/config.txt **
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
[... 1516 lines cut ...]
--------------------------------------------------------------------------------
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Aug 30 2024 19:17:39 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)

** rpicam-hello --version **
rpicam-apps build: v1.5.2 a8ccf9f3cd9d 03-10-2024 (22:33:44)
libcamera build: v0.3.2+27-7330f29b

6by9, 2024-11-04:
[What to do if your camera is not detected](https://forums.raspberrypi.com/viewtopic.php?t=362707)

>You can tell if a tutorial is talking about the obsolete camera stack if you see the following:
>
>    Command line applications such as: raspistill, raspivid
>    Use of "vc_gencmd get_camera"
>    A Python module called PiCamera (or picamera, python-picamera or python3-picamera)
>    Instructions about putting "start_x" or "gpu_mem" into your /boot/config.txt file.
>    **Instructions that tell you to "enable the camera" using the raspi-config tool.**
>    Terms such as MMAL or mmal.
>
>Any instructions referring to the above terms are meaningless and indeed harmful for the modern camera stack.

closed 2024-11-04 by naushir