raspberrypi/rpicam-apps #747: [BUG] RTSP stream not starting on RPi 5

GitHubGoody, 2024-11-06:
**Describe the bug**
I have an RPi 5 8Gb running headless on the latest Raspberry Pi OS available via the RPi Imager with an RPi Camera Module 3 Wide connected. I'm simply trying to get it configured to stream RTSP but I keep getting errors and the stream does not start.

**Bug report**
I attached the bug report generated using this command:
`camera-bug-report -t <timeout in seconds> -o <output file> -c "<rpicam-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h264>`

[bug.txt](https://github.com/user-attachments/files/17652790/bug.txt)

This is the issue I posted on the forum [here](https://forums.raspberrypi.com/viewtopic.php?t=378996):

I have an RPi 5 8Gb running headless on the latest Raspberry Pi OS available via the RPi Imager with an RPi Camera Module 3 Wide connected. I'm simply trying to get it configured to stream RTSP.

After running all of these commands to update it...

```
sudo apt update
supo apt upgrade
sudo apt install vlc
sudo apt install ffmpeg
```

...I tried the instructions offered here: https://www.raspberrypi.com/documentation/computers/camera_software.html#rtsp

When I try...
`rpicam-hello`
...it starts the stream

When I try...
`rpicam-jpeg --output test.jpg`
...it streams for 5 seconds and outputs a test.jpg image that I can retrieve via WinSCP and it looks great.

When I try...
`rpicam-vid  -t 10s -o test.mp4`
...it streams for 10 seconds and outputs a test.mpg video that I can retrieve via WinSCP at it looks great.

When I try to start a stream with this...
`rpicam-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h264`
...I get this result and errors...

```
pi@xtoolp2lasercamera:~ $ rpicam-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h264
VLC media player 3.0.21 Vetinari (revision 3.0.21-0-gdd8bfdbabe8)
[000055563b0f0e50] main interface error: no suitable interface module
[000055563b009560] main libvlc error: interface "globalhotkeys,none" initialization failed
[000055563b0f0e50] dummy interface: using the dummy interface module...
[18:24:36.359722123] [18476]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[18:24:36.367574459] [18489]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[18:24:36.385163086] [18489]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Preview window unavailable
Mode selection for 640:480:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 1486.67
    SRGGB10_CSI2P,2304x1296/0 - Score: 1786.67
    SRGGB10_CSI2P,4608x2592/0 - Score: 2686.67
Stream configuration adjusted
[18:24:36.386235628] [18476]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-YUV420 (1) 1536x864-BGGR_PISP_COMP1
[18:24:36.386348425] [18489]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[NULL @ 0x555661e5b2f0] Unable to find a suitable output format for '-'
ERROR: *** libav: cannot allocate output context, try setting with --libav-format ***
[00007fff1c004290] h26x demux error: this doesn't look like a h264 ES stream, continuing anyway
[00007fff1c004e80] main stream error: unknown query 0x30e in demux_vaControlHelper
```

Specifically, these lines are in red:

```
[00005555e8a84e50] main interface error: no suitable interface module
[00005555e899d560] main libvlc error: interface "globalhotkeys,none" initialization failed

[NULL @ 0x5555cd22f2f0] Unable to find a suitable output format for '-'
ERROR: *** libav: cannot allocate output context, try setting with --libav-format ***
[00007fff4c004290] h26x demux error: this doesn't look like a h264 ES stream, continuing anyway
[00007fff4c004e80] main stream error: unknown query 0x30e in demux_vaControlHelper
```

naushir, 2024-11-08:
Can you try adding ``--libav-format h264`` into the command line?

slain2, 2024-11-10:
See my reply to that forum post for a version that works on a Pi5 Bookworm with v3 wide camera.

naushir, 2024-11-21:
Any updates on this?  If not I will close down shortly.

GitHubGoody, 2024-11-25:
@naushir Sorry, I wasn't able to check in here for a couple weeks. I abandoned my initial approach for this project...first partially with mediamtx (which seemed to be working well considering the documented rpicam commands did not work at all) then completely since the RPi 5 uses 22 pin mini-CSI camera connectors. I need a 1 m long cable for this project, but for some unknown reason, I cannot find a 15-pin to 22-pin cable that long, so I'm trying to make a 3B+ (15 pin regular CSI connector) work for this streamer. In any case, if the RPi 5 requires different commands as (@slain2 suggested) for the native camera library to work, shouldn't that be documented more clearly at the location I linked in my OP? Conversely, if the provided commands should work, then it seems there's still a bug that needs addressing. 🤷🏼‍♂️

naushir, 2024-11-25:
I think it does mention to use ``--libav-format`` [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#libav).  If it's unclear please do open a PR to update the text.

GitHubGoody, 2024-12-08:
> --libav-format h264

[Done](https://github.com/raspberrypi/documentation/pull/3955), please modify as necessary since I wasn't able to test this.

Tcoton, 2025-12-06:
Whichever command found in the documentation to start a stream on local network does not provide a video streaming to other computers. I am battling with all the commands to try to make it work with a Raspberry Pi 5 using the latest Raspberry OS Trixie, I can see on the local Pi 5 a preview of the video using the updated command in this page but when trying to connect from another computer using vlc and opening the rstp streaming, I get a "cannot connect to server" error and I am on the same LAN. Is there some kind of firewall? I am also testing go2rtc and this one can stream on the very same ports as the one in the command line.