raspberrypi/rpicam-apps #748: Is it possible to invert the colors with rpicam-apps?

LibreTechnica, 2024-11-17:
I am trying to built a digital magnifier for people with low vision.  A common function for the commercial magnifiers is to invert the colors.  As i'm understanding more about the Raspberry Pi camera concepts, it appears that this ability used to exist in the legacy application `raspicam` but was inexplicably removed in `rpicam-apps`.  (Source: https://www.raspberrypi.com/documentation/computers/camera_software.html#differences-between-rpicam-and-raspicam) 

Is there another way I could achieve this with `rpicam-apps`?  I'm using the **DRM/KMS preview rendering** so I don't have any x11 overhead to slow things down, but I'll entertain using x11 if I can still achieve fast boot times.  I appreciate any direction and suggestions at all.

6by9, 2024-11-17:
The effects in the legacy camera stack were internally referred to as the legacy effects framework, and were all post-processing of the images.

The libcamera docs cover post-processing, which includes a negate option - https://www.raspberrypi.com/documentation/computers/camera_software.html#built-in-stages

The other potential option is to use a gamma map to invert the image. Pi0-3 support an output gamma map in the DRM driver (Pi4 has a hardware issue in that block, and I can't recall for Pi5).
I can't recall if the ISP can implement an inverted gamma map when producing the images.

LibreTechnica, 2024-11-17:
Hi!  I just tested this and is worked like a charm!  That page alro made me interested in the Sobel filter, but I'll explore that when I have more time.  Thanks so much @6by9!

closed 2024-11-17 by LibreTechnica