raspberrypi/rpicam-apps #749: [BUG] json configs are referencing models that are not yet in rpi stable.

mtycvy, 2024-11-23:
**Describe the bug**
running
`sudo rpicam-hello -t 0 --post-process-file ./assets/hailo_yolov6_inference.json --lores-width 640 --lores-height 640`
returns the following error
```
Reading post processing stage "hailo_yolo_inference"
ERROR: *** No such node (hef_file) ***
```
It looks like commit c17a416 is referencing models that hailo have not yet published to raspberry pi stable.

naushir, 2024-11-25:
These new models can be downloaded with the [download-hailo-models.sh](https://github.com/raspberrypi/rpicam-apps/blob/main/utils/download-hailo-models.sh) scripts.  We are preparing a software package release now so they should be available on apt soon.

mtycvy, 2024-11-25:
thankyou - that has fixed some of the commands. maybe the documentation [https://www.raspberrypi.com/documentation/computers/ai.html](hdocumentation/computers/ai.html )
should be update to included this step - new users like me will encounter this error and potentially get stuck until the models appear in apt. There are also messages in the hailo forums from people encountering this issue.

also some of the commands in the documentation don't work when using main branch.

mtycvy, 2024-11-25:
im still having some problems reading the model for my hailo8 - it looks like the problem is around https://github.com/raspberrypi/rpicam-apps/blob/6318ac74e44ad973d5170540c3818016a31bb94c/post_processing_stages/hailo/hailo_postprocessing_stage.cpp#L20

Im not a c++ developer, but is HAILO_ARCH_HAILO8 ever defined?

closed 2024-11-28 by naushir