raspberrypi/rpicam-apps #750: Build error following libcamera build

jamesmaino, 2024-11-27:
I am running a fresh install of:

Raspberry Pi OS with desktop
Release date: November 19th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)

I then run the following:

```

# libcamera deps
sudo apt install -y libboost-dev
sudo apt install -y libgnutls28-dev openssl libtiff5-dev pybind11-dev
sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
sudo apt install -y meson cmake
sudo apt install -y python3-yaml python3-ply

cd ~
git clone https://github.com/raspberrypi/libcamera.git
cd libcamera
meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=disabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled
ninja -C build
sudo ninja -C build install

sudo apt remove -y --purge rpicam-apps 

# rpicam-apps deps
sudo apt install -y libepoxy-dev libjpeg-dev libtiff5-dev libpng-dev
sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
sudo apt install -y libavcodec-dev libavdevice-dev libavformat-dev libswresample-dev

sudo apt install -y cmake libboost-program-options-dev libdrm-dev libexif-dev
sudo apt install -y meson ninja-build

cd ~
git clone https://github.com/raspberrypi/rpicam-apps.git
cd rpicam-apps
meson setup build -Denable_libav=enabled -Denable_drm=enabled -Denable_egl=enabled -Denable_qt=enabled -Denable_opencv=disabled -Denable_tflite=disabled -Denable_hailo=disabled
meson compile -C build
sudo meson install -C build

sudo ldconfig

```

Adter install, the following works: 

```
rpicam-vid
```

However, there are type errors in python which cause issues with scripting. 

For example, this returns a type error:

```
python3 -c 'import libcamera; print( libcamera.controls.Contrast.type)' 
```

naushir, 2024-11-27:
I wonder if you are not using the expected libcamera package.  What is the output of

```
python3 -c 'import libcamera; print(libcamera.__file__)'
```

On my setup it points to 
```
/usr/local/lib/aarch64-linux-gnu/python3.11/site-packages/libcamera/__init__.py
```

jamesmaino, 2024-11-28:
Thank you for the clue! 

My setup it was pointing to the wrong package. 

```shell
$ python3 -c 'import libcamera; print(libcamera.__file__)'

/usr/lib/python3/dist-packages/libcamera/__init__.py
```

I wasted a lot of time thinking it was a build issue! Changing the path fixed the issue.

`export PYTHONPATH=/usr/local/lib/aarch64-linux-gnu/python3.11/site-packages:$PYTHONPATH`

This is my first time building `libcamera` and `rpicam-apps` but I assumed the `meson` option `-Dpycamera=enabled` would update the path. Is this not the case?

naushir, 2024-11-28:
I don't think it would update the ``PYTHON_PATH`` automatically.  There may be a way to do this through ``meson devenv``, but right now the build scripts do not support that option as far as I know.

Closing now.

closed 2024-11-28 by naushir