raspberrypi/rpicam-apps #753: Document required version of libcamera

skadisch, 2024-12-05:
I'm trying to build both [libcamera](https://github.com/raspberrypi/libcamera) and this project from source. I have checked the linked [instructions](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps), but I don't seem to find anywhere, what versions of these projects are compatible. In [meson.build](https://github.com/raspberrypi/rpicam-apps/blob/b78871ec21b466d8cbf04430af88b660aff5a487/meson.build#L37), only libcamera is specified as a dependency. So of course, the meson setup succeeds, but then the build yields errors afterwards. I have tried some combinations of versions, all having problems of their own. Am I missing something? If not, maybe the required version of libcamera should be specified in the meson.build file or documented somewhere else.

naushir, 2024-12-06:
We aim to ensure that this tree will always correctly compile with the main branch of our libcamera tree: https://github.com/raspberrypi/libcamera.  When we are close to making a camera package release, there may be a day or two when these two repos are out of sync when preparing for the release, but otherwise you can assume they will be in sync.

It would help if you could share your build errors so we can help you get a working build.

skadisch, 2024-12-06:
Meanwhile I basically did, what you said - use the newest versions of both repositories. I see that currently there is a lot of development going on, and this would cause some overhead. Probably it makes sense to revisit this, once both repositories have stabilized in terms of dependencies (i.e. libcamera >= 1.0.0). What do you think?

naushir, 2024-12-06:
I think it may be quite a while before libcamera reaches 1.0.0 - it's in active development with lots of changes planned for the future.  However, if you want a stable release, we do tag both branches when we make release packages.  Is that suitable for you?

naushir, 2024-12-11:
Any other questions related to this?  If not, I'll close this shortly.

closed 2024-12-12 by naushir