raspberrypi/rpicam-apps #756: How to --annotate-text ?

int2018, 2024-12-30:
Hallo rpicam team,

I recently upgraded my little pi zero to bookworm and found that programs like raspistill, raspivid didn't make it into the new world.
Ok, I guess that's how it goes with extinction and so I tried the "new world" program set libcamera-* (an newly called rpicam-*).

Ok, updating some scripts can't be a big problem since the camera hardware is still around and the usual features are working.

However, I could not figure out how add/merge the timestamp text into the image as I did with raspistill:
raspistill -o $f -t 1 --annotate 12 --annotate-text "$(date +'%d.%m.%Y %H:%M')"

In rpicam-still is an --annotate-text parameter but only with fixed arguments for title bar display, not in the image.
And a -datetime option can be passed to create output filenames.

Maybe I missed something (in the documentation ?) but is there any chance I can achieve the annotate-text feature or do I have to make an additional step with other programs to get the info into the image? It was already implemented so it should be "easy" to migrate or re-implement?

Thanks for a hint and thanks for your effort!

And have a good start into 2025!

6by9, 2025-01-02:
https://www.raspberrypi.com/documentation/computers/camera_software.html#annotate_cv-stage

> annotate_cv stage
> This stage writes text into the top corner of images using the same % substitutions as the [info-text](https://www.raspberrypi.com/documentation/computers/camera_software.html#info-text) option.

int2018, 2025-01-02:
Yes, I tried that but it didn't work and no error messages. 
sudo rpicam-still --post-process-file /home/pi/annotate_cv.json -o /home/pi/tmp.jpg

Seems the standard build doesn't have what is needed?

$ rpicam-still --version
rpicam-apps build: v1.5.3 50958df98d3c 26-11-2024 (15:45:26)
rpicam-apps capabilites: egl:0 qt:0 drm:1 libav:0
libcamera build: v0.3.2+99-1230f78d

I just want to put some text in the image, I don't need face recognition and AI framework...
As a compromise I installed ImageMagick (which also is a feature overkill) and now I do the "postprocessing" like this: 

sudo nice -n 10 convert /home/pi/tmp.jpg -gravity SouthEast -pointsize 36 -fill "rgba(255,255,255,0.5)" -annotate +20+20 "$(date '+%d.%m.%Y %H:%M')" /var/www/html/last.jpg

Maybe the new concept of rpicam-* is very powerful but it lost some fast and easy features...
Well, like the old saying: add a feature - lose a feature...

Thanks for the hint anyway

naushir, 2025-01-02:
Try ``--post-process-file /use/share/rpi-camera-assets/annotate_cv.json``.

naushir, 2025-01-06:
Any further information needed on this?  If not, I'll close this shortly.

int2018, 2025-01-06:
Well I tried the suggestion but it is not working. 
sudo rpicam-still --post-process-file /home/pi/annotate_cv.json -o /home/pi/tmp.jpg
sudo rpicam-still --post-process-file /usr/share/rpi-camera-assets/annotate_cv.json -o /home/pi/tmp.jpg

I think moving the annotate_cv.json into another folder didn't help.

As I mentioned before, could it be that this feature needs OpenCV support linked in to the bin? Do I have it?
$ rpicam-still --version
rpicam-apps build: v1.5.3 https://github.com/raspberrypi/rpicam-apps/commit/50958df98d3cf77b54706a794226d556d649981c 26-11-2024 (15:45:26)
rpicam-apps capabilites: egl:0 qt:0 drm:1 libav:0
libcamera build: v0.3.2+99-1230f78d

So, no, doens't work as suggested....

int2018, 2025-01-06:
I used the Raspberry Pi Lite OS image to update to Bookworm and since I just need this "small" feature to add text to an image I don't want to go the compile/build path ...

So, I guess the reason is found and the case can be closed :-)

closed 2025-01-06 by naushir

arturdev, 2025-04-17:
@int2018 how did you manage to solve it? Did you succeed to add texts on video frames on Pi Lite OS? :)