raspberrypi/rpicam-apps #757: [BUG] Detecting the camera tool names (aka libcamera tool renaming)

raffaeler, 2024-12-31:
**Describe the bug**
This issue is relevant to anyone is using the libcamera apps **programmatically** inside scripts or libraries.
More specifically I refer to this issue: https://github.com/dotnet/iot/issues/2371

What is the correct way to detect the presence of `libcamera-*` or `rpicam-*` tools?
Should we assume to use the new tool names from a certain OS version on?
Should we check instead the presence in a specific directory of the OS? Which one?
What is instead the command or API that can be used to correctly adopt the correct tool set?

TIA

naushir, 2025-01-06:
Unfortunately there is no programmatic way via an API to detect the package names. ``rpicam-*`` has been the official name for Bookworm packages from the start.  Would that be enough for you to hardcode in your usage?

raffaeler, 2025-01-06:
@naushir thanks for the answer, but hardcoding is not possible because it would break backward compatibility.
The older Raspbian installations are still there and are using the `libcamera*` tools.
If we now change the hardcoded names of the tools to `rpicam-*`, a simple upgrade of the library would break the installation to a big crowd of people.

I should find a reliable way to inspect a settings (OS version?) and dynamically change the behavior at runtime.
Is there any official documentation about the tools included in each OS version?

Thanks

closed 2025-01-09 by naushir