raspberrypi/rpicam-apps #759: [BUG] Crash with exif data

FrankEscobar, 2025-01-06:
I'm trying to add some metadata to the exif list like:

```
options->exif.clear();
options->exif.push_back("ImageDescription=Test Description");

try {
    jpeg_save(spans, imgBuf.info, imgBuf.metadata, imgBuf.filename, app.CameraModel(), options);
    std::cout << "Saved Flawlessly" << std::endl;
} catch (const std::exception& e) {
    std::cerr << "Error saving JPEG: " << e.what() << std::endl;
}
```
but it always fails with

`Error saving JPEG: failed to read EXIF IFD and tag`

Any tip on this? Thank you in advance

naushir, 2025-01-09:
Can you run under a debugger (like gdb) and see exactly where this is blowing up?

naushir, 2025-01-24:
Any updates on this?

naushir, 2025-01-28:
Closing down due to inactivity.  Please do reopen if you have more info.

closed 2025-01-28 by naushir