raspberrypi/rpicam-apps #761: Pi5 (pios lite) Unable to find appropriate H.264 codec

petercwallis, 2025-01-09:
It's a brand new install of Pios (64) lite, and I've installed sway (i3 for wayland) over the top.
 rpicam-vid gives the above Error.

I tried a fix that involves unpacking mesa-vulkan-drivers:arm64 (and hence pocketsphinx ??) and am about to try that.  I am assuming it won't help, but...

naushir, 2025-01-09:
The lite OS does not include libav/ffmpeg needed for encoding on Pi 5.  Are you able to use the full Raspberry Pi OS instead?  Alternatively, you can install the dependencies manually for the codec support, but that's more tricky.

petercwallis, 2025-01-09:
Oh okay.  I see firefox has it's own plugin for H.264 so it runs under sway. Could you perhaps point me to something that says how to install the dependencies manually? - you're right; I'll probably end up installing the pios desktop and then sway but ..

petercwallis, 2025-01-09:
Ah I found this:
https://forums.raspberrypi.com/viewtopic.php?t=361268
from a year ago.  Yep it looks painful on a pi5 if things have not changed..

petercwallis, 2025-01-09:
Thanks naushir that worked.

closed 2025-01-09 by petercwallis