raspberrypi/rpicam-apps #764: [BUG] Error: "Could not connect to display" and Qt platform plugin issues with `rpicam-hello` using YOLOv8 Pose post-processing

tobiashungwe, 2025-01-17:
# Error: "Could not connect to display" and Qt platform plugin issues with `rpicam-hello` using YOLOv8 Pose post-processing

## Description:
I encountered an error when executing the `rpicam-hello` example with the YOLOv8 Pose post-processing file (`hailo_yolov8_pose.json`). The application fails to connect to the display and terminates with a Qt platform plugin error. Other post-processing files, such as `hailo_yolov5_personface.json`, work without any issues under the same setup.

---

## Steps to Reproduce:
1. Follow the setup instructions provided in the Raspberry Pi AI documentation:  
   [https://www.raspberrypi.com/documentation/computers/ai.html](https://www.raspberrypi.com/documentation/computers/ai.html).
2. SSH into the Raspberry Pi using an `-X` option to enable X11 forwarding for display output:
   ```bash
   ssh -X <your_raspberry_pi_ip>
   ```
3. Run the following command:
   ```bash
   rpicam-hello -t 0 --post-process-file /usr/share/rpi-camera-assets/hailo_yolov8_pose.json
   ```
4. Observe the error.

---

## Expected Behavior:
The application should initialize, apply the YOLOv8 Pose post-processing, and display the output without any errors.

---

## Observed Behavior:
The following error is displayed, and the application aborts:

```
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted
```

Other post-processing files (e.g., `hailo_yolov5_personface.json`) work without issues when executed with the same setup.

---

## Environment Details:
- **Hardware:** Raspberry Pi 5
- **Camera Module:** IMX500
- **Operating System:** Raspberry Pi OS (64-bit)
- **Kernel Version:** 6.1.21-v8+
- **libcamera Version:** v0.3.2+99-1230f78d
- **Qt Version:** 5.15.2
- **HailoRT Version:** 4.18.0
- **Connection Method:** SSH with `-X` option for X11 forwarding

---

## Troubleshooting Steps Taken:
1. Verified that the `hailo_yolov8_pose.json` file exists and is accessible.
2. Ensured the X server is running and the `DISPLAY` variable is correctly set.
3. Tested with other post-processing files like `hailo_yolov5_personface.json`, which work as expected.
4. Checked that the required Qt libraries and plugins (`xcb`) are installed.

---

## Possible Cause:
The issue appears to be specific to the `hailo_yolov8_pose.json` post-processing file, potentially due to its configuration or compatibility with the display setup.

---

## Suggested Fixes or Workarounds:
- Validate the configuration and dependencies of the `hailo_yolov8_pose.json` file.
- Check if the file requires any specific version of the libraries or hardware support.
- Test the file on a different setup to see if the issue persists.

---

## Additional Context:
I followed the Raspberry Pi AI documentation for setup and usage instructions:  
[https://www.raspberrypi.com/documentation/computers/ai.html](https://www.raspberrypi.com/documentation/computers/ai.html).

If additional logs or information are needed, let me know.

naushir, 2025-01-28:
Any update on this?

tobiashungwe, 2025-01-30:
It also does not work when running it locally, I am running it on the Raspberry Pi OS lite

naushir, 2025-01-31:
Raspberry Pi OS lite does not install the QT library dependences.  Can you confirm this by running ``rpicam-hello --version`` and post the output here?

naushir, 2025-02-24:
Closing due to inactivity.  Please reopen if this is still causing problems.

closed 2025-02-24 by naushir