raspberrypi/rpicam-apps #766: libcamera-hello segmentation fault

AshishShrikhande2408, 2025-01-19:
I am building a yocto setup for raspberry pi5. I am getting segmentation fault whenever I run libcamera-hello from desktop but it works perfectly if I run it from terminal. Below are the details of my build:

   1.  Linux build:
    Linux ashish 6.6.36-v8-16k SMP PREEMPT Mon Jul 1 19:39:21 UTC 2024 aarch64 GNU/Linux

    2. I have attached config.txt.

    3. I have attached build log for rpicam-apps.

    4. I have attached booting log.

    5. My desktop environment is XFCE, x11.

    6. below is my libcamera error:
    ~$ libcamera-hello [0:02:36.511728112] [798] INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d-dirty (2025-01-18T13:29:08UTC)[0:02:36.525967191] [799] INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf-dirty 30-12-2024 (06:30:32)0:02:36.637680849] [799] INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
    Segmentation fault (core dumped)

    7. With debug enabled:
    $ export LIBCAMERA_LOG_LEVELS=*:DEBUG
    ashish:$ libcamera-hello
    [0:04:17.746001578] [872] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/libcamera/ipa_rpi_pisp.so is signed
    [0:04:17.746182280] [872] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/libcamera/ipa_rpi_pisp.so'
    [0:04:17.746199544] [872] INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d-dirty (2025-01-18T13:29:08UTC)
    [0:04:17.746303889] [873] DEBUG Camera camera_manager.cpp:73 Starting camera manager
    [0:04:17.753357204] [873] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
    [0:04:17.753512917] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
    [0:04:17.754880486] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
    [0:04:17.754902677] [873] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
    [0:04:17.755003559] [873] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
    [0:04:17.755020731] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
    [0:04:17.755195041] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
    [0:04:17.755211546] [873] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
    [0:04:17.755318170] [873] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
    [0:04:17.755335935] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 9 missing dependencies
    [0:04:17.755447060] [873] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
    [0:04:17.755464583] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 9 missing dependencies
    [0:04:17.756717304] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
    [0:04:17.756735772] [873] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
    [0:04:17.757979434] [873] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
    [0:04:17.757997162] [873] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
    [0:04:17.758275077] [873] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'rpi/pisp'
    [0:04:17.758298213] [873] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
    [0:04:17.758327574] [873] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
    [0:04:17.758415599] [873] INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf-dirty 30-12-2024 (06:30:32)
    [0:04:17.759044933] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Exposure (0x00980911)
    [0:04:17.759074627] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Horizontal Flip (0x00980914)
    [0:04:17.759090594] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Vertical Flip (0x00980915)
    [0:04:17.759105710] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Wide Dynamic Range (0x009a0915)
    [0:04:17.759119418] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Camera Orientation (0x009a0922)
    [0:04:17.759138479] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Camera Sensor Rotation (0x009a0923)
    [0:04:17.759153279] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Vertical Blanking (0x009e0901)
    [0:04:17.759166728] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Horizontal Blanking (0x009e0902)
    [0:04:17.759181176] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Analogue Gain (0x009e0903)
    [0:04:17.759195384] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Red Pixel Value (0x009e0904)
    [0:04:17.759209129] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Green (Red) Pixel Value (0x009e0905)
    [0:04:17.759222781] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Blue Pixel Value (0x009e0906)
    [0:04:17.759236489] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Green (Blue) Pixel Value (0x009e0907)
    [0:04:17.759250197] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Notify Gains (0x009e0909)
    [0:04:17.759266035] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Link Frequency (0x009f0901)
    [0:04:17.759281650] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Pixel Rate (0x009f0902)
    [0:04:17.759296414] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Test Pattern (0x009f0903)
    [0:04:17.759315716] [873] DEBUG V4L2 v4l2_device.cpp:634 'imx708_wide': Control: Digital Gain (0x009f0905)
    [0:04:17.759628030] [873] DEBUG CameraSensor camera_sensor.cpp:408 'imx708_wide': No static test pattern map for 'imx708_wide'
    [0:04:17.839218789] [873] DEBUG V4L2 v4l2_device.cpp:634 'dw9807 4-000c': Control: Focus, Absolute (0x009a090a)
    [0:04:17.840097177] [873] DEBUG IPAManager ipa_manager.cpp:306 IPA module /usr/lib/libcamera/ipa_rpi_pisp.so signature is valid
    [0:04:17.840161085] [873] DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /usr/lib/libcamera/ipa_rpi_pisp.so
    [0:04:17.850212152] [873] DEBUG RPiBlackLevel black_level.cpp:40 Read black levels red 4096 green 4096 blue 4096
    [0:04:17.850437885] [873] DEBUG RPiAgc agc.cpp:49 Read AGC channel
    [0:04:17.850464634] [873] DEBUG RPiAgc agc_channel.cpp:219 AgcConfig
    [0:04:17.850577018] [873] DEBUG RPiAgc agc.cpp:49 Read AGC channel
[... 21 lines cut ...]
    [0:04:17.853757046] [873] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video25[23:cap]: Opened device platform:1000880000.pisp_be: pispbe: pispbe
    [0:04:17.853802745] [873] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video28[24:out]: Opened device platform:1000880000.pisp_be: pispbe: pispbe
    [0:04:17.853821602] [873] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video21[25:out]: Opened device platform:1000880000.pisp_be: pispbe: pispbe
    [0:04:17.853866690] [873] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video26[26:cap]: Opened device platform:1000880000.pisp_be: pispbe: pispbe
    [0:04:17.853909573] [873] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video22[27:out]: Opened device platform:1000880000.pisp_be: pispbe: pispbe
    [0:04:17.853952659] [873] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video27[28:cap]: Opened device platform:1000880000.pisp_be: pispbe: pispbe
    [0:04:17.854002100] [873] INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
    [0:04:17.854025681] [873] DEBUG Camera camera_manager.cpp:159 Pipeline handler "rpi/pisp" matched
    [0:04:17.854039889] [873] DEBUG RPI pisp.cpp:865 Unable to acquire a CFE instance

    7. libcamera version is:
    ~$ libcamera-hello --version
    rpicam-apps build: v1.5.3 50958df98d3c-dirty 18-01-2025 (14:32:12)
[... cut ...]
    libcamera build: v0.3.2+99-1230f78d-dirty (2025-01-18T13:29:08UTC)

    8. qcam works perfectly indicating basic setup for libcamera has no issues.

I have been pondering on this issue now since 2 months. Can you please let me know what exactly might be the issue?

[boot log.txt](https://github.com/user-attachments/files/18468023/boot.log.txt)

[rpicam build log.txt](https://github.com/user-attachments/files/18468024/rpicam.build.log.txt)

[config.txt](https://github.com/user-attachments/files/18468026/config.txt)

naushir, 2025-01-20:
Would you be able to attach a debugger to your executable and share a callstack of when it crashed?

AshishShrikhande2408, 2025-01-20:
Unfortunately I don't have a debugger

naushir, 2025-01-20:
Can you try using ``gdb``?

`
gdb --args rpicam-hello -t 0
`

AshishShrikhande2408, 2025-01-20:
I am getting below output

~$ gdb --args rpicam-hello -t 0
GNU gdb (GDB) 15.1
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-poky-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rpicam-hello...
(No debugging symbols found in rpicam-hello)

naushir, 2025-01-20:
In the gdb prompt, type ``run``.  When the program crashes, type ``bt`` in the command prompt, to get the backtrace.

naushir, 2025-01-20:
Ok, the error is in the rpicam-apps library, specifically in creating the EGL preview window.  This is why the error does not occur in qcam, as it uses QT for the preview.

Can you put log messages through that function and see exactly where it might be failing:
https://github.com/raspberrypi/rpicam-apps/blob/e821e4aa1b3a8d3b1f11878cb5acdeb1007616fd/preview/egl_preview.cpp#L170

AshishShrikhande2408, 2025-01-20:
The code fails after line number 184. It is failing at line
if (!eglInitialize(egl_display_, &egl_major, &egl_minor))

naushir, 2025-01-20:
Assuming ``egl_display_`` is non-NULL (the code checks), the error looks to be in the EGL driver that you have built though yocto.  I'm sorry, but I do not have the expertise to help further here.  You may be able to work-around this by using the QT preview instead?

AshishShrikhande2408, 2025-01-20:
The preview from qcam is not as clear as libcamera. Thats why I thought of getting rpicam-hello to work

AshishShrikhande2408, 2025-01-20:
What do you suggest in case I want to solve this error

naushir, 2025-01-20:
Perhaps you can get yocto to build the egl library in debug mode and use ``gdb`` to trace where in the library things are failing?  Beyond that I don't have any other suggestions, sorry :(