raspberrypi/rpicam-apps #767: Help: Draw elements on the camera preview

IHateNameStealers, 2025-01-20:
Hope someone here can help.

I am trying to draw elements on the camera preview, but am having a really hard time to do so.

Any suggestions how this can be dome?

Thank's

naushir, 2025-01-20:
It's really hard to recommend anything specific without knowing exactly what you are trying to do.  Perhaps you can look at our object detection drawing stage that uses OpenCV to draw boxes and annotate text on the image?

https://github.com/raspberrypi/rpicam-apps/blob/e821e4aa1b3a8d3b1f11878cb5acdeb1007616fd/post_processing_stages/object_detect_draw_cv_stage.cpp#L58

IHateNameStealers, 2025-01-20:
So true. I should have been more clear.

I am using the rpicam to monitor a finish line and stop a timer when a person crosses it.
I got this all working fine, however aligning the camera exactly in line with the finish line is a pain in the neck. 
Drawing a virtual line on the camera preview would make things so much easier for me, and I was hoping there was an easy way of doing this.

naushir, 2025-01-24:
Closing now - feel free to reopen if you have any further questions.

closed 2025-01-24 by naushir