raspberrypi/rpicam-apps #768: the cv for AI camera (only black & white for font & bounding box)

okeyla, 2025-01-22:
Dear Sirs,
 
For AI camera's object detection demo, I would like to change the bounding box outline and font color. 
It seems it's only one channel. (black & white) from source code:
post_processing_stages/object_detect_draw_cv_stage.cpp

So I change code as below:
from Mat image(info.height, info.width, CV_8U, ptr, info.stride);
to Mat image(info.height, info.width, CV_8UC3, ptr, info.stride);

But system occurs runtime error:
terminate called after throwing an instance of 'cv::Exception'
what(): OpenCV(4.6.0) ./modules/core/src/matrix.cpp:434: error: (-215:Assertion failed) _step >= minstep in function 'Mat'
Could it be three channel(BGR) supported?

`
bool ObjectDetectDrawCvStage::Process(CompletedRequestPtr &completed_request)
{
	if (!stream_)
		return false;

	BufferWriteSync w(app_, completed_request->buffers[stream_]);
	libcamera::Span<uint8_t> buffer = w.Get()[0];
	uint32_t *ptr = (uint32_t *)buffer.data();
	StreamInfo info = app_->GetStreamInfo(stream_);

	std::vector<Detection> detections;

	completed_request->post_process_metadata.Get("object_detect.results", detections);

        //Mat image(info.height, info.width, CV_8U, ptr, info.stride);      //<--- This is origin
	Mat image(info.height, info.width, CV_8UC3, ptr, info.stride);    //<----I change to 3 channel to display in RED color
	Scalar colour = Scalar(0, 0, 255);
	int font = FONT_HERSHEY_SIMPLEX;

	for (auto &detection : detections)
	{
		Rect r(detection.box.x, detection.box.y, detection.box.width, detection.box.height);
		rectangle(image, r, colour, line_thickness_);
		std::stringstream text_stream;
		text_stream << detection.name << " " << (int)(detection.confidence * 100) << "%";
		std::string text = text_stream.str();
		int baseline = 0;
		Size size = getTextSize(text, font, font_size_, 2, &baseline);
		Point text_origin(detection.box.x + 5, detection.box.y + size.height + 5);
		putText(image, text, text_origin, font, font_size_, colour, 8);
	}

	return false;
}
`

eG9yIGF4LGF4, 2025-01-24:
+1 as faced the same problem and will be happy to get it fixed... or at least describe some workaround for this.

naushir, 2025-01-24:
Unfortunately this is not possible with the existing code.  The image that the rectangles get drawn on is in YUV format, this is why ``CV_8U`` is used as the format.  This is also why colour changes will not work, as the drawing happens on the Y/luma component only.

If you really want to change the colour, one way would be to convert the YUV image to RGB, draw on the image with ``CV_8U``, then convert back to YUV.

okeyla, 2025-01-27:
As below code inside object_detect_draw_cv_stage.cpp,
I try to use a memory to convert image format, but no matter what I change, there's no boundingbox at all...

`	Mat image(info.height, info.width, CV_8U, (uint8_t *)buffer.data(), info.stride);
	Mat rgb = Mat(info.height, info.width, CV_8UC3);
	cvtColor(image,rgb,COLOR_YUV2RGB_NV12); //<--- COLOR_YUV2RGB_NV21, COLOR_YUV2RGB_I420
`
Any ideas?

naushir, 2025-01-27:
You need to convert back from RGB -> YUV420 and replace the contents of the original buffer (``buffer.data()``) once you have drawn the lines.

naushir, 2025-01-27:
Can you consult the opencv documentation and see what ``putText`` does exactly?

naushir, 2025-01-31:
Don't plan on doing anything with rpicam-apps.  Hopefully you can use picamera2 when this is much easier to achieve.

closed 2025-01-31 by naushir