raspberrypi/rpicam-apps #770: [BUG] Preview window unavailable.

CheaterScript, 2025-02-02:
**Describe the bug**
Hi, I am using rpicam-hello on a Raspberry Pi with the Bookworm system. The window management system is Wayland, and the display is an SPI-connected LED screen. When I run `rpicam-hello`, I get the error `"Preview window unavailable,"` but it works fine with HDMI. I would like to know what the issue is.

**Additional context**
`dtoverlay=vc4-kms-v3d`
display file is `/dev/fb0`

naushir, 2025-02-03:
Are you running this through the console or through the GUI?

CheaterScript, 2025-02-07:
Thank you for your response. I am running it through the console, and the display driver I am using is `panel-mipi-dbi`. 
I noticed that an additional `DRM` device, `card1` and `card1-spi-1`, has appeared. Although `rpicam-hello` does not display a preview, the `-o` parameter can indeed save images. However, when I try to capture images using Python to call `rpicam`, it also fails, and the error message is the same as with `--qt-preview`: 

```shell
[1:28:57.358323895] [938]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[1:28:58.123560227] [943]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:28:58.129963238] [943]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[1:28:58.130341359] [943]  INFO RPI pipeline_base.cpp:1120 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:28:58.226454549] [938]  INFO Camera camera.cpp:1197 configuring streams: (0) 2592x1944-BGR888 (1) 2592x1944-SGBRG10_CSI2P
[1:28:58.227401050] [943]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted

```

I have seen discussions in the community suggesting that it might be due to insufficient memory. I am using a Raspberry Pi Zero 2W.

closed 2025-02-07 by CheaterScript