raspberrypi/rpicam-apps #775: [BUG] Accumulating SWAP usage on a Rpi Zero 2W until stream errors and freeze.

axeljerabek, 2025-02-09:
**Describe the bug**
I have a few Raspberry Pi Zero 2W with a FullHD streaming script onto a RTMP server. The streams start quite fine (with several tweaks and optimizations), but after 1-2 hours, there are several errors showing up frequently until the stream (the Rpi Zero 2W) freezes completely. The stream-script uses about 5 MB more SWAP every 5 minutes until it reaches 500-600 MB swap usage.

**Bug report**
Not applicable

I use the following script to stream:

rpicam-vid -t 0 --width 1920 --height 1080 --nopreview --exposure normal --sharpness 1.0 --contrast 1.0 --brightness 0.0 --saturation 1.0 --ev +3 --awb auto  \
--denoise auto --rotation 0 --profile main --level 4.2 --codec libav --libav-format flv -n --framerate 30 -b 3600000 --low-latency 1 --flush \
--libav-video-codec h264_v4l2m2m --libav-video-codec-opts "num_output_buffers=2;num_capture_buffers=8;strict=-2" -g 16 --metadata - --buffer-count 2 \
--libav-audio --audio-source alsa --audio-device hw:0,0 --audio-channels 1 --audio-codec aac --audio-samplerate 48000 --audio-bitrate 64000 --av-sync 0 \
--inline 1 -o "rtmp://<some-server-ip>/zero3/test"

**Additional context**

and "journalctl -u stream.service -f"

shows the errors after a while:

Feb 09 18:38:19 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:12 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:12 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:12 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:12 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:12 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:12 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:12 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:39 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:39 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:39 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:39 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:43 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:39:45 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:40:06 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:40:06 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.
Feb 09 18:41:01 zero.jerabek.fi streamfull[2409]: [alsa @ 0x5562997880] ALSA buffer xrun.

and 

[mp4 @ 0x1692c10] Encoder did not produce proper pts, making some up.
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=105, track[105]=233
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=106, track[106]=234
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=107, track[107]=235
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=108, track[108]=236
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=109, track[109]=237
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=110, track[110]=238
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=111, track[111]=239
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=116, track[116]=244
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=117, track[117]=245
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=118, track[118]=246
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=119, track[119]=247
[h264_v4l2m2m @ 0x1692830] Pkt tracking failure: pts=120, track[120]=248

Maybe also important:
I have this behaviour only on Raspberries with Alsa audio. On the ones streaming only video, it seems completely fine.
And the more swap is used, the more errors i get.

6by9, 2025-02-12:
av_sync only tells libav what the offset is between audio and video streams. If something is retimestamping frames, then it's irrelevant.

Netflix can use non-realtime encoding, so can use B-frames and other more intensive encode options to get the bitrate down at the expense of encode time.
I know nothing about what a Mero Cam might or might not be doing.

-low-latency does nothing when not using libx264 for the encode (you're using h264_v4l2m2m).

Some gpu_mem is required for the hardware H264 encode, but generally only enough for around 3 frames. The default is about right for most use cases.

axeljerabek, 2025-02-12:
Yes, the B-Frames make a difference. But usually my RTMP streams also work with a lot lower bandwith. 3,6 should be ok. And even the 8Mbit do have the same issue.

Also default values should be ok in my case. The CPU is at around 10 percent. The streams usually also work fine with nothing overclocked. I tried to overclock to 1200 MHz and the GPU to 600MHz. The system is actually not stressed with the stream at all. And there is still enough of free RAM available. The streaming script itself uses around 23MB.

[image]

naushir, 2025-02-13:
About your problem with memory usage creeping up - I can't see anything in rpicam-apps/libav_encoder that would cause this.  Perhaps there's something in alsa or even libav that could be leaking memory?

Have you tried using pulseaudio instead of alsa?

axeljerabek, 2025-02-13:
I ran valgrind, but it did not show anything. Tried pluseaudio also, then pipewire, but without any changes.
Did you try it on a Rpi Zero or Zero2? It takes about one hour with my actual settings to fill up the swap to 50 MB. Around that point, the errors start occuring. Until then, the stream is just fine and performant, without overclocking, stutters, artefacts or dropped frames.

My swap logging looks like this:

2025-02-13 03:10:01 - Swap-Verbrauch von rpicam-vid ist OK: 0 KB
2025-02-13 03:15:01 - Swap-Verbrauch von rpicam-vid ist OK: 0 KB
2025-02-13 03:20:02 - Swap-Verbrauch von rpicam-vid ist OK: 0 KB
2025-02-13 03:25:02 - Swap-Verbrauch von rpicam-vid ist OK: 13568 KB
2025-02-13 03:30:01 - Swap-Verbrauch von rpicam-vid ist OK: 13312 KB
2025-02-13 03:35:01 - Swap-Verbrauch von rpicam-vid ist OK: 13952 KB
2025-02-13 03:40:01 - Swap-Verbrauch von rpicam-vid ist OK: 13952 KB
2025-02-13 03:45:01 - Swap-Verbrauch von rpicam-vid ist OK: 28672 KB
2025-02-13 03:50:01 - Swap-Verbrauch von rpicam-vid ist OK: 28672 KB
2025-02-13 03:55:02 - Swap-Verbrauch von rpicam-vid ist OK: 36736 KB
2025-02-13 04:00:01 - Swap-Verbrauch von rpicam-vid ist OK: 36736 KB
2025-02-13 04:05:01 - Swap-Verbrauch von rpicam-vid ist OK: 36736 KB
2025-02-13 04:10:01 - Swap-Verbrauch von rpicam-vid ist OK: 36736 KB
2025-02-13 04:15:02 - Swap-Verbrauch von rpicam-vid ist OK: 44160 KB
2025-02-13 04:20:01 - Swap-Verbrauch von rpicam-vid ist OK: 44160 KB
2025-02-13 04:25:01 - Swap für rpicam-vid zu hoch (62080 KB) – starte stream neu...
2025-02-13 04:25:02 - Service stream erfolgreich neu gestartet.
2025-02-13 04:30:01 - Swap-Verbrauch von rpicam-vid ist OK: 0 KB
2025-02-13 04:35:02 - Swap-Verbrauch von rpicam-vid ist OK: 13568 KB
2025-02-13 04:40:02 - Swap-Verbrauch von rpicam-vid ist OK: 13568 KB
2025-02-13 04:45:01 - Swap-Verbrauch von rpicam-vid ist OK: 13568 KB
2025-02-13 04:50:02 - Swap-Verbrauch von rpicam-vid ist OK: 13184 KB
2025-02-13 04:55:02 - Swap-Verbrauch von rpicam-vid ist OK: 13312 KB
2025-02-13 05:00:02 - Swap-Verbrauch von rpicam-vid ist OK: 13312 KB
2025-02-13 05:05:02 - Swap-Verbrauch von rpicam-vid ist OK: 16128 KB
2025-02-13 05:10:01 - Swap-Verbrauch von rpicam-vid ist OK: 16128 KB
2025-02-13 05:15:02 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:20:01 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:25:02 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:30:01 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:35:02 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:40:02 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:45:01 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:50:01 - Swap-Verbrauch von rpicam-vid ist OK: 44928 KB
2025-02-13 05:55:01 - Swap für rpicam-vid zu hoch (68736 KB) – starte stream neu...
2025-02-13 05:55:02 - Service stream erfolgreich neu gestartet.
2025-02-13 06:00:02 - Swap-Verbrauch von rpicam-vid ist OK: 0 KB
2025-02-13 06:05:01 - Swap-Verbrauch von rpicam-vid ist OK: 0 KB
2025-02-13 06:10:01 - Swap-Verbrauch von rpicam-vid ist OK: 0 KB
2025-02-13 06:15:01 - Swap-Verbrauch von rpicam-vid ist OK: 13568 KB
2025-02-13 06:20:01 - Swap-Verbrauch von rpicam-vid ist OK: 13312 KB
2025-02-13 06:25:02 - Swap-Verbrauch von rpicam-vid ist OK: 13312 KB
2025-02-13 06:30:01 - Swap-Verbrauch von rpicam-vid ist OK: 13312 KB
2025-02-13 06:35:02 - Swap-Verbrauch von rpicam-vid ist OK: 23168 KB
2025-02-13 06:40:01 - Swap-Verbrauch von rpicam-vid ist OK: 23168 KB
2025-02-13 06:45:02 - Swap-Verbrauch von rpicam-vid ist OK: 45696 KB
2025-02-13 06:50:02 - Swap-Verbrauch von rpicam-vid ist OK: 45696 KB
2025-02-13 06:55:01 - Swap-Verbrauch von rpicam-vid ist OK: 45696 KB
2025-02-13 07:00:01 - Swap-Verbrauch von rpicam-vid ist OK: 45696 KB
2025-02-13 07:05:02 - Swap-Verbrauch von rpicam-vid ist OK: 45824 KB
2025-02-13 07:10:02 - Swap-Verbrauch von rpicam-vid ist OK: 45824 KB
2025-02-13 07:15:01 - Swap für rpicam-vid zu hoch (62848 KB) – starte stream neu...
2025-02-13 07:15:02 - Service stream erfolgreich neu gestartet.

I simply restart the stream as soon as it hits the 50MB swap usage, it is not usable at that point anymore anyways.

At the moment, I compile a new ffmpeg to have newer (and hopefully better) libav libs. But I tried that a while ago and it did not bring any advantages. I think that this behaviour is base for several issues on the smaller Rpi versions, since it does not show immediately. And if you only check RAM and CPU usage, it looks pretty fine. The swap is the actual problem, even without any other processes in the background.

axeljerabek, 2025-02-25:
Is there anything new on this topic? Did someone test it on a Rpi Zero or Zero2?
I tried switching audio servers, tested all the options, but it still consumes too much swap until errors occur. If I limit swap space, the script consumes around 13MB and does not grow anymore, since there is no more swap to be used.
still, i would need more swap for other applications on the Zero2.

naushir, 2025-02-25:
I've been away last week so unfortunately not had a chance to look at this on a Pi Zero platform.  Once I've gone through my backlog, I'll attempt to reproduce.

axeljerabek, 2025-04-01:
Additionally, the --av-sync option is also not really usefull, since the async is not the same amount of milliseconds everytime the camera stream starts. It varies in several seconds sometimes. I hope, you can have a look into the audio issues soon. This would enhance the rpicam software greatly for the smaller Raspberries :)

axeljerabek, 2025-06-20:
Thanks for this one, I will try it on one or two of my clients. Will report :)

axeljerabek, 2025-06-22:
Quick update: It seems to compile on my Pi Zero 2W. Takes while, will post later :)

axeljerabek, 2025-06-22:
Ok, both, libcamera and rpicam-apps compiled and installed. Now, there is not --libav audio option anymore. the rpicam-vid gives me these options:

root@zero3:/home/axel/dev/rpicam-apps# rpicam-vid --help
Valid options are:
  -h [ --help ] [=arg(=1)] (=0)         Print this help message
  --version [=arg(=1)] (=0)             Displays the build version number
  --list-cameras [=arg(=1)] (=0)        Lists the available cameras attached to the system.
  --camera arg (=0)                     Chooses the camera to use. To list the available indexes, use the
                                        --list-cameras option.
  -v [ --verbose ] [=arg(=2)] (=1)      Set verbosity level. Level 0 is no output, 1 is default, 2 is verbose.
  -c [ --config ] [=arg(=config.txt)]   Read the options from a file. If no filename is specified, default to
                                        config.txt. In case of duplicate options, the ones provided on the command line
                                        will be used. Note that the config file must only contain the long form
                                        options.
  --info-text arg (=#%frame (%fps fps) exp %exp ag %ag dg %dg)
                                        Sets the information string on the titlebar. Available values:
                                        %frame (frame number)
                                        %fps (framerate)
                                        %exp (shutter speed)
                                        %ag (analogue gain)
                                        %dg (digital gain)
                                        %rg (red colour gain)
                                        %bg (blue colour gain)
                                        %focus (focus FoM value)
                                        %aelock (AE locked status)
                                        %lp (lens position, if known)
                                        %temp (sensor temperature, if available)
                                        %afstate (AF state, if supported)
  --width arg (=0)                      Set the output image width (0 = use default value)
  --height arg (=0)                     Set the output image height (0 = use default value)
  -t [ --timeout ] arg (=5sec)          Time for which program runs. If no units are provided default to ms.
  -o [ --output ] arg                   Set the output file name
  --post-process-file arg               Set the file name for configuring the post-processing
  --post-process-libs arg               Set a custom location for the post-processing library .so files
  -n [ --nopreview ] [=arg(=1)] (=0)    Do not show a preview window
  -p [ --preview ] arg (=0,0,0,0)       Set the preview window dimensions, given as x,y,width,height e.g. 0,0,640,480
  -f [ --fullscreen ] [=arg(=1)] (=0)   Use a fullscreen preview window
  --qt-preview [=arg(=1)] (=0)          Use Qt-based preview window (WARNING: causes heavy CPU load, fullscreen not
                                        supported)
  --hflip [=arg(=1)] (=0)               Request a horizontal flip transform
  --vflip [=arg(=1)] (=0)               Request a vertical flip transform
  --rotation arg (=0)                   Request an image rotation, 0 or 180
  --roi arg (=0,0,0,0)                  Set region of interest (digital zoom) e.g. 0.25,0.25,0.5,0.5
  --shutter arg (=0)                    Set a fixed shutter speed. If no units are provided default to us
  --analoggain arg (=0)                 Set a fixed gain value (synonym for 'gain' option)
  --gain arg                            Set a fixed gain value
  --metering arg (=centre)              Set the metering mode (centre, spot, average, custom)
  --exposure arg (=normal)              Set the exposure mode (normal, sport)
  --ev arg (=0)                         Set the EV exposure compensation, where 0 = no change
  --awb arg (=auto)                     Set the AWB mode (auto, incandescent, tungsten, fluorescent, indoor, daylight,
                                        cloudy, custom)
  --awbgains arg (=0,0)                 Set explict red and blue gains (disable the automatic AWB algorithm)
  --flush [=arg(=1)] (=0)               Flush output data as soon as possible
  --wrap arg (=0)                       When writing multiple output files, reset the counter when it reaches this
                                        number
  --brightness arg (=0)                 Adjust the brightness of the output images, in the range -1.0 to 1.0
  --contrast arg (=1)                   Adjust the contrast of the output image, where 1.0 = normal contrast
  --saturation arg (=1)                 Adjust the colour saturation of the output, where 1.0 = normal and 0.0 =
                                        greyscale
  --sharpness arg (=1)                  Adjust the sharpness of the output image, where 1.0 = normal sharpening
  --framerate arg (=-1)                 Set the fixed framerate for preview and video modes
  --denoise arg (=auto)                 Sets the Denoise operating mode: auto, off, cdn_off, cdn_fast, cdn_hq
  --viewfinder-width arg (=0)           Width of viewfinder frames from the camera (distinct from the preview window
                                        size
  --viewfinder-height arg (=0)          Height of viewfinder frames from the camera (distinct from the preview window
                                        size)
[... cut ...]
  --frames arg (=0)                     Run for the exact number of frames specified. This will override any timeout
                                        set.
  --sync arg (=off)                     Whether to synchronise with another camera. Use "off", "server" or "client".

Do I need to compile it with --libav support somehow? I followed the libcamera and rpicam-apps compile from the documentation:
https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-rpicam-apps

Any quick tips? :)

naushir, 2025-06-23:
Oops, you don't have the libav build dependencies installed so your rpicam-apps removed support for this feature.  From the [docs](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps-without-building-libcamera), you need to run the following:
```bash
sudo apt install libavcodec-dev libavdevice-dev libavformat-dev libswresample-dev
```
before setting up the rpicam-apps meson project.

Sorry about that!

naushir, 2025-06-23:
@axeljerabek sorry I just noticed an issue with the changes on the branch https://github.com/raspberrypi/rpicam-apps/tree/av_sync.

I've fixed the problem, please update before you rebuild.  Thanks!

naushir, 2025-06-23:
Can you run ``rpicam-vid --version`` and post the output here please?

naushir, 2025-06-23:
I'm not too sure why, but alsa seems to be behaving quite badly for me with these changes.  I cannot recall if I ever really tested alsa thoroughly with the original code either.

Nevertheless, I think the changes I'm proposing for timestamp handling are better than what we had before, so I'll tidy them up and submit them.  Hopefully I can get to the bottom of the alsa problems as well, but I may not have time for that right now.

naushir, 2025-06-23:
I've just updated the branch and hopefully make the alsa source handling better.  Would you be able to give it one final test?

axeljerabek, 2025-06-23:
Sure, test the av_sync branch or the main? The code in alsa would be great :)

naushir, 2025-06-23:
Yes, just pull the latest version of the av_sync branch, thanks!

axeljerabek, 2025-06-23:
Will do and test it later, will write in about 1 hour about the results.

naushir, 2025-06-23:
Great, I'll merge the existing changes in now.

naushir, 2025-06-23:
Thanks for testing this.  I'll resolve this issue now!

closed 2025-06-23 by naushir

axeljerabek, 2025-06-29:
I got the git on most rpis now since you merged. But it still only works with pulse audio, the alsa still gives buffer xruns. For complete stream steering I would need the alsa portion. also, not all my rpis have a gui and start the pulse server, most are headless. Is there a chance that this audio timing portion makes its way into the alsa part of the code? :)

axeljerabek, 2025-06-30:
First: In general, the situation is a lot better than before. So thanks for this huge help already.

I still have the more complex command line as above, will switch to alsa back then to see if it helps. I need to take the tasks to 1 or 2 processor cores only, it is a lot more stable. I will try my system service again and give feedback. I constantly monitor the system itself if anything is writing to SD card or any other process is keeping the system busy. Will enhance my monitoring now and check the logs tomorrow.

naushir, 2025-07-07:
Another thing to try is to match the input audio sample rate to the encoded output sample rate - this will avoid any expensive sample rate conversions.

axeljerabek, 2025-07-07:
Yes, I have tried pulse also and it gives less errors I think. But still, the errors also appear with pulse audio. The microphone has 48000 sample rate and so does the stream itself, so this should fit actually. My problem with pulse is, that it does not run in the root environment, only as user, but my stream has to be run as root, since I am also steering some other things in parallel. But as I said, the errors also appear with pulse.

naushir, 2025-07-07:
What's the CPU usage like during the encode operation? It feels like things may be just about maxing out and cause these occasional glitches.

axeljerabek, 2025-07-07:
CPU and RAM usage are easy, i disabled all other services. RAM still about 45 percent available, CPU on 4 cores about 18-20 percent. When I taskset it to 2 core only, both are stable at around 44 percent. Temperature is between 55 and 65, depending on outside heat, bot no throttling appears. Check the screenshot above, that is the average system load while encoding and streaming. Should actually be fine?

axeljerabek, 2025-07-07:
oh, i do not even have to test this over night:

Jul 07 19:08:23 zero.jerabek.fi streamfull[1049]: [alsa @ 0x556203a370] ALSA buffer xrun.
Jul 07 19:08:25 zero.jerabek.fi streamfull[1049]: [alsa @ 0x556203a370] ALSA buffer xrun.
Jul 07 19:08:27 zero.jerabek.fi streamfull[1049]: [alsa @ 0x556203a370] ALSA buffer xrun.
Jul 07 19:08:27 zero.jerabek.fi systemd[1]: Stopping stream.service - ZeroCam stream service...
Jul 07 19:08:27 zero.jerabek.fi systemd[1]: stream.service: Killing process 1059 (rpicam-vid) with signal SIGKILL.
Jul 07 19:08:27 zero.jerabek.fi systemd[1]: stream.service: Deactivated successfully.
Jul 07 19:08:27 zero.jerabek.fi systemd[1]: Stopped stream.service - ZeroCam stream service.
Jul 07 19:08:27 zero.jerabek.fi systemd[1]: stream.service: Consumed 7min 19.814s CPU time.
Jul 07 19:08:27 zero.jerabek.fi systemd[1]: Starting stream.service - ZeroCam stream service...
Jul 07 19:08:28 zero.jerabek.fi systemd[1]: Started stream.service - ZeroCam stream service.
Jul 07 19:08:33 zero.jerabek.fi streamfull[1418]: [0:15:14.442617724] [1418]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+59-d83ff0a4
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [0:15:14.676426257] [1425]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [0:15:14.681832553] [1425]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media0
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [0:15:14.681950886] [1425]  INFO RPI pipeline_base.cpp:1121 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: Mode selection for 1920:1080:12:P(30)
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]:     SRGGB10_CSI2P,1536x864/120.135 - Score: 2200
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]:     SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1150
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]:     SRGGB10_CSI2P,4608x2592/14.3536 - Score: 33342.8
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [0:15:14.696090036] [1418]  INFO Camera camera.cpp:1205 configuring streams: (0) 1920x1080-YUV420 (1) 2304x1296-SBGGR10_CSI2P
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [0:15:14.698116753] [1425]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [h264_v4l2m2m @ 0x557b8f8a00]  <<< v4l2_encode_init: fmt=179/0
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [h264_v4l2m2m @ 0x557b8f8a00] Using device /dev/video11
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [h264_v4l2m2m @ 0x557b8f8a00] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: [h264_v4l2m2m @ 0x557b8f8a00] requesting formats: output=YU12 capture=H264
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: Input #0, alsa, from 'plughw:0,0':
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]:   Duration: N/A, start: 1751908114.359334, bitrate: 1536 kb/s
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]:   Stream #0:0: Audio: pcm_s16le, 48000 Hz, 2 channels, s16, 1536 kb/s
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]: Output #0, flv, to 'rtmp://192.168.178.25/zero/test':
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]:   Stream #0:0: Video: h264 (High), drm_prime(tv, bt709), 1920x1080, q=2-31, 6000 kb/s, 30 fps, 30 tbr, 1000k tbn
Jul 07 19:08:34 zero.jerabek.fi streamfull[1418]:   Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp, 128 kb/s
Jul 07 19:08:36 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:38 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:41 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:43 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:45 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:47 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:49 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:52 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.
Jul 07 19:08:54 zero.jerabek.fi streamfull[1418]: [alsa @ 0x557b901370] ALSA buffer xrun.

there is a buffer xrun every second almost :/
And it just continues.

naushir, 2025-07-08:
I'm afraid I'm out of simple ideas to try.  Going back to ``pulse_audio``, you mentioned that it cannot run as root... do you get any explicit errors?

axeljerabek, 2025-07-08:
Jost grabbing it and will test asap :) Thanks for your efforts again!

axeljerabek, 2025-07-08:
Ok, just tested it and set the period size from 256, 512, 1024, 2048, 4096, 8192 and 16384. but I keep getting the xruns still.

Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: [h264_v4l2m2m @ 0x558896fcf0]  <<< v4l2_encode_init: fmt=179/0
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: [h264_v4l2m2m @ 0x558896fcf0] Using device /dev/video11
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: [h264_v4l2m2m @ 0x558896fcf0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: [h264_v4l2m2m @ 0x558896fcf0] requesting formats: output=YU12 capture=H264
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: libav: Set ALSA period size to 2049 frames
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: Input #0, alsa, from 'plughw:0,0':
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]:   Duration: N/A, start: 1751998192.277649, bitrate: 1536 kb/s
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]:   Stream #0:0: Audio: pcm_s16le, 48000 Hz, 2 channels, s16, 1536 kb/s
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: Output #0, flv, to 'rtmp://192.168.178.25/zero/test':
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]:   Stream #0:0: Video: h264 (High), drm_prime(tv, bt709), 1920x1080, q=2-31, 4000 kb/s, 30 fps, 30 tbr, 1000k tbn
Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]:   Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp, 128 kb/s
Jul 08 20:09:59 zero.jerabek.fi streamfull[66123]: [alsa @ 0x558898c930] ALSA buffer xrun.
Jul 08 20:10:09 zero.jerabek.fi streamfull[66123]: [alsa @ 0x558898c930] ALSA buffer xrun.
Jul 08 20:10:19 zero.jerabek.fi streamfull[66123]: [alsa @ 0x558898c930] ALSA buffer xrun.
Jul 08 20:10:29 zero.jerabek.fi streamfull[66123]: [alsa @ 0x558898c930] ALSA buffer xrun.
Jul 08 20:10:39 zero.jerabek.fi streamfull[66123]: [alsa @ 0x558898c930] ALSA buffer xrun.

It does not seem to be able to set the period size correctly, since this line:

Jul 08 20:09:52 zero.jerabek.fi streamfull[66123]: libav: Set ALSA period size to 2049 frames

seems that alsa is setting it internally back to 2049?

naushir, 2025-07-10:
I don't really understand ALSA enough to say what's going on here I'm afraid.  But back to pulse audio, you can't use it because it doesn't run under root.  Maybe we can explore trying to see if that can work?

axeljerabek, 2025-07-10:
Yes, ALSA is a pain. Just to make sure, I purged all ALSA packages including the pulse connections and configs and reinstalled everything from scratch. Still buffer xruns, but since your update from last week, it really got a lot better. I had xruns every few minutes and many in a row, now it is around 2 per hour. This keeps the streams a lot better. I am happy to test some more for you, if you have any updates.

What bugs me more at the moment is the DTS problem, that is the second stream killer:

journalctl -u stream.service -f
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]: [h264_v4l2m2m @ 0x557da8b5e0]  <<< v4l2_encode_init: fmt=179/0
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]: [h264_v4l2m2m @ 0x557da8b5e0] Using device /dev/video11
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]: [h264_v4l2m2m @ 0x557da8b5e0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]: [h264_v4l2m2m @ 0x557da8b5e0] requesting formats: output=YU12 capture=H264
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]: Input #0, alsa, from 'hw:0,0':
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]:   Duration: N/A, start: 1752074026.204727, bitrate: 768 kb/s
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]:   Stream #0:0: Audio: pcm_s16le, 48000 Hz, 1 channels, s16, 768 kb/s
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]: Output #0, flv, to 'rtmp://192.168.178.25/zero/test':
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]:   Stream #0:0: Video: h264 (High), drm_prime(tv, bt709), 1920x1080, q=2-31, 6000 kb/s, 30 fps, 30 tbr, 1000k tbn
Jul 09 17:13:46 zero.jerabek.fi streamfull[1193]:   Stream #0:1: Audio: aac (LC), 48000 Hz, mono, fltp, 128 kb/s
**Jul 09 17:40:44 zero.jerabek.fi streamfull[1193]: [flv @ 0x557da8b9a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 1617396 >= 1617363
Jul 09 17:40:44 zero.jerabek.fi streamfull[1193]: terminate called after throwing an instance of 'std::runtime_error'
Jul 09 17:40:44 zero.jerabek.fi streamfull[1193]:   what():  libav: error writing output: Invalid argument
Jul 09 17:40:44 zero.jerabek.fi streamfull[1192]: /usr/bin/streamfull: Zeile 10:  1193 Abgebrochen**             nice -n -19 chrt -r 99 taskset -c 0,1,2,3 rpicam-vid -t 0 --width 1920 --height 1080 --nopreview 1 --low-latency 1 --hdr off --flush 1 --buffer-count 4 --exposure long --sharpness 1.1 --contrast 1.2 --brightness 0.0 --saturation 1.0 --ev +1.3 --awb auto --profile high --level 4.2 --codec libav --libav-audio 1 --audio-source alsa --audio-device hw:0,0 --audio-channels 1 --audio-codec aac --audio-samplerate 48000 --audio-bitrate 128000 --libav-format flv -n --framerate 30 -b 6000000 --autofocus-mode manual --lens-position 0.8 --denoise auto --autofocus-window 0.25,0.25,0.5,0.5 --inline 1 -o "rtmp://192.168.178.25/zero/test"
Jul 09 17:40:44 zero.jerabek.fi systemd[1]: stream.service: Deactivated successfully.
Jul 09 17:40:44 zero.jerabek.fi systemd[1]: stream.service: Consumed 17min 59.290s CPU time.
Jul 09 17:40:49 zero.jerabek.fi systemd[1]: stream.service: Scheduled restart job, restart counter is at 2.
Jul 09 17:40:49 zero.jerabek.fi systemd[1]: Stopped stream.service - ZeroCam stream service.

In general, I would be happy. if audio timings and buffers would somehow sync when a stream starts and then stay stable for the time playing. I am not an audiophile, but no great Hollywood movie is without proper (and synched) sounds :)

(Except the ones from Charlie Chaplin)

But a lot of thanks for your help, it got A LOT better! :)

naushir, 2025-07-11:
Perhaps it's worth getting pulse audio working under root?  I've found this: https://stackoverflow.com/questions/66775654/how-can-i-make-pulseaudio-run-as-root which might help?