raspberrypi/rpicam-apps #780: Request full-range YUV420?

schoolpost, 2025-02-25:
Is there a way to request full-range YUV420? I am inspecting the luma channel on an image captured from a yuv420 Lores stream and it looks to be 16-235, I would like 0-255 if that is possible?

naushir, 2025-02-25:
This should be possible, but it may not be plumbed in fully right now.  What exact command line are you running, and I assume this is on a Pi 5?

schoolpost, 2025-02-25:
> This should be possible, but it may not be plumbed in fully right now. What exact command line are you running, and I assume this is on a Pi 5?

Yes Pi 5, I am working with a custom app on top of rpicam-apps, so I'm ok to dive into the source to make changes if need be. 

I see some of this going on in the ConfigureVideo:
```// Now we get to override any of the default settings from the options_->
	StreamConfiguration &cfg = configuration_->at(0);
	cfg.pixelFormat = libcamera::formats::YUV420;
	cfg.bufferCount = 6; // 6 buffers is better than 4
	if (options_->buffer_count > 0)
		cfg.bufferCount = options_->buffer_count;
	if (options_->width)
		cfg.size.width = options_->width;
	if (options_->height)
		cfg.size.height = options_->height;
	if (flags & FLAG_VIDEO_JPEG_COLOURSPACE)
		cfg.colorSpace = libcamera::ColorSpace::Sycc;
	else if (cfg.size.width >= 1280 || cfg.size.height >= 720)
		cfg.colorSpace = libcamera::ColorSpace::Rec709;
	else
		cfg.colorSpace = libcamera::ColorSpace::Smpte170m;
	configuration_->orientation = libcamera::Orientation::Rotate0 * options_->transform;

	post_processor_.AdjustConfig("video", &configuration_->at(0));```

and looks like lores just copies the same attributes from the main stream:

```	if (have_lores_stream)
	{
		Size lores_size(options_->lores_width, options_->lores_height);
		lores_size.alignDownTo(2, 2);
		if (lores_size.width > configuration_->at(0).size.width ||
			lores_size.height > configuration_->at(0).size.height)
			throw std::runtime_error("Low res image larger than video");
		configuration_->at(lores_index).pixelFormat = lores_format_;
		configuration_->at(lores_index).size = lores_size;
		configuration_->at(lores_index).bufferCount = configuration_->at(0).bufferCount;
	}
	configuration_->orientation = libcamera::Orientation::Rotate0 * options_->transform;```

But how would I specify the output range?

naushir, 2025-02-25:
@davidplowman

naushir, 2025-02-25:
It seems we only ever use the default colourspace for the low res stream. Perhaps something like this is all you need?

```diff
diff --git a/core/rpicam_app.cpp b/core/rpicam_app.cpp
index e1038dc38c83..f887f1f92d12 100644
--- a/core/rpicam_app.cpp
+++ b/core/rpicam_app.cpp
@@ -588,6 +588,7 @@ void RPiCamApp::ConfigureVideo(unsigned int flags)
                configuration_->at(lores_index).pixelFormat = lores_format_;
                configuration_->at(lores_index).size = lores_size;
                configuration_->at(lores_index).bufferCount = configuration_->at(0).bufferCount;
+               configuration_->at(lores_index).colorSpace = configuration_->at(0).colorSpace;
        }
        configuration_->orientation = libcamera::Orientation::Rotate0 * options_->transform;
```

schoolpost, 2025-02-25:
I'll give that a go.

I suppose it means Colorspace and Output range are coupled in that case? at least from what I can tell here:

https://github.com/raspberrypi/libcamera/blob/29156679717bec7cc4784aeba3548807f2c27fca/src/libcamera/pipeline/rpi/pisp/pisp.cpp#L264
```
void setupOutputClipping(const V4L2DeviceFormat &v4l2Format,
			 pisp_be_output_format_config &outputFormat)
{
	const PixelFormat &pixFormat = v4l2Format.fourcc.toPixelFormat();
	const PixelFormatInfo &info = PixelFormatInfo::info(pixFormat);

	if (info.colourEncoding != PixelFormatInfo::ColourEncodingYUV)
		return;

	if (v4l2Format.colorSpace == ColorSpace::Sycc) {
		outputFormat.lo = 0;
		outputFormat.hi = 65535;
		outputFormat.lo2 = 0;
		outputFormat.hi2 = 65535;
	} else if (v4l2Format.colorSpace == ColorSpace::Smpte170m ||
			v4l2Format.colorSpace == ColorSpace::Rec709) {
		outputFormat.lo = 16 << 8;
		outputFormat.hi = 235 << 8;
		outputFormat.lo2 = 16 << 8;
		outputFormat.hi2 = 240 << 8;
	} else {
		LOG(RPI, Warning)
			<< "Unrecognised colour space "
			<< ColorSpace::toString(v4l2Format.colorSpace)
			<< ", using full range";
		outputFormat.lo = 0;
		outputFormat.hi = 65535;
		outputFormat.lo2 = 0;
		outputFormat.hi2 = 65535;
	}
}
```

schoolpost, 2025-02-25:
Yea unfortunately no luck with this change:
```
 		configuration_->at(lores_index).pixelFormat = lores_format_;
 		configuration_->at(lores_index).size = lores_size;
 		configuration_->at(lores_index).bufferCount = configuration_->at(0).bufferCount;
+		configuration_->at(lores_index).colorSpace = libcamera::ColorSpace::Sycc;
 	}
```

Inspecting the luma channel, I'm still seeing a limited range:

[image]

6by9, 2025-02-25:
Colorspace and Output range shouldn't need to be coupled.

The `colorSpace` class has separate fields for Primaries, TransferFunction, YcbcrEncoding, and Range - https://github.com/raspberrypi/libcamera/blob/main/include/libcamera/color_space.h#L17
This is in common with V4L2 formats, but many devices don't support all the options.

schoolpost, 2025-02-25:
Right, but so far as I can tell; that is not implemented on PISP pipeline side of libcamera?

6by9, 2025-02-25:
> Right, but so far as I can tell; that is not implemented on PISP pipeline side of libcamera? the use of the range attribute in the case.

I'll agree that it appears not to be implemented for PISP, and based on a `git grep` no other pipeline does either (the display sinks do)

naushir, 2025-02-25:
Just backtracking a bit to get a better understanding of the problem - You want full-range YUV on the low-res stream *and* the main stream, is that correct?  With the rpicam-apps unmodified code, do you get full range on the main stream by using ``Sycc`` colourspace?

It looks like all the plumbing in libcamera is present to set the low res stream for full range, provided you also set the colourspace to ``Sycc``.

schoolpost, 2025-02-25:
> Just backtracking a bit to get a better understanding of the problem - You want full-range YUV on the low-res stream _and_ the main stream, is that correct? With the rpicam-apps unmodified code, do [...]

Did not work when I set the Lores exclsuively to `Sycc`, but applying `Sycc` to the MainStream did give us full-range on the Lores stream as well.

closed 2025-02-25 by schoolpost

schoolpost, 2025-02-26:
It would be preferential if I can find a method to set the colorspace for the streams I'm using from the application level, without having to modify the core source of rpicam-apps; Is this possible?

naushir, 2025-02-26:
@davidplowman should we assume the lores colourspace must follow the main colourspace like picamera2 does?

naushir, 2025-02-28:
This should now be available with https://github.com/raspberrypi/rpicam-apps/pull/782