raspberrypi/rpicam-apps #781: [BUG] rpicam-hello symbol lookup error

jpendric, 2025-02-26:
**Describe the bug**
rpicam-hello throws this error: 
 $ rpicam-hello
rpicam-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp22compute_optimal_strideER24pisp_image_format_config

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.

[bug.txt](https://github.com/user-attachments/files/18993991/bug.txt)

**Additional context**
Add any other context about the problem here.

aaronwmorris, 2025-02-27:
I think something is wrong with the package dependencies somewhere, but if you upgrade your packages, the issue should get corrected.

```
sudo apt-get update

sudo apt-get upgrade
```

jpendric, 2025-03-01:
I rebuilt libcamera, and that fixed it.
Thanks.

I think something is wrong with the package dependencies somewhere, but if you upgrade your packages, the issue should be corrected.