raspberrypi/rpicam-apps #783: [BUG] Not able to detect camera rpi hd after upgrade system

pat685, 2025-03-02:
I have rpi 5 and rpi hd camera. On new sd mini card, I installed 64 bits system version. After system installation, I used rpicam command to verify the camera is detected by system. The camera is detected. After, I process upgrade system (sudo apt-get update and sudo apt-get upgrade). After upgrade, I re-used rpicam command. The system don t detect camera. I tried to install allsky anyway and it doesn t detect the camera too. I used report camera bug. the file is attached. I attached too command of cmd windows.

What is wrong ?
Thanks
Patrice

[bug.txt](https://github.com/user-attachments/files/19044678/bug.txt)

[cmd command.txt](https://github.com/user-attachments/files/19044738/cmd.command.txt)

pat685, 2025-03-03:
I'm not a linux specialist. Can you write the commands that I have need to do ?

naushir, 2025-03-04:
In that case, I think you are best off reinstalling your SD card image and updating all your libraries when prompts you on first boot.  That will get you a working system without any local dependencies.

But since you have it working now, maybe just leave as-is 😄

closed 2025-03-04 by naushir