raspberrypi/rpicam-apps #785: Error "Failed to import fd 29" after upgrade

johnbailon, 2025-03-18:
Hi, on a [10" Waveshare touchscreen with built in ov5647 5mp camera](https://www.waveshare.com/wiki/RPI-HMI-080D-101D#Install_the_System_and_Driver_Manually), running `rpicam-hello` used to work until the last  update (`sudo apt update && sudo apt upgrade -y`). Now I am getting this error:

```sh
pi@raspberry:~ $ rpicam-hello
[0:19:20.198950137] [8325]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:19:20.210660664] [8328]  INFO RPI pisp.cpp:720 libpisp version v1.1.0 e7974a156008 27-01-2025 (21:50:51)
[0:19:20.235959806] [8328]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/ov5647@36 to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_D0
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI3: Screen seems not DRI3 capable
Made X/EGL preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:19:20.294394699] [8325]  INFO Camera camera.cpp:1202 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-GBRG_PISP_COMP1
[0:19:20.294498496] [8328]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected CFE format: 1296x972-PC1g
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 29
Aborted
```
Notes:
I tried running on a fresh install and `rpicam-hello` works on the same hardware.
`rpicam-hello --qt-preview` still works but I need the non QT window.
This is all running locally, not remote or an ssh/vnc session so that is not the issue.

Anyone else run into this problem or have any idea fixing this?

6by9, 2025-03-18:
Duplicate of https://github.com/raspberrypi/bookworm-feedback/issues/381.
Fix is being tested at the moment, and should hit apt within 48 hours.

johnbailon, 2025-03-22:
This is fixed. See this [thread](https://github.com/raspberrypi/bookworm-feedback/issues/381#issuecomment-2733738594).

closed 2025-03-22 by johnbailon