raspberrypi/rpicam-apps #787: no pkg-config for rpicam-apps

GrubbyHalo, 2025-03-20:
It would be nice to maybe have a rpicam-apps-dev package to install a pkg-config for the rpicam-apps libraries so that build systems can locate and link against them. 

Also, is there a specific reason why the standard 'lib' prefix was omitted on the rpicam_app library ?

naushir, 2025-03-26:
@XECDesign for visibility.

> It would be nice to maybe have a rpicam-apps-dev package to install a pkg-config for the rpicam-apps libraries so that build systems can locate and link against them.

I can setup the meson build file to create rpicam_app.so and install it appropriately.

> Also, is there a specific reason why the standard 'lib' prefix was omitted on the rpicam_app library ?

Unfortunately this is a hangover from the days that the library was called libcamera_apps.so.  Unfortunately I don't think we will be changing that at this point.

naushir, 2025-04-01:
https://github.com/raspberrypi/rpicam-apps/pull/790 should hopefully provide this file.

closed 2025-04-23 by naushir