raspberrypi/rpicam-apps #788: Preview not work on the SPI screen.

CheaterScript, 2025-03-29:
Hi, I'm a beginner and would like to ask a follow-up question about my previous issue #770 .

I've noticed that `rpicam` seems to only trigger previews on `HDMI` displays. When I connect only my `SPI` screen:

- The `SPI` screen correctly shows the command line interface
- 
- I can successfully write images via `/dev/fb0`
- 
- However, running `rpicam-hello -t 0` doesn't display any preview

I'm curious:

- Is this because the code specifically restricts previews to `HDMI`?
- 
- Or is it caused by some `Linux` system configuration (e.g., driver/device mapping)?

I'd really appreciate your insights on this. Thank you for your help!

6by9, 2025-03-31:
Similar discussion as https://github.com/raspberrypi/picamera2/issues/1231

drm_preview actually explicitly looks for the "vc4" DRM driver - https://github.com/raspberrypi/rpicam-apps/blob/main/preview/drm_preview.cpp#L247

Note that you'll also need a DRM driver for your SPI display, not an FBTFT one. Your other issue says you're using "panel-mipi-dbi", so you're covered there.

CheaterScript, 2025-04-02:
> Similar discussion as [raspberrypi/picamera2#1231](https://github.com/raspberrypi/picamera2/issues/1231)
> 
> drm_preview actually explicitly looks for the "vc4" DRM driver - https://github.com/raspberrypi/rpicam-apps/blob/main/preview/drm_preview.cpp#L247
> 
> Note that you'll also need a DRM driver for your SPI display, not an FBTFT one. Your other issue says you're using "panel-mipi-dbi", so you're covered there.

Thank you for your response! I really appreciate your help.

closed 2025-04-02 by CheaterScript