raspberrypi/rpicam-apps #791: [BUG] --sync option did not work as described in the document (early timeout)

EwingKang, 2025-04-01:
**Describe the bug**
Following the guide for `--sync` from the document
https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-vid
As I understand it, with the following two commands on two terminal, I should get two 20s videos that has (software) synced frame.
```
rpicam-vid -n -t 20s --camera 1 --codec libav -o client.mp4 --sync client
```
```
rpicam-vid -n -t 20s --camera 0 --codec libav -o server.mp4 --sync server
```
It seems like the timeout option is enforced regardless of the sync status. 
For example, first I start the client, and took 5 seconds to start the server. After ~3 seconds (100 frames) I can see the sync info `RPiSync sync.cpp:281 *** Sync achieved! Difference 34us` on both terminals. However, both terminals stops the recording with message `Halting: reached timeout of 20000 milliseconds.` 
The process produces a 17-seconds `server.mp4`, and a 12-seconds `client.mp4`

**Bug report**

[bug-report.txt](https://github.com/user-attachments/files/19548152/bug-report.txt)

[client-report.txt](https://github.com/user-attachments/files/19548274/client-report.txt)
[server-report.txt](https://github.com/user-attachments/files/19548275/server-report.txt)

**Additional context**
I tried some perturbations with `initial`, `keypress`, and `signal` option. None of the above worked. I suspect something is going very wrong with the library and application.
I have `apt upgrade` on my rpi5 with libcamera-ipa:arm64 version `0.4.0+rpt20250213-1`

naushir, 2025-04-01:
This has already been fixed in commit https://github.com/raspberrypi/rpicam-apps/commit/b7632f4d4b5c40a64c17dd4ee7096a779c698d00.  Unfortunately that fix did not make the last 1.6.0 release.

Are you able to manually build/install rpicam-apps?  If so, you can run with this until we make the next package release of rpicam-apps, which will include this fix.

EwingKang, 2025-04-02:
Hay Naushir thanks for your apt response. I tried the latest build, the --sync portion of the `rpicam-vid` worked as expected.
However I do find some problems.
### One less frame
With following server and client command
```
# client
./rpicam-vid --camera 1 -n -t 20s --framerate 10 --width 1640 --height 1232 --level 4.2 --codec libav -o cam1_client.mp4 --sync client
# server
./rpicam-vid --camera 0 -n -t 20s --framerate 10 --width 1640 --height 1232 --level 4.2 --codec libav -o cam0_server.mp4 --sync server
```
The client side always produces 199 frames (I20/P99/B80), one less then the server side of 200 frames (I20/P100/B80).
* `--buffer-count` doesn't help as described in the [document](https://www.raspberrypi.com/documentation/computers/camera_software.html#buffer-count).
* Setting `-t 10s` does produces two videos with 100 frames
* Setting `-t 25s` produces 249 frames from client (I25/P124/B100) and 250 frames from server (I25/P125/B100)
* Setting `-t 40s` produces 400 frames from client (I40/P200/B160) and 399 frames from server (I40/P199/B160)
* Removing `--width` and `--height` and thus reverting it to the default 640*480 does not change the result
* Removeing the `--framerate 10` and thus reverting it to the default 30 FPS produces correct frames in both 20s (I20/P300/B280) and 25s (I25/P375/B350) case.
* I do recall seeing once that is the reverse situation: the server side produces one less frame than the client side, but I have no evidance

### Unable to start `rpicam-vid` paused with `--initial` option
Regardless the use of `--sync` option, the `--initial pause` doesn't seems to work. 
* `--initial pause --keypress` starts the recording immediately without waiting for user input. 'x' + Enter does stops the recording at any time
* `--initial 'pause' --signal` starts the recording immediately without waiting for signal. `kill -s SIGUSR2 $(ps -C rpicam-vid)` does stops the recording at any time

For the further issues, would you like me to start a new ticket, edit the title of this ticket, or anything else?

naushir, 2025-04-02:
Using ``-t`` to capture a fixed number of frames is not advisable.  This is because the time sampling is non-deterministic.  Instead, you should use the ``--frames`` option to capture a fixed number of frames.  For example, ``--frames 300`` will capture and encode exactly 300 frames and then stop.

For the ``--initial`` problem, can you open another issue please?

EwingKang, 2025-04-04:
@naushir Thank you for your response, I'll open a dedicated ticket for the `initial` next Monday with detailed reports. I think this can be closed. Thanks again.

naushir, 2025-04-04:
Sure, closing this now.

closed 2025-04-04 by naushir