raspberrypi/rpicam-apps #796: [BUG] rpicam-vid --initial option not working at all

EwingKang, 2025-04-07:
**Describe the bug**
`--initial pause` for `rpicam-vid` does not prevent video from recording and timeout.
For example the following command finishes in ~7s and produces a correct length video without waiting for the user to press "Enter".
```bash
./rpicam-vid --camera 0 -n --frames 200 --width 1640 --height 1232 --level 4.2 --codec libav -o cam0_server.mp4 --initial pause -k
```

Neither `-k` nor `--signal` work.

**Bug report**
[report_initial.txt](https://github.com/user-attachments/files/19627040/report_initial.txt)

**Additional context**
I'm trying to use it with `--sync` and `--signal` option. I want to would start a client, start a server, and trigger the server/client with SIGUSR1. 
Reference document: https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-vid

naushir, 2025-04-23:
Unfortunately, this is because of the use of the libav backend, which does not correctly support pause/split/segment, etc correctly.  We will look into fixing this hopefully soon.

naushir, 2025-06-27:
With https://github.com/raspberrypi/rpicam-apps/pull/832 merged, you should now be able to use this feature, but only for elementary H.264 streams without any containers.

closed 2025-06-27 by naushir

EwingKang, 2025-08-13:
Thanks for the apt fixes.
I tried the latest debian release, the behavior is a little weird. 
rpicam-apps build: v1.8.1 eb293b055248 21-07-2025 (11:50:44)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.5.1+100-e53bdf1f

The following command still times out in 10 seconds. It produces a zero-byte cam0_server.h264 file. It seems like the `--frames 300` option start counting the frame even if the state should be paused. If any key is pressed within 10 seconds, the video will still timeout at the same point, but it seems like everything after key press is recorded.
```bash
./rpicam-vid --camera 0 -n --frames 300 -t 0 --width 1640 --height 1232 --level 4.2 --codec libav -o cam0_server.h264 --initial pause -k
```

However, if we omit the `--frames` option, the key trigger would work normally. And I can stop the recording with "x" and "Enter". 
```
./rpicam-vid --camera 0 -n --frames 300 -t 0 --width 1640 --height 1232 --level 4.2 --codec libav -o cam0_server.h264 --initial pause -k
```

@naushir I wonder if we should consider this as a bug? Intuitively, the `--frames` option should control **the amount of frames recorded** instead of the frames that reached the application (which I presume is what is happening)