raspberrypi/rpicam-apps #799: Rpi 4 8gb + HQ Cam + Thomas Jacquin Allsky + Arducam Ethernet Extension

MrNbaphoto, 2025-04-22:
Hi everyone,

I am having an issue where images are hanging, when stopping Allsky then Rebooting it seems I get a small handful of photos then it stops. I thought I was having an issue with RJ45 so I replaced it and no change. I then formatted the Micro SD and did a fresh install with the newest version of PI and also TJ Allsky. Same issue.. Done a fair bit of googling and tried a few things but to no avail...
 
I have had some discussions on the Allsky Git Page [https://github.com/AllskyTeam/allsky/discussions/4409](url)

We have exhausted options and I was suggested to report it here in hope of some help.

I am very basic when it comes to Linux so if you can be a little gentle and explain how to do things that would be great. I am logging into Rpi via SSH until I get a HDMI to Micro HDMI to connect it to a screen

Rpi 4 8gb, Rpi HQ Cam, TJ Allsky Current Version, Arducam Ethernet extension, Fresh Formatted 128gb Card, Fresh Install 64bit PI with Desktop

Thanks in advance. Micheal

naushir, 2025-04-22:
From https://github.com/AllskyTeam/allsky/discussions/4409:

```
[93:12:50.409301353] [47239] INFO Camera camera.cpp:1008 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a ***
```

This suggests (as mentioned in the above discussion) that one of the earlier ``rpicam-still`` processes has stalled.  Can you run ``rpicam-hello --version`` and provide the output please?  Also, can you confirm the full command line of the process that has stalled?

6by9, 2025-04-22:
Have you tested without the Arducam ethernet (really CAT5) extension?

I'm assuming it is https://docs.arducam.com/Camera-Extension-Solution/Ethernet-Extension-Kit/, in which case AIUI it is an active device which internally requires configuring for the CSI2 configuration. Arducam have prebaked some settings into it, but no-one else knows what those settings are. I can't even see a reference as to what chipset they are using.

MrNbaphoto, 2025-04-22:
Was running Snagless Cat6 cable FYI.

That is a negative, though my previous iteration ran POE on that version of TJ Allsky, ran fine to the camera. This iteration ran for over a year with no problems at all till more recently.

You are assuming correctly that is what I am using.

naushir, 2025-04-22:
As @6by9 mentioned, my first suggestion would be to remove the Adrucam extention hardware from your setup and see if you still observe the lockup.

EricClaeys, 2025-04-22:
@MrNbaphoto, Michael,
To provide @naushir with the command being run, do this:
~~~
grep "Running:" /var/log/allsky.log | tail -1
~~~
That will display the last rpicam-still command.  Make sure you copy the command and all the arguments after it.

@naushir,
Most of the rpicam-still command lines from users have the same arguments in the same order, but obviously with different values.
Users select settings like auto/manual exposure via a web interface, and Allsky creates and executes the command.  The only "free form" field, which isn't used very often, is the "Extra Parameters" setting which is primarily for uncommon rpicam-still settings Allsky doesn't know about, like the focus-related ones.

> 6:37:52.605105589] [6074] INFO Camera camera.cpp:1008 Pipeline handler in use by another process 

Unless there's a bug in Allsky, it's very unlikely there will ever be multiple `rpicam-still` command running at once.  ***Before every image***, the Allsky C program runs
~~~
system("pkill --signal SIGKILL rpicam-still");
~~~

MrNbaphoto, 2025-04-23:
@EricClaeys I tried running that and it doesn't come up with anything at all?

`grep "Running:" /var/log/allsky.log | tail -1`

EricClaeys, 2025-04-23:
@MrNbaphoto, sorry about that.
Temporarily change the `Debug Level` in the WebUI to 2, wait a couple minutes and run the command again.  The "Running" command is only logged at level 2 or higher.

MrNbaphoto, 2025-04-23:
Thanks 👍 

nbaphoto@nbaphoto:~ $ grep "Running:" /var/log/allsky.log | tail -1
2025-04-23T15:16:35.629431+10:00 nbaphoto allsky[19755]:  > Running: rpicam-still --thumb none --output '/home/nbaphoto/allsky/tmp/image-20250423151635.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 3982 --analoggain 1 --awb auto --quality 100

MrNbaphoto, 2025-04-23:
@EricClaeys something interesting for you. Stopped and restarted Allsky, still had debug level 2. 
Got this message: Can't determine what command to use for RPi camera.   (April 23, 4:08:12 pm AEST)

[image]

EricClaeys, 2025-04-23:
When Allsky starts it runs a couple different commands to try and find the camera, but none of them worked.  It's most likely due to whatever is causing rpicam-still to hang.

naushir, 2025-04-23:
The command line all looks reasonable to me.

My (very uneducated) suspicion is that the earlier ``rpicam-still`` process may be stalled because of a sensor device timeout.  This causes subsequent ``rpicam-still`` processes to fail in this way.  Is there any way of gathering logs for all runs so we can check this?  A timeout error message will be displayed if this happens.

EricClaeys, 2025-04-25:
@naushir, the Allsky log file records every `rpicam-still` command it runs, but we only keep stdout/stderr from rpicam-still from the last run.  That's where this output:
~~~
[93:12:50.409301353] [47239] INFO Camera camera.cpp:1008 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a ***
~~~
came from.  What other information are you referring to with "Is there any way of gathering logs for all runs" ?
I can probably give the users having this problem a modified Allsky that did more/different logging.

If it makes any difference, Allsky actually calls `rpicam-still` with `LIBCAMERA_LOG_LEVELS=ERROR,FATAL` prepended to the command line.

Shouldn't running
~~~
system("pkill --signal SIGKILL rpicam-still");
~~~
kill any prior hung `raspi-still` command?  SIGKILL should kill anything, although I believe it's possible the process can be killed but because no other process did a "wait()" on it, it becomes a zombie process that may still keep resources open.

brianboru82, 2025-04-26:
I've had the same issue, attached is the trace file for the process after it hung. 
Here's the version info: 

[image]

[rpicam_debug_trace_479584.txt](https://github.com/user-attachments/files/19922563/rpicam_debug_trace_479584.txt)

naushir, 2025-04-28:
@MrNbaphoto and @brianboru82 did you have a chance to run this with the sensor attached directly to the Raspberry Pi and without the Adrucam Ethernet extender?

MrNbaphoto, 2025-04-28:
@naushir my unit is sealed on the roof, hence the extension kit. Don't have a spare camera to test with the hat off

naushir, 2025-04-28:
We definitely want to check if the preceding process has had a HW timeout event occurring to cause this.  With the absence of any other information, I believe this is what's happening.

When a timeout occurs, we attempt to restart the device, but this may not be successful, causing the process to lockup in an undetermined state.  The below change will instead just quit the process with an exception.  Would you be able to try this out and see if it unblocks the next process from running correctly?

```diff
diff --git a/apps/rpicam_still.cpp b/apps/rpicam_still.cpp
index 5e97aa79fbde..ba4797bfd5ab 100644
--- a/apps/rpicam_still.cpp
+++ b/apps/rpicam_still.cpp
@@ -222,10 +222,7 @@ static void event_loop(RPiCamStillApp &app)
                RPiCamApp::Msg msg = app.Wait();
                if (msg.type == RPiCamApp::MsgType::Timeout)
                {
-                       LOG_ERROR("ERROR: Device timeout detected, attempting a restart!!!");
-                       app.StopCamera();
-                       app.StartCamera();
-                       continue;
+                       throw std::runtime_error("ERROR: Device timeout detected, quitting!!!");
                }
                if (msg.type == RPiCamApp::MsgType::Quit)
                        return;
```

naushir, 2025-04-28:
> Shouldn't running
>
> ```
> system("pkill --signal SIGKILL rpicam-still");
> ```
>
> kill any prior hung `raspi-still` command? SIGKILL should kill anything, although I believe it's possible the process can [...]

``SIGKILL`` should indeed kill it completely AFAIK.  Perhaps it's worth adding a ``sleep`` call to wait a number of seconds before starting the new ``rpicam-still`` process as the kill may take a few seconds?

EricClaeys, 2025-04-28:
@naushir,
How would someone get your modified program?

Where was the `LOG_ERROR("ERROR...` sent?  That message never appeared in the rpicam-still output.

naushir, 2025-04-28:
> [@naushir](https://github.com/naushir), How would someone get your modified program?

Right now the users would have to manually apply the change to ``rpicam-vid.cpp`` and rebuild/install locally.  If this is not possible, I can provide a pre-built binary with this change.

> Where was the `LOG_ERROR("ERROR...` sent? That message never appeared in the rpicam-still output.

The exception should be logged to ``stderr`` if I'm not mistaken.

naushir, 2025-04-28:
Do you capture both ``stdout`` and ``stderr`` as I'm really guessing where it logs to.

brianboru82, 2025-04-28:
> [@MrNbaphoto](https://github.com/MrNbaphoto) and [@brianboru82](https://github.com/brianboru82) did you have a chance to run this with the sensor attached directly to the Raspberry Pi and without the Adrucam Ethernet extender?

My setup has the HQ camera directly connected to the Raspberry Pi with the ribbon cable, not using any extender.

brianboru82, 2025-04-28:
> Right now the users would have to manually apply the change to `rpicam-vid.cpp` and rebuild/install locally. If this is not possible, I can provide a pre-built binary with this change.
>
>
> The [...]

I could probably try this in the next couple days. Is everything I need to do in the command line in that code segment above? If it's not too much trouble, having a pre-built binary might be easier.

EricClaeys, 2025-04-28:
> Do you capture both `stdout` and `stderr` as I'm really guessing where it logs to.

Yes, we capture stdout and stderr.

brianboru82, 2025-04-29:
Okay, I built the rpicam-apps with the changes mentioned above. I'll let you know how it goes. The process hangs have been random, sometimes once a day, sometimes a week between.

naushir, 2025-05-01:
@brianboru82 do you have any updates on your testing?

brianboru82, 2025-05-01:
It has been working fine since the change. I'll let you know if something happens. Let me know if there's any log or something that would be helpful.

EricClaeys, 2025-05-02:
@naushir, @brianboru82,
No one's ever reported the `ERROR: Device timeout detected, attempting a restart!!!` message which implies that's never been the problem, so I'm not sure if the change will help.

@brianboru82,
Any output from the `rpicam-still` command is written to `~/allsky/tmp/capture_RPi_debug.txt`, so if there's a problem that's a good file to look in.  Note that the file only has the LAST command's output.

EricClaeys, 2025-05-02:
@naushir,
Every image is taken by running
~~~
rpicam-still --shutter ....    >  tmp/capture_RPi_debug.txt  2>&1
~~~
so the file is overwritten every frame.

If `rpicam-still` fails 5 times in a row, Allsky displays a message in the Web interface, adds a message to the /var/log/allsky.log file, then quits.  In the cases above, however, `rpicam-still` hung, so Allsky didn't quit.

Assuming the user saves`capture_RPi_debug.txt` before restarting Allsky, the file will be from the _last run_.  The examples above are from the last invocation.

In a couple cases the user ran `strace` on the hung `rpicam-still` command.  If they also saved `capture_RPi_debug.txt` it would have been from the hung `rpicam-still`.

<hr>

If it would help you, I can have Allsky copy the contents of `capture_RPi_debug.txt` to the Allsky log file whenever `rpicam-still fails`, which will tell us if prior failures were for the same reason.

FYI, the name of the C++ program that calls `rpicam-still` is `capture_RPi`.  Allsky also supports ZWO cameras, and its capture program is called `capture_ZWO`.  ZWO has a SDK so the `capture_ZWO` program directly controls the camera and gets metadata directly from the camera.  That's a LOT easier to deal with than having to call a command to take a picture, then read a metadata file.

EricClaeys, 2025-05-02:
@naushir,
In the future we may move to Python, but not yet.  Where is the `libcamera` API?  Is it production-quality?

naushir, 2025-05-02:
> [@naushir](https://github.com/naushir), In the future we may move to Python, but not yet. Where is the `libcamera` API? Is it production-quality?

The ``rpicam-apps`` API (and consequently ``rpicam-still``) is a wrapper around the [``libcamera API``](https://libcamera.org/api-html/index.html) with the addition of image/video encode, preview, postprocessing, etc functionality.  This is the same for Picamera2, only done in Python.

The ``libcamera`` library is production ready - we have been using it under the hood for over 5 years now, but the API is subject to change as it's still in development.

naushir, 2025-05-07:
Any further updates on the testing?

brianboru82, 2025-05-07:
It's been working, I had to restart the pi the other day because it said it couldn't detect the camera, but I don't think that is related to this?

EricClaeys, 2025-05-07:
@brianboru82,
Not finding the camera (which @MrNbaphoto  also reported) and hanging commands in my view are both cases where the communication between the Pi (or the software on the Pi) and the camera is not working properly.  People with ZWO cameras don't report these problems. 

@MrNbaphoto,
Any update?

@naushir,

Given the arguments to the futex() calls:
futex(0x7fe6f406fc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, FUTEX_BITSET_MATCH_ANY
futex(0x7fc181fd88, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, FUTEX_BITSET_MATCH_ANY

Is there any way to determine where in the code it's hanging, and hence what the problem is?

&nbsp;
For these messages, can we get a version of the command that lists the PID and ideally the process name of the "other process"?

INFO Camera camera.cpp:1008 Pipeline handler in use by another process
ERROR: *** failed to acquire camera

MrNbaphoto, 2025-05-07:
@EricClaeys nothing here. Still can't get it fired up. Have run sudo updates etc. But still can't find camera. I have a HDMi to Micro HDMI so I can connect to a monitor if you have some logs you want me to download?

MrNbaphoto, 2025-05-08:
Lib camera hello didn't pick up on a camera will try list when I'm home tonight

brianboru82, 2025-05-08:
> [@brianboru82](https://github.com/brianboru82), Not finding the camera (which [@MrNbaphoto](https://github.com/MrNbaphoto) also reported) and hanging commands in my view are both cases where the communication between the Pi (or the software on the Pi) and the camera is not working properly. People with ZWO cameras don't report these problems.
> 
> [@MrNbaphoto](https://github.com/MrNbaphoto), Any update?
> 
> [@naushir](https://github.com/naushir),
> 
> Given the arguments to the futex() calls: futex(0x7fe6f406fc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, FUTEX_BITSET_MATCH_ANY futex(0x7fc181fd88, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, FUTEX_BITSET_MATCH_ANY
> 
> Is there any way to determine where in the code it's hanging, and hence what the problem is?
> 
>   For these messages, can we get a version of the command that lists the PID and ideally the process name of the "other process"?
> 
> INFO Camera camera.cpp:1008 Pipeline handler in use by another process ERROR: *** failed to acquire camera

Yeah, makes sense. I did grab the debug log before I restarted, it is attached below.

The errors from this same time from the allsky.log are also attached

[allsky_log_excerpt.txt](https://github.com/user-attachments/files/20095991/allsky_log_excerpt.txt)
 

[capture_RPi_debug.txt.txt](https://github.com/user-attachments/files/20095980/capture_RPi_debug.txt.txt)

EricClaeys, 2025-05-08:
@brianboru82,

This is a new error:
~~~
dmaHeap allocation failure for rpicam-apps0
ERROR: *** failed to allocate capture buffers for stream ***
~~~
Interesting that yesterday @NeilAllison147 reported this same error.

naushir, 2025-05-08:
> Is there any way to determine where in the code it's hanging, and hence what the problem is?

The only way I can think of is to create a debug build of rpicam-apps, and then ``gdb`` can provide a full stack trace when things go wrong.

naushir, 2025-05-08:
In order to debug any issues here effectively I suggest we strip back to basics.  Take a Pi4 or Pi5 and attach the (official RPi) IMX477 camera directly to it - without any Arducam extender.  With a fresh OS image with all official updates applied, run ``rpicam-still`` in a loop continuously outside of the Allsky framework.  Is this error reproducible in that environment?

I've tried this setup over a couple of days and unfortunately I cannot reproduce any of the above lockups.

naushir, 2025-05-14:
Here's the ``rpicam_apps.so`` and ``rpicam-still`` binaries built with debug information which might give us more clues:
[rpicam_app.zip](https://github.com/user-attachments/files/20202406/rpicam_app.zip)

@brianboru82 did you see any more errors with the timeout patch above?

XECDesign, 2025-05-14:
You should also be able to run `sudo apt install rpicam-apps-dbgsym` (or `rpicam-apps-lite-dbgsym` if you're using the lite version), attach gdb with `gdb --pid=whatever` and run `bt` to get a backtrace.

brianboru82, 2025-05-14:
@naushir I have not seen any further errors since last week. 

I can install that debug version tonight most likely.

NeilAllison147, 2025-05-15:
How would I go about installing the debug version?
Bear in mind I'm a complete novice when it comes to the RPi OS :)

brianboru82, 2025-05-15:
So my camera was working all today, and this evening ran the `sudo apt install rpicam-apps-dbgsym` to install the debug version.

Now it isn't seeing the camera at all. 

Running 'rpicam-still --list-cameras' returns 'Aborted'.

Here is what I'm seeing:

[image]

Let me know what else I can try.

naushir, 2025-05-15:
The latest rpicam-apps package is linked to the latest libcamera package which is version 0.5.0.  Can you make sure you have updated the libcamera package as well?

brianboru82, 2025-05-15:
Yes, I did update the libcamera package as well, after that, it said libcamera0.4 was no longer used and I could use autoremove to remove it, which I did.

After that, when I run rpicam-hello or libcamera-hello, I get the following error. It sounds like something is still referencing the old package? 

[image]

brianboru82, 2025-05-15:
I think I found my issue for the above, I had done the local build of rpicam-apps with the change mentioned last week, and that was interfering with the debug version installed just above.

brianboru82, 2025-05-25:
I was running the system manually from the command line (something is up with the allsky service at the moment), so I hit ctrl C, and it spit this information out. Not sure if it's helpful. 

[image]

brianboru82, 2025-05-26:
Yes, when I hit ctrl C with the debug at 4, it says 'got SIGINT to exit'

[image]

Here's the backtrace after I had restarted things. I'll definitely try it next time it hangs. 
It was interesting, when I did the gdb command, it seemed to pause/hang allsky while it was active, then when I ended it, allsky continued on.
[image]

EricClaeys, 2025-05-26:
@brianboru82,

Regarding gdb, recall that Allsky invokes rpicam-still, and waits for it to finish.  If it hangs, Allsky waits forever (or until you restart Allsky).  When running gdb, it may be pausing rpicam-still, so Allsky continues waiting.  Ending gdb let rpicam-still finish, when then let Allsky continue.

<hr>

There appears to be a bug in Allsky - when you press CTRL-C Allsky gets the signal, outputs "got SIGINT to exit", and displays the "Allsky Not Running" image.  But it appears Allsky isn't exiting.
Would you please attach /var/log/allsky.log IN A NEW ALLSKY GITHUB PAGE, not here, since this is an Allsky issue, not rpicam-still.

You can set the Debug Level back to what it was before.  Level 4 shows the "got SIGINT" message, and I wanted to make sure that part worked.

brianboru82, 2025-05-26:
@EricClaeys actually it did exit a few lines later, it appeared it was finishing up the other tasks it was running before it stopped completely.

Also, it appears when running allsky from the command line rather than through the service, nothing actually gets added to the log. I'll create an issue related to that.

EricClaeys, 2025-05-30:
@naushir,
Any thoughts on the `gdb` output from @brianboru82 ?
Anything else he should try when it hangs again?

Eric

EricClaeys, 2025-05-30:
@brianboru82,
for the command @naushir asked you to run, before running it run:
~~~
sudo systemctl stop allsky
cd ~/allsky/tmp
~~~
This makes sure Allsky's not running and puts the `test.jpg` file in ~/allsky/tmp, which is in memory, so it won't write to the SD card.

To get the process id (PID) of the `rpicam-still` command to run `gdb` when it hangs, run `pgrep rpicam-still`.

That's a lot easier and more accurate than `ps -ef | grep rpicam-still | grep -v grep`...

brianboru82, 2025-06-03:
@naushir @EricClaeys My setup has the RPI HQ cam directly connected to the Pi using the ribbon cable, no extender at all.

I can try this, is this something I do once it hangs again, or do I get it running at a 'normal' state and it will catch the hang when it happens?

@EricClaeys , after I do the stop allsky and cd allsky/tmp, and run @naushir 's command, do I start allsky again, or is this a separate thing to run until we get a hang?

naushir, 2025-06-03:
If the process hangs and the loop locks up, you can gdb into the ``rpicam-still`` process and get a call trace.  The direct call to ``rpicam-still`` should guarantee no allsky service will attempt to kill the process.

brianboru82, 2025-06-04:
I did catch a hang today, and ran `gdb `on the process. Here are the screenshots:

[image]

[image]

[image]

[image]

[image]

The log file it created (partway through) is also attached. 

[gdb.txt](https://github.com/user-attachments/files/20599411/gdb.txt)

After I closed the gdb session, the camera routine (the `while true;` command line from above) started working again.

brianboru82, 2025-06-05:
1. Unfortunately, as soon as a quit the GDB session, the rpicam-still command while loop started running again by itself, and I wasn't able to grab a screenshot. I'll try to do that next time it happens..
2. Bug report file here:
[bug-report.txt](https://github.com/user-attachments/files/20610983/bug-report.txt)
3. The camera is mounted on my roof, it'll be a few days at least before I can get up there.

brianboru82, 2025-06-06:
Here's the photos of my Raspberry Pi HQ camera board.

[image]

[image]

[image]

naushir, 2025-06-09:
Thanks for the debug info!  I'm afraid I've got no idea what might be happening, the error seems to be deep within libcamera.  I'll attempt to reproduce this so I can debug it myself.  @brianboru82 approximately how long did it take for the loop to crash?

brianboru82, 2025-06-09:
> Thanks for the debug info! I'm afraid I've got no idea what might be happening, the error seems to be deep within libcamera. I'll attempt to reproduce this so I can debug it myself. [@brianboru82](https://github.com/brianboru82) approximately how long did it take for the loop to crash?

Yes, you're welcome! I'd say about a day and a half for this time, though it varies. There have been times it takes just a few hours, others it takes a week or two. I haven't noticed any pattern or correlation for it.

[40 more replies]