raspberrypi/rpicam-apps #804: question about restarting video stream

Pantelorian, 2025-04-25:
We need ability to reconfigure stream params(fps, resolution) in real time.
So the question is: what is the correct way to do that?
Do we need stop camera, encoder, teardown, close camera then reconfigure and open and start everything? Or there's way to do less things? We need as minimal as possible time for changing params

naushir, 2025-04-25:
FPS can be reconfigured while running using the ``FrameDurations`` control.  Resolution must require a stop/configure/start cycle.

closed 2025-04-25 by Pantelorian

Pantelorian, 2025-07-02:
Well, it looks like we reconfiguring, not really stable anywhere, but it works.
But we figured out some very strange problem, for example if we start video 1280x720, than reconfigure to 640x480 and then again 1280x720, new 720p not the same as old 720p. It looks like it tooks 640x480 and crops it to 720p's aspect ratio
What could be the problem? Options_ tooks correct resolution

naushir, 2025-07-02:
You will have to provide more information on this:

- libcamera console logs
- Sensor detail
- Changes you have made

to start with so we can help debug further.

Pantelorian, 2025-07-02:
sensor: imx219 camera v2.1

logs: 

             ├─618 rpicam-vid -t 0 --inline -n --rotation 180 --framerate 30 --width 1640 --height 922 -g 10 -b 5000000 -o -

июн 06 00:37:48 raspberrypi systemd[1]: Started send_video.service - Video Streaming Service.
июн 06 00:37:50 raspberrypi send_video.sh[618]: [0:00:15.696350617] [618]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
июн 06 00:37:50 raspberrypi send_video.sh[618]: [0:00:15.781859836] [646]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
июн 06 00:37:50 raspberrypi send_video.sh[618]: [0:00:15.786773794] [646]  WARN RPI vc4.cpp:393 Mismatch between Unicam and CamHelper for embedded data usage!
июн 06 00:37:50 raspberrypi send_video.sh[618]: [0:00:15.788003326] [646]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
июн 06 00:37:50 raspberrypi send_video.sh[618]: [0:00:15.788096919] [646]  INFO RPI pipeline_base.cpp:1120 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
июн 06 00:37:50 raspberrypi send_video.sh[618]: Mode selection for 1640:922:12:P(30)
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,640x480/206.654 - Score: 5220.23
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,1640x1232/41.8515 - Score: 2420.22
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,1920x1080/47.5737 - Score: 1112.39
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,3280x2464/21.1941 - Score: 20750.1
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB8,640x480/206.654 - Score: 6220.23
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB8,1640x1232/83.703 - Score: 3420.22
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB8,1920x1080/47.5737 - Score: 2112.39
июн 06 00:37:50 raspberrypi send_video.sh[618]:     SRGGB8,3280x2464/21.1941 - Score: 21750.1
июн 06 00:37:50 raspberrypi send_video.sh[618]: Stream configuration adjusted
июн 06 00:37:50 raspberrypi send_video.sh[618]: [0:00:15.808102284] [618]  INFO Camera camera.cpp:1197 configuring streams: (0) 1640x922-YUV420 (1) 1920x1080-SRGGB10_CSI2P
июн 06 00:37:50 raspberrypi send_video.sh[618]: [0:00:15.808635774] [646]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected unicam format: 1920x1080-pRAA
июн 06 00:37:50 raspberrypi send_video.sh[618]: Camera streams configured
июн 06 00:37:50 raspberrypi send_video.sh[618]: Available controls:
июн 06 00:40:05 raspberrypi send_video.sh[618]: Setting framerate to 24
июн 06 00:40:05 raspberrypi send_video.sh[618]: Setting resolution to 1920x1080
июн 06 00:40:05 raspberrypi send_video.sh[618]: Restarting camera with new parameters...
июн 06 00:40:05 raspberrypi send_video.sh[618]: ERROR: Device timeout detected, attempting a restart!!!
июн 06 00:40:05 raspberrypi send_video.sh[618]: Restart already in progress, skipping this request
июн 06 00:40:05 raspberrypi send_video.sh[618]: ERROR: Device timeout detected, attempting a restart!!!
июн 06 00:40:05 raspberrypi send_video.sh[618]: Restart already in progress, skipping this request
июн 06 00:40:05 raspberrypi send_video.sh[618]: ERROR: Device timeout detected, attempting a restart!!!
июн 06 00:40:05 raspberrypi send_video.sh[618]: Restart already in progress, skipping this request
июн 06 00:40:05 raspberrypi send_video.sh[618]: ERROR: Device timeout detected, attempting a restart!!!
июн 06 00:40:05 raspberrypi send_video.sh[618]: Restart already in progress, skipping this request
июн 06 00:40:05 raspberrypi send_video.sh[618]: ERROR: Device timeout detected, attempting a restart!!!
июн 06 00:40:05 raspberrypi send_video.sh[618]: Restart already in progress, skipping this request
июн 06 00:40:05 raspberrypi send_video.sh[618]: Mode selection for 1920:1080:10:P(24)
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,640x480/206.654 - Score: 5093.33
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,1640x1232/41.8515 - Score: 1937.83
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,1920x1080/47.5737 - Score: 0
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB10_CSI2P,3280x2464/21.1941 - Score: 7637.68
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB8,640x480/206.654 - Score: 6093.33
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB8,1640x1232/83.703 - Score: 2937.83
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB8,1920x1080/47.5737 - Score: 1000
июн 06 00:40:05 raspberrypi send_video.sh[618]:     SRGGB8,3280x2464/21.1941 - Score: 8637.68
июн 06 00:40:05 raspberrypi send_video.sh[618]: Stream configuration adjusted
июн 06 00:40:05 raspberrypi send_video.sh[618]: [0:02:30.571627597] [655]  INFO Camera camera.cpp:1197 configuring streams: (0) 1920x1080-YUV420 (1) 1920x1080-SRGGB10_CSI2P
июн 06 00:40:05 raspberrypi send_video.sh[618]: [0:02:30.583762493] [646]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected unicam format: 1920x1080-pRAA
июн 06 00:40:05 raspberrypi send_video.sh[618]: Camera streams configured
июн 06 00:40:05 raspberrypi send_video.sh[618]: Available controls:
июн 06 00:40:05 raspberrypi send_video.sh[618]: Camera restarted with new parameters: resolution 1920x1080, framerate 24
[... cut ...]
	auto duration = std::chrono::duration_cast<std::chrono::milliseconds>(end_time - start_time).count();

	LOG(1, "Camera restarted with new parameters: resolution " << options->width << "x" << options->height << 
		", framerate " << (options->framerate ? options->framerate.value() : DEFAULT_FRAMERATE) << std::endl<<
		" (время перезапуска: " << duration << " мс"  << std::endl);
	
	
	restart_in_progress = false;
}

socket control gets new params, puts them in options_ and then calls restart func 
loaded src to this repo:
https://github.com/Pantelorian/rpicam-apps-1.5-socket_control

naushir, 2025-07-08:
> июн 06 00:40:05 raspberrypi send_video.sh[618]: ERROR: Device timeout detected, attempting a restart!!!
> июн 06 00:40:05 raspberrypi send_video.sh[618]: Restart already in progress, skipping this [...]

This suggests something might be fundamentally wrong with your sensor.  It's not likely related to software at all.