raspberrypi/rpicam-apps #805: [BUG] Build failing

bou6, 2025-04-30:
**Describe the bug**
Build issue probably related to [this](https://github.com/raspberrypi/rpicam-apps/issues/547)

I followed the instructions mentioned in this [page](https://www.raspberrypi.com/documentation/computers/camera_software.html#advanced-rpicam-apps) for building.

I cloned the latest [libcamera](https://git.linuxtv.org/libcamera.git/), but still having build issues

`c++ -Iapps/rpicam-vid.p -Iapps -I../apps -I. -I.. -I/usr/include/libcamera -I/usr/include -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -DBOOST_ALL_NO_LIB -MD -MQ apps/rpicam-vid.p/rpicam_vid.cpp.o -MF apps/rpicam-vid.p/rpicam_vid.cpp.o.d -o apps/rpicam-vid.p/rpicam_vid.cpp.o -c ../apps/rpicam_vid.cpp
In file included from ../apps/rpicam_vid.cpp:14:
../core/rpicam_encoder.hpp: In member function ‘void RPiCamEncoder::StartEncoder()’:
../core/rpicam_encoder.hpp:38:58: error: ‘SyncMode’ is not a member of ‘libcamera::controls::rpi’
   38 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeOff);
      |                                                          ^~~~~~~~
../core/rpicam_encoder.hpp:38:94: error: ‘SyncModeOff’ is not a member of ‘libcamera::controls::rpi’
   38 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeOff);
      |                                                                                              ^~~~~~~~~~~
../core/rpicam_encoder.hpp:40:58: error: ‘SyncMode’ is not a member of ‘libcamera::controls::rpi’
   40 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeServer);
      |                                                          ^~~~~~~~
../core/rpicam_encoder.hpp:40:94: error: ‘SyncModeServer’ is not a member of ‘libcamera::controls::rpi’
   40 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeServer);
      |                                                                                              ^~~~~~~~~~~~~~
../core/rpicam_encoder.hpp:42:58: error: ‘SyncMode’ is not a member of ‘libcamera::controls::rpi’
   42 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeClient);
      |                                                          ^~~~~~~~
../core/rpicam_encoder.hpp:42:94: error: ‘SyncModeClient’ is not a member of ‘libcamera::controls::rpi’
   42 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeClient);
      |                                                                                              ^~~~~~~~~~~~~~
../core/rpicam_encoder.hpp: In member function ‘bool RPiCamEncoder::EncodeBuffer(CompletedRequestPtr&, Stream*)’:
../core/rpicam_encoder.hpp:54:91: error: ‘SyncReady’ is not a member of ‘libcamera::controls::rpi’
   54 |                 if (GetOptions()->sync && !completed_request->metadata.get(controls::rpi::SyncReady).value_or(false))
      |                                                                                           ^~~~~~~~~
[16/26] Compiling C++ object apps/rpicam-raw.p/rpicam_raw.cpp.o
FAILED: apps/rpicam-raw.p/rpicam_raw.cpp.o 
c++ -Iapps/rpicam-raw.p -Iapps -I../apps -I. -I.. -I/usr/include/libcamera -I/usr/include -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -DBOOST_ALL_NO_LIB -MD -MQ apps/rpicam-raw.p/rpicam_raw.cpp.o -MF apps/rpicam-raw.p/rpicam_raw.cpp.o.d -o apps/rpicam-raw.p/rpicam_raw.cpp.o -c ../apps/rpicam_raw.cpp
In file included from ../apps/rpicam_raw.cpp:10:
../core/rpicam_encoder.hpp: In member function ‘void RPiCamEncoder::StartEncoder()’:
../core/rpicam_encoder.hpp:38:58: error: ‘SyncMode’ is not a member of ‘libcamera::controls::rpi’
   38 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeOff);
      |                                                          ^~~~~~~~
../core/rpicam_encoder.hpp:38:94: error: ‘SyncModeOff’ is not a member of ‘libcamera::controls::rpi’
   38 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeOff);
      |                                                                                              ^~~~~~~~~~~
../core/rpicam_encoder.hpp:40:58: error: ‘SyncMode’ is not a member of ‘libcamera::controls::rpi’
   40 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeServer);
      |                                                          ^~~~~~~~
../core/rpicam_encoder.hpp:40:94: error: ‘SyncModeServer’ is not a member of ‘libcamera::controls::rpi’
   40 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeServer);
      |                                                                                              ^~~~~~~~~~~~~~
../core/rpicam_encoder.hpp:42:58: error: ‘SyncMode’ is not a member of ‘libcamera::controls::rpi’
   42 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeClient);
      |                                                          ^~~~~~~~
../core/rpicam_encoder.hpp:42:94: error: ‘SyncModeClient’ is not a member of ‘libcamera::controls::rpi’
   42 |                         cl.set(libcamera::controls::rpi::SyncMode, libcamera::controls::rpi::SyncModeClient);
      |                                                                                              ^~~~~~~~~~~~~~
../core/rpicam_encoder.hpp: In member function ‘bool RPiCamEncoder::EncodeBuffer(CompletedRequestPtr&, Stream*)’:
../core/rpicam_encoder.hpp:54:91: error: ‘SyncReady’ is not a member of ‘libcamera::controls::rpi’
   54 |                 if (GetOptions()->sync && !completed_request->metadata.get(controls::rpi::SyncReady).value_or(false))
`
**Question**
which commits/version should I pick up from rpicam-apps and libcamera to resolve the build Problem

Thanks in Advance

naushir, 2025-04-30:
Please try building/installing libcamera from this tree: https://github.com/raspberrypi/libcamera

This should then resolve your compilation issues above.

elcalzado, 2025-05-14:
I'm currently working on integrating `libcamera-apps` into a Buildroot environment for a Raspberry Pi Zero 2W. My end goal is to successfully run the [uvc-gadget](https://gitlab.freedesktop.org/camera/uvc-gadget) while utilizing `libcamera` for camera functionality. However, I keep running into a persistent error: **"No cameras detected"**.

Here's the relevant things I've done so far:

1. **Started with Buildroot Defconfig:**

   * I used the `raspberrypizero2w_64_defconfig` as my base configuration.

2. **Modified Toolchain:**

   * Adjusted the toolchain settings to include the necessary headers and dependencies required by `libcamera-apps`.

3. **Enabled Required Packages:**

   * Enabled `libcamera` and `libcamera-apps` in the Buildroot configuration.
   * Set `/dev management` to use the `eudev` option, as it seemed necessary for device detection.

4. **Version Pinned Dependencies:**

   * I manually updated the `.mk` files for both `libcamera` and `libcamera-apps` to use specific commits that I know are compatible. These commits were tested successfully on Raspberry Pi OS Lite. Specific commit hashes are below.

5. **Modified libcamera Source Repository:**

   * Configured the `libcamera` package in Buildroot to pull directly from the `raspberrypi/libcamera` GitHub repository instead of the official upstream repository.

6. **Verified Compatibility on Raspberry Pi OS:**

   * Using the same versions of `libcamera` and `libcamera-apps`, I was able to successfully compile and run the applications on Raspberry Pi OS Lite. This confirms that the versions and configuration are compatible, but the issue seems isolated to Buildroot.

---

### Observed Behavior

When running the UVC gadget in my Buildroot setup—before I changed the toolchain and tried compiling `libcamera-apps`—I consistently encountered the error:
**"No cameras detected"**, as well as a **"no IPA modules found"** warning.

After changing the toolchain, enabling `libcamera-apps`, and making the changes mentioned above to the `.mk` files, I encounter a new error when I run `make`:

```cpp
../core/rpicam_app.cpp: In member function ‘void RPiCamApp::StartCamera()’:
../core/rpicam_app.cpp:642:78: error: ‘controls::rpi’ has not been declared
  642 |         if (!controls_.get(controls::ScalerCrop) && !controls_.get(controls::rpi::ScalerCrops))
      |                                                                              ^~~
../core/rpicam_app.cpp:673:49: error: ‘controls::rpi’ has not been declared
  673 | controls_.set(controls::rpi::ScalerCrops, libcamera::Span<const Rectangle>(crops.data(), crops.size()));
      |                                                 ^~~
[11/33] Compiling C++ object rpicam_app.so.1.7.0.p/image_jpeg.cpp.o
```

---

### Questions

* Is there any additional configuration required in Buildroot to ensure proper camera detection?
* Has anyone successfully integrated `libcamera-apps` with Buildroot? I don't understand why it fails to build in Buildroot when I'm using two compatible versions. Is changing the version not enough?

Any help or guidance would be greatly appreciated! If additional logs or specifics are needed, let me know, and I'll provide them.

Thanks in advance!

---

### Additional Context

* [`libcamera`](https://github.com/raspberrypi/libcamera) Commit: `d83ff0a4ae4503bc56b7ed48cd142c3dd423ad3b`
* [`libcamera-apps`](https://github.com/raspberrypi/rpicam-apps) Commit: `5a3f5965aca96c2d575261e4e4045f0e0481279b`

naushir, 2025-05-19:
@elcalzado please open another issue to discuss your problem.

closed 2025-05-19 by naushir

FilipDominec, 2025-05-22:
@elcalzado Coincidentally, I, too, need to get the ```libcamera``` & co. functionality under buildroot environment on Rp Zero W. (I have described my situation here: https://github.com/issues/created?issue=showmewebcam%7Cshowmewebcam%7C243) 

Getting this working is important for my work. Can I help you in any way?