raspberrypi/rpicam-apps #809: [BUG] DMA Heap Missing on Fresh Install - camera fails to start

wallarug, 2025-05-28:
**Describe the bug**
On a fresh install of the latest Raspberry Pi OS (using RPi Imager), there is an issue with how the camera is setup and configured.

Two errors occur that need separate solutions.

# Issue 1 - DMA heap access

**Error:** 
```bash
user@hostname:~ $ libcamera-still --list-cameras
Could not open any dmaHeap device
Available cameras
-----------------
0 : ov64a40 [9248x6944 10-bit] (/base/axi/pcie@1000120000/rp1/i2c@88000/ov64a40@36)
    Modes: 'SBGGR10_CSI2P' : 1920x1080 [45.65 fps - (784, 1312)/7712x4352 crop]
                             2312x1736 [26.75 fps - (0, 0)/9280x6976 crop]
                             3840x2160 [14.89 fps - (784, 1312)/7712x4352 crop]
                             4624x3472 [7.66 fps - (0, 0)/9280x6976 crop]
                             8000x6000 [2.58 fps - (624, 472)/8048x6032 crop]
                             9248x6944 [2.02 fps - (0, 0)/9280x6976 crop]

1 : ov64a40 [9248x6944 10-bit] (/base/axi/pcie@1000120000/rp1/i2c@80000/ov64a40@36)
    Modes: 'SBGGR10_CSI2P' : 1920x1080 [45.65 fps - (784, 1312)/7712x4352 crop]
                             2312x1736 [26.75 fps - (0, 0)/9280x6976 crop]
                             3840x2160 [14.89 fps - (784, 1312)/7712x4352 crop]
                             4624x3472 [7.66 fps - (0, 0)/9280x6976 crop]
                             8000x6000 [2.58 fps - (624, 472)/8048x6032 crop]
                             9248x6944 [2.02 fps - (0, 0)/9280x6976 crop]
```

## Solution
As outlined here:  https://github.com/raspberrypi/rpicam-apps/issues/218#issuecomment-1020738780

# Issue 2 - DMA System folder missing
The `vidbuf_cached` folder (symlink) is not present on fresh install of Raspbian.  This presents itself as the error message below from running `libcamera-still`

```bash
[0:03:42.776568363] [1726]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/ov64a40@36 - Selected sensor format: 4624x3472-SRGGB10_1X10 -
dmaHeap allocation failure for rpicam-apps1
ERROR: *** failed to allocate capture buffers for stream ***
```

I discovered this by fluke comparing two systems.  A working one, and a broken one (fresh install).
Evidence:
```bash
user@hostname:~ $ ls /dev/dma_heap/
linux,cma  system
user@hostname:~ $ ls -l /dev/dma_heap/
total 0
crw-rw---- 1 root video 253, 1 May 28 21:00 linux,cma
crw-rw---- 1 root video 253, 0 May 28 21:00 system
```
Expected:
```bash
user@hostname:~ $ ls -l /dev/dma_heap/
total 0
crw-rw---- 1 root video 253, 1 May 28 21:00 linux,cma
crw-rw---- 1 root video 253, 0 May 28 21:00 system
lrwxrwxrwx 1 root root       6 May 28 21:03 vidbuf_cached -> system
```

## Solution
```bash
user@hostname:~ $ sudo ln -s system /dev/dma_heap/vidbuf_cached
ln: failed to create symbolic link '/dev/dma_heap/vidbuf_cached': File exists
user@hostname:~ $ sudo ln -sf system /dev/dma_heap/vidbuf_cached
user@hostname:~ $ ls -l /dev/dma_heap/
total 0
crw-rw---- 1 root video 253, 1 May 28 21:00 linux,cma
crw-rw---- 1 root video 253, 0 May 28 21:00 system
lrwxrwxrwx 1 root root       6 May 28 21:03 vidbuf_cached -> system
```

# Request

Can this be fixed at a system / OS level?  Rather than a user having to find this out for themselves?

wallarug, 2025-05-28:
> Would you be able to try the regular OS (i.e. not Lite) and see if the symlink is missing?

I could give it a go in about an hour.  Just finishing off an important work thing (which needed the camera).  
I'll check the full OS when I'm done.

wallarug, 2025-05-28:
Further details - this happens when using specific overlays.  The folder is present on the DMA `vidbuf_cached` folder disappears when I change or update the camera overlay.

The `vidbuf_cached` is present with no cameras found / fresh install.

So this might actually be an @arducam issue rather than a rpi-apps issue.   I'll continue to have a look and let you know.

naushir, 2025-05-29:
@XECDesign for visibility.

naushir, 2025-06-03:
@wallarug any updates on this issue?  If not, I'll close it shortly.

wallarug, 2025-06-03:
Hey @naushir,  If you give me another 48 hours - I'll look into it and report back here.

wallarug, 2025-06-03:
> Looking at the [udev rule](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/lib/udev/rules.d/60-dma-heap.rules) responsible for creating the symlink, I'm not seeing how an overlay could break it. I'd expect a symlink to linux,cma at least.
> 
> I see that the kernel isn't something you'd find on a fresh install of anything we ship, and there's mention of overlays and arducam, so "fresh install" doesn't sound like an accurate and complete picture to figure out what's going wrong.

Hey - I did run the rpi-updater to grab the latest firmware.   I don't think that caused the symlink to disappear.  I'll get back to you all in the next 48 hours.

wallarug, 2025-06-05:
Apologies for the delays getting back to you all.  Work has been crazy.  Here is my breakdown of what I've discovered.  

# Testing Results
- OS: Raspbian 64-Bit Lite
- uname: `Linux cscs-0010 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux`
- Device: Raspberry Pi 5, 8GB

## After Install (clean)
```bash
0:~ $ ls -l /dev/dma_heap/
total 0
crw-rw---- 1 root video 253, 1 May 13 10:14 linux,cma
crw-rw---- 1 root video 253, 0 May 13 10:14 system
lrwxrwxrwx 1 root root       6 May 13 10:14 vidbuf_cached -> system
```

## Ran Updates and other commands
Nothing I installed triggered the symlink to disappear

## Rebooted after changing config.txt (no cameras attached)
```bash
:~ $ ls -l /dev/dma_heap/
total 0
crw------- 1 root root 253, 1 Jun  5 15:16 linux,cma
crw------- 1 root root 253, 0 Jun  5 15:16 system
```

DMA Heap is gone.  So it appears to be when updating the /boot/firmware/config.txt with the overlays that the dma_heap is being removed.  Regardless of if the physical hardware is present or not.

## Config file dump
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtoverlay=ov64a40,cam0,link-frequency=360000000
dtoverlay=ov64a40,cam1,link-frequency=360000000
camera_auto_detect=0
```

## Test 2 - Changing back to without overlays
```bash
:~ $ ls -l /dev/dma_heap/
total 0
crw------- 1 root root 253, 1 Jun  5 15:20 linux,cma
crw------- 1 root root 253, 0 Jun  5 15:20 system
```

Still gone.

## Test 3 - Add overlays back, manually add symlink, remove overlays
1.  Put in the same config as above (with dtoverlays)
2.  Put symlink back
3. Reboot
4. Check - NO SYMLINK
5. (in theory) put symlink back
6. Remove overlays from config.txt
7. Reboot
8. Check - NO SYMLINK

**So it looks like swapping in and out `dtoverlay` is causing the symlink to be broken or removed.**

Let me know if you need more tests run.

naushir, 2025-06-05:
I'm unable to reproduce this with the exact same ``config.txt`` files on a fresh OS image with updates applied.

> Ran Updates and other commands

The only thing I ran was ``sudo apt update && sudo apt full-upgrade -y``.  Did you install any additional packages?

XECDesign, 2025-06-05:
Should be fixed now.

wallarug, 2025-06-06:
Thank you both.  

What's the best approach to check this?   Wipe both SD cards and go again?

XECDesign, 2025-06-06:
That would be the safest and cleanest approach, but you may be able to fix it by running `sudo apt update && sudo apt install raspberrypi-sys-mods`.

naushir, 2025-06-09:
Resolving now...

wallarug, 2025-07-09:
Hey,  Thanks for all the help with this. 

On a brand new system this issue no longer presents itself.

I struggled with an existing system, however, after following the the recommended resolution path and a reboot - we seem to be ok now.

Apologies for the delayed thank you - but you guys did a great job and extremely fast.   I very much appreciate it!  

Thanks so much 👍