raspberrypi/rpicam-apps #812: [BUG] Cannot hflip or vflip with the GS Camera

henrykotze, 2025-06-03:
**Describe the bug**
```
Using the command
rpicam-vid --hflip --vflip
```
does not flip the camera view when using the GS Camera. We have tested on the HQ and RPi3 camera, and all perform as expected.

We have compiled the rpicam-apps from source, since we are running Ubuntu 22.04, and on the following commit:

Here is the camera-bug-report when running the following command:

camera-bug-report

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.

The bug report tool uses the following syntax:

```
camera-bug-report -t <timeout in seconds> -o <output file> -c "<rpicam-app command>"
```

For example,

```
camera-bug-report -t 5 -o bug.txt -c "rpicam-still -t 1000 -o test.jpg"
```
will attempt to run rpicam-still and timeout after 5 seconds if the application has stalled. The script will generate a ``bug.txt`` file that captures all the output logs and system state to help us debug the issue.

You can also run without a timeout:

```
camera-bug-report -o bug.txt -c "rpicam-vid -t 0 -o test.264"
```
This will run ``rpicam-vid`` indefinitely until either you press ``Ctrl+C`` or the application terminates, after which the necessary output logs and system state will be captured.

If you cannot run your application through the ``camera-bug-report`` tool, run it without the ``-c`` command line argument **after running the camera application.**  In these cases, please also provide the command line used to run the application, as well as any output generated during the run.

**Additional context**
Add any other context about the problem here.

6by9, 2025-06-03:
@naushir Global Shutter is imx296, not imx708 (v3).

Flips have been added by us over the mainline driver (ought to upstream those patches).
12b310f8b15c for rpi-6.12.y, but it's present in rpi-6.6.y as well.

henrykotze, 2025-06-03:
Thanks for these comments. Much appreciated.

This is outside my area of expertise, but can you point me to documentation on how I can compile these changes and reregistered the sensor driver into ubuntu kernel?

henrykotze, 2025-06-09:
Running ROS2. 

Last when i attempted to compile ROS2 on Raspberry Pi OS, i either failed or, some programs didn't run, and using Ubuntu was more seamless. 

Also, confidence when moving from SITL -> HITL. I would test it in simulation on my pc(ubuntu) and then when its only successful there, I would start moving it over to hardware.

naushir, 2025-06-09:
Perhaps the quickest way is to copy the IMX296 driver from the Raspberry Pi kernel tree into the Ubuntu kernel tree and build/deploy with instructions similar to https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel?

naushir, 2025-06-23:
Resolving now...