raspberrypi/rpicam-apps #814: [BUILD ISSUE]  AV_CODEC_CONFIG_SAMPLE_FORMAT .. etc. not declared

ntutangyun, 2025-06-04:
Hi all,

I have just started with Raspberry Pi 5, installed Ubuntu 24.10 OS from Raspberry PI Imager. I have the Raspberry Pi Camera Module 3 connected to the PI. 

I'm trying to setup the libraries:

I followed [this](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera) tutorial to build the `raspberrypi/libcamera` and the building seems to be fine.

but when I tried to build the `raspberrypi/rpicam-apps`, I encounter the following issue:

Below is the configuration output:
```bash
~/Desktop/rpicam-apps$ meson setup --reconfigure build -Denable_libav=enabled -Denable_drm=enabled -Denable_egl=enabled -Denable_qt=enabled -Denable_opencv=disabled -Denable_tflite=disabled -Denable_hailo=disabled
The Meson build system
Version: 1.5.1
Source dir: /home/cranfield-rpi1/Desktop/rpicam-apps
Build dir: /home/cranfield-rpi1/Desktop/rpicam-apps/build
Build type: native build
Project name: rpicam-apps
Project version: 1.7.0
C compiler for the host machine: cc (gcc 14.2.0 "cc (Ubuntu 14.2.0-4ubuntu2) 14.2.0")
C linker for the host machine: cc ld.bfd 2.43.1
C++ compiler for the host machine: c++ (gcc 14.2.0 "c++ (Ubuntu 14.2.0-4ubuntu2) 14.2.0")
C++ linker for the host machine: c++ ld.bfd 2.43.1
Host machine cpu family: aarch64
Host machine cpu: aarch64
Dependency dl found: YES unknown (cached)
Dependency libcamera found: YES 0.5.0 (cached)
Dependency boost (modules: program_options) found: YES 1.83.0 (cached)
Dependency threads found: YES unknown (cached)
Dependency libavcodec found: YES 61.3.100 (cached)
Dependency libavdevice found: YES 61.1.100 (cached)
Dependency libavformat found: YES 61.1.100 (cached)
Dependency libavutil found: YES 59.8.100 (cached)
Dependency libswresample found: YES 5.1.100 (cached)
Dependency libexif found: YES 0.6.24 (cached)
Dependency libjpeg found: YES 2.1.5 (cached)
Dependency libtiff-4 found: YES 4.5.1 (cached)
Dependency libpng found: YES 1.6.44 (cached)
Dependency libdrm found: YES 2.4.122 (cached)
Dependency x11 found: YES 1.8.7 (cached)
Dependency epoxy found: YES 1.5.10 (cached)
Dependency qt5 (modules: Core, Widgets) found: YES 5.15.15 (cached)
Configuring postproc_lib.h using configuration
Dependency opencv4 skipped: feature enable_opencv disabled
Dependency tensorflow-lite skipped: feature enable_tflite disabled
Dependency HailoRT (modules: HailoRT::libhailort) skipped: feature enable_hailo disabled
Dependency hailo-tappas-core skipped: feature enable_hailo disabled
WARNING: Library target 'rpicam_app' has 'name_prefix' set. Compilers may not find it from its '-lrpicam_app' linker flag in the 'rpicam_app.pc' pkg-config file.
WARNING: Library target 'rpicam_app' has 'name_prefix' set. Compilers may not find it from its '-lrpicam_app' linker flag in the 'rpicam_app-uninstalled.pc' pkg-config file.
Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
Build targets in project: 8

rpicam-apps 1.7.0

  libcamera
    location             : /usr/local/lib/aarch64-linux-gnu
    version              : 0.5.0

  Build configuration
    libav encoder        : YES
    drm preview          : YES
    egl preview          : YES
    qt preview           : YES
    OpenCV postprocessing: NO
    TFLite postprocessing: NO
    Hailo postprocessing : NO
    IMX500 postprocessing: NO

  User defined options
    enable_drm           : enabled
    enable_egl           : enabled
    enable_hailo         : disabled
    enable_libav         : enabled
    enable_opencv        : disabled
    enable_qt            : enabled
    enable_tflite        : disabled

Found ninja-1.12.1 at /usr/bin/ninja
Cleaning... 0 files.

```

and below is the compilation output:

```bash
~/Desktop/rpicam-apps$ meson compile -C build
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/cranfield-rpi1/Desktop/rpicam-apps/build
ninja: Entering directory `/home/cranfield-rpi1/Desktop/rpicam-apps/build'
[4/15] Compiling C++ object rpicam_app.so.1.7.0.p/encoder_libav_encoder.cpp.o
FAILED: rpicam_app.so.1.7.0.p/encoder_libav_encoder.cpp.o 
c++ -Irpicam_app.so.1.7.0.p -I. -I.. -I/usr/local/include/libcamera -I/usr/include -I/usr/include/aarch64-linux-gnu -I/usr/include/webp -I/usr/include/libpng16 -I/usr/include/libdrm -I/usr/include/aarch64-linux-gnu/qt5/QtCore -I/usr/include/aarch64-linux-gnu/qt5 -I/usr/include/aarch64-linux-gnu/qt5/QtWidgets -I/usr/include/aarch64-linux-gnu/qt5/QtGui -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -fPIC -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -pthread -DBOOST_ALL_NO_LIB -MD -MQ rpicam_app.so.1.7.0.p/encoder_libav_encoder.cpp.o -MF rpicam_app.so.1.7.0.p/encoder_libav_encoder.cpp.o.d -o rpicam_app.so.1.7.0.p/encoder_libav_encoder.cpp.o -c ../encoder/libav_encoder.cpp
../encoder/libav_encoder.cpp: In member function ‘void LibAvEncoder::initAudioOutCodec(const VideoOptions*, const StreamInfo&)’:
../encoder/libav_encoder.cpp:326:67: error: ‘AV_CODEC_CONFIG_SAMPLE_FORMAT’ was not declared in this scope
  326 |         avcodec_get_supported_config(codec_ctx_[AudioOut], codec, AV_CODEC_CONFIG_SAMPLE_FORMAT, 0,
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../encoder/libav_encoder.cpp:326:9: error: ‘avcodec_get_supported_config’ was not declared in this scope; did you mean ‘avcodec_get_hw_config’?
  326 |         avcodec_get_supported_config(codec_ctx_[AudioOut], codec, AV_CODEC_CONFIG_SAMPLE_FORMAT, 0,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         avcodec_get_hw_config
[5/15] Compiling C++ object post_processing_stages/core-postproc.so.p/acoustic_focus_stage.cpp.o
FAILED: post_processing_stages/core-postproc.so.p/acoustic_focus_stage.cpp.o 
c++ -Ipost_processing_stages/core-postproc.so.p -Ipost_processing_stages -I../post_processing_stages -I. -I.. -I/usr/local/include/libcamera -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -Wno-psabi -ftree-vectorize -fPIC -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -MD -MQ post_processing_stages/core-postproc.so.p/acoustic_focus_stage.cpp.o -MF post_processing_stages/core-postproc.so.p/acoustic_focus_stage.cpp.o.d -o post_processing_stages/core-postproc.so.p/acoustic_focus_stage.cpp.o -c ../post_processing_stages/acoustic_focus_stage.cpp
../post_processing_stages/acoustic_focus_stage.cpp: In lambda function:
../post_processing_stages/acoustic_focus_stage.cpp:81:73: error: ignoring return value of ‘int system(const char*)’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
   81 |                                 std::thread([](std::string cmd) { system(cmd.c_str()); }, cmd).detach();
      |                                                                   ~~~~~~^~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
[7/15] Compiling C++ object apps/rpicam-raw.p/rpicam_raw.cpp.o
ninja: build stopped: subcommand failed.

```

Please help. thanks. 

Is it because I'm using an unsupported version of libavcodec-dev?

below is the current version:

```bash
~/Desktop/rpicam-apps$ sudo apt install -y libavcodec-dev
libavcodec-dev is already the newest version (7:7.0.2-3ubuntu1.1).
The following package was automatically installed and is no longer required:
  python3-netifaces
Use 'sudo apt autoremove' to remove it.

Summary:
  Upgrading: 0, Installing: 0, Removing: 0, Not Upgrading: 0
```

and I don't have ffmpeg installed in the Ubuntu system yet.

naushir, 2025-06-05:
If you don't have ffmpeg installed, you can disable the ``libav`` integration by using the ``enable_libav=disabled`` meson configuration option.

naushir, 2025-06-05:
For the second error, can you apply the following patch and report if it works?

```patch
diff --git a/post_processing_stages/acoustic_focus_stage.cpp b/post_processing_stages/acoustic_focus_stage.cpp
index a7e89ff86978..2943d78e7fb2 100644
--- a/post_processing_stages/acoustic_focus_stage.cpp
+++ b/post_processing_stages/acoustic_focus_stage.cpp
@@ -78,7 +78,7 @@ public:
                                std::string duration_str = oss.str();

                                std::string cmd = "/usr/bin/play -nq -t alsa synth " + duration_str + " sine " + std::to_string(freq);
-                               std::thread([](std::string cmd) { system(cmd.c_str()); }, cmd).detach();
+                               std::thread([](std::string cmd) { [[maybe_unused]] int i = system(cmd.c_str()); }, cmd).detach();
                        }
                }
                return false;
```

naushir, 2025-06-18:
Fix for the unused variable has now been merged.

closed 2025-06-18 by naushir