raspberrypi/rpicam-apps #815: Gentoo: rpicam-hello --list-cameras No cameras available!

costel78, 2025-06-08:
Pretty sure that is not a bug, hopefully, but don't know where to ask for help. 
Here are the previous attempts to get it solved:
[Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=388252)
[Gentoo Forums](https://forums.gentoo.org/viewtopic-t-1174376.html)
with no success, or, at least, a point in the right direction.

Some sections from camera-bug-report are completed by hand, as on my Gentoo, there is no sudo and no /boot/firmware.
On RaspberryPi OS, the camera is working fine. Same case, cable and camera - just dual boot by swapping the /boot (/boot/firmware) content. This excludes the hardware issues, I believe.

Thank you!

[bug-report.txt](https://github.com/user-attachments/files/20644692/bug-report.txt)

naushir, 2025-06-09:
This feels like a groups/permission issue.  Is your username part of the video group?  Maybe try as a superuser and see if that helps?

costel78, 2025-06-09:
LIBCAMERA_LOG_LEVELS=*:0 rpicam-hello -t 0
[0:01:08.710047773] [1480] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib64/libcamera/ipa_rpi_vc4.so is signed
[0:01:08.710105477] [1480] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib64/libcamera/ipa_rpi_vc4.so'
[0:01:08.710646903] [1480] DEBUG IPAModule ipa_module.cpp:333 ipa_soft_simple.so: IPA module /usr/lib64/libcamera/ipa_soft_simple.so is signed
[0:01:08.710679532] [1480] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib64/libcamera/ipa_soft_simple.so'
[0:01:08.710693199] [1480]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0
[0:01:08.710810180] [1485] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:01:08.719687847] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[0:01:08.719914236] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[0:01:08.721247403] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:01:08.721274532] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[0:01:08.722969329] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "em28xx" created from /dev/media4
[0:01:08.722992588] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media4: em28xx
[0:01:08.723140254] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpi-hevc-dec" created from /dev/media1
[0:01:08.723149329] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 1 missing dependencies
[0:01:08.723314958] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:01:08.723326940] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rpi-hevc-dec
[0:01:08.723456440] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[0:01:08.723478254] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 8 missing dependencies
[0:01:08.723602106] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media3
[0:01:08.723619958] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 8 missing dependencies
[0:01:08.724643403] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[0:01:08.724663143] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[0:01:08.725669236] [1485] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[0:01:08.725689773] [1485] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: pispbe
[0:01:08.726028217] [1485] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
[0:01:08.726444791] [1485] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:01:08.726454958] [1485] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:01:08.726461866] [1485] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'simple'
[0:01:08.726479662] [1485] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'ipu3'
[0:01:08.726692495] [1485] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'uvcvideo'
Preview window unavailable
ERROR: *** no cameras available ***

costel78, 2025-06-09:
modprobe bcm2835-unicam, Unable to acquire a Unicam instance is still there.
rmmod bcm2835-unicam ; modprobe bcm2835-unicam-legacy
[0:08:01.491681467] [1773] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:08:01.491692838] [1773] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
still present.

What are those dependencies: Defer media device /dev/media2 due to 8 missing dependencies ?

naushir, 2025-06-09:
Are you running on a Pi 5? It seems like your libcamera build doesn't have Pi 5 support added.

costel78, 2025-06-09:
Yes, is a Pi5 8GB RAM. 
src_configure() {
        local emesonargs=(
                -Ddocumentation=disabled
                $(meson_feature libevent cam)
                $(meson_feature gstreamer)
                $(meson_feature qt6 qcam)
                $(meson_feature trace tracing)
                $(meson_feature v4l2)
                -Dipas=rpi/vc4,simple
                -Dpipelines=uvcvideo,ipu3,simple,rpi/vc4
        )
If I'am correct, support for pispbe is missing from libcamera. What else is missing ?

naushir, 2025-06-09:
```
-Dipas=rpi/vc4,rpi/pisp,simple
-Dpipelines=uvcvideo,ipu3,simple,rpi/vc4,rpi/pisp
```
ought to do the trick.

costel78, 2025-06-09:
It is working now.
libcamera-hello --buffer-count 1 --viewfinder-buffer-count 3
[0:45:24.818721134] [19738]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0
[0:45:24.837329299] [19741]  INFO RPI pisp.cpp:720 libpisp version 000000000000-invalid 09-06-2025 (14:16:47)
[0:45:24.948671032] [19741]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Preview window unavailable
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:45:24.949887438] [19738]  INFO Camera camera.cpp:1205 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR_PISP_COMP1
[0:45:24.950115530] [19741]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
#0 (0.00 fps) exp 32680.00 ag 8.00 dg 1.00
#1 (30.01 fps) exp 32680.00 ag 8.00 dg 1.00
#2 (30.01 fps) exp 32680.00 ag 8.00 dg 1.00
#3 (30.01 fps) exp 32680.00 ag 8.00 dg 1.00
....

Do you know how to circumvent buffer allocation error and the need of  --buffer-count 1 --viewfinder-buffer-count 3 arguments ?

naushir, 2025-06-09:
> Do you know how to circumvent buffer allocation error and the need of --buffer-count 1 --viewfinder-buffer-count 3 arguments ?

I'm not sure what you are referring to here?  Do you seen any error message without those parameters?

costel78, 2025-06-09:
Oh, I forgot to post the errors:
libcamera-hello
[1:22:08.845447978] [31897]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0
[1:22:08.860136587] [31900]  INFO RPI pisp.cpp:720 libpisp version 000000000000-invalid 09-06-2025 (14:16:47)
[1:22:08.966083375] [31900]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[1:22:09.066426263] [31897]  INFO Camera camera.cpp:1205 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR_PISP_COMP1
[1:22:09.066887223] [31900]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
dmaHeap allocation failure for rpicam-apps2
ERROR: *** failed to allocate capture buffers for stream ***

LIBCAMERA_LOG_LEVELS=*:0 rpicam-hello -t 0
[1:23:28.311258770] [31940] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib64/libcamera/ipa_rpi_pisp.so is signed
[1:23:28.311334714] [31940] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib64/libcamera/ipa_rpi_pisp.so'
[1:23:28.311417361] [31940] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib64/libcamera/ipa_rpi_vc4.so is signed
[1:23:28.311436639] [31940] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib64/libcamera/ipa_rpi_vc4.so'
[1:23:28.311473806] [31940] DEBUG IPAModule ipa_module.cpp:333 ipa_soft_simple.so: IPA module /usr/lib64/libcamera/ipa_soft_simple.so is signed
[1:23:28.311490842] [31940] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib64/libcamera/ipa_soft_simple.so'
[1:23:28.311498546] [31940]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0
[1:23:28.311650064] [31943] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[1:23:28.320203255] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[1:23:28.320406125] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[1:23:28.321622747] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[1:23:28.321640673] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[1:23:28.323176849] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "em28xx" created from /dev/media4
[1:23:28.323203757] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media4: em28xx
[1:23:28.323310330] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpi-hevc-dec" created from /dev/media1
[1:23:28.323325737] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 1 missing dependencies
[1:23:28.323468237] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[1:23:28.323481959] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rpi-hevc-dec
[1:23:28.323577255] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[1:23:28.323592273] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 8 missing dependencies
[1:23:28.323681828] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media3
[1:23:28.323696587] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 8 missing dependencies
[1:23:28.324664267] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[1:23:28.324680619] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[1:23:28.325851964] [31943] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[1:23:28.325867927] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: pispbe
[1:23:28.326249035] [31943] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
[1:23:28.326266202] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[1:23:28.326284591] [31943] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
[1:23:28.326373738] [31943] DEBUG ClockRec clock_recovery.cpp:69 configure 100 2000 10 50000
[1:23:28.326382109] [31943] DEBUG ClockRec clock_recovery.cpp:83 reset
[1:23:28.326448516] [31943]  INFO RPI pisp.cpp:720 libpisp version 000000000000-invalid 09-06-2025 (14:16:47)
[1:23:28.327126993] [31943] DEBUG CameraSensor camera_sensor_raw.cpp:210 imx708: unsupported number of sinks (0) or sources (2)
[1:23:28.327183919] [31943] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Exposure (0x00980911)
[1:23:28.327204382] [31943] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Horizontal Flip (0x00980914)
[1:23:28.327214234] [31943] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Vertical Flip (0x00980915)
[... cut ...]
[1:23:28.574819960] [31943] DEBUG Timer timer.cpp:93 Starting timer 0x7fffa882cc00: deadline 1:23:30.574811516
[1:23:28.574913015] [31943] DEBUG Event event_dispatcher_poll.cpp:215 next timer 0x7fffa882cc00 expires in 1.999899130
[1:23:28.575191810] [31943] DEBUG Timer timer.cpp:93 Starting timer 0x7fffa882d3d0: deadline 1:23:30.575191014
[1:23:28.575250476] [31943] DEBUG Event event_dispatcher_poll.cpp:215 next timer 0x7fffa882d3d0 expires in 1.999941186
ERROR: *** failed to allocate capture buffers for stream ***