raspberrypi/rpicam-apps #841: [BUG]

qq765490023, 2025-07-23:
In main branch:
fox@fox:~/Documents/projects/rpicam-apps $ git log
commit 3b01de4a987e8506f44450c0c5402a05188dd64b (HEAD -> main, origin/main, origin/HEAD)
Author: Naushir Patuck <naush@raspberrypi.com>
Date:   Tue Jul 22 12:08:31 2025 +0100

    core: Fix posible static initialiser order fiasco problems
    
    Remove direct access to maps with appropriate getters.
    
    Signed-off-by: Naushir Patuck <naush@raspberrypi.com>

I successfully built this project, but when I try to use: 

 ./rpicam-hello -t 0 --camera 0

It just won't show the preview window, while using
 rpicam-hello -t 0 --camera 0
which I installed with APT is working properly, what coulde be the cause that the window not showing up. 

log info is as below when I run :  ./rpicam-hello -t 0 --camera 0

[9:16:14.827420753] [61169]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
#7 (0.00 fps) exp 33251.00 ag 9.85 dg 1.02
#8 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#9 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#10 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#11 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#12 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#13 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#14 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#15 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#16 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#17 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#18 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#19 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02

qq765490023, 2025-07-23:
fox@fox:~/Documents/projects/rpicam-apps $ rpicam-hello --version
rpicam-apps build: v1.8.1 eb293b055248 21-07-2025 (11:50:44)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.5.1+100-e53bdf1f
fox@fox:~/Documents/projects/rpicam-apps $ 
-------------------------------------------------------------
this one works fine!

fox@fox:~/Documents/projects/rpicam-apps $ ./build/apps/rpicam-hello --version
rpicam-apps build: v1.8.1 3b01de4a987e 23-07-2025 (18:18:54)
rpicam-apps capabilites: egl:0 qt:0 drm:0 libav:0
libcamera build: v0.5.1+100-e53bdf1f
fox@fox:~/Documents/projects/rpicam-apps $ 
------------------------------------------------
This one no preview window.

qq765490023, 2025-07-23:
fox@fox:/Documents/projects/rpicam-apps $ rpicam-hello --version
rpicam-apps build: v1.8.1 https://github.com/raspberrypi/rpicam-apps/commit/eb293b0552484c35dbddf7788badf42c709a4783 21-07-2025 (11:50:44)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.5.1+100-e53bdf1f
fox@fox:/Documents/projects/rpicam-apps $
this one works fine!

fox@fox:/Documents/projects/rpicam-apps $ ./build/apps/rpicam-hello --version
rpicam-apps build: v1.8.1 https://github.com/raspberrypi/rpicam-apps/commit/3b01de4a987e8506f44450c0c5402a05188dd64b 23-07-2025 (18:18:54)
rpicam-apps capabilites: egl:0 qt:0 drm:0 libav:0
libcamera build: v0.5.1+100-e53bdf1f
fox@fox:/Documents/projects/rpicam-apps $
This one no preview window.

naushir, 2025-07-23:
> rpicam-apps capabilites: egl:0 qt:0 drm:0 libav:0

You seem to have not built the preview plugins for rpicam-apps.  What does ``meson setup build --wipe`` show?

qq765490023, 2025-07-23:
fox@fox:~/Documents/projects/rpicam-apps $ meson setup build --wipe
The Meson build system
Version: 1.5.1
Source dir: /home/fox/Documents/projects/rpicam-apps
Build dir: /home/fox/Documents/projects/rpicam-apps/build
Build type: native build
Project name: rpicam-apps
Project version: 1.8.1
C compiler for the host machine: cc (gcc 12.2.0 "cc (Debian 12.2.0-14+deb12u1) 12.2.0")
C linker for the host machine: cc ld.bfd 2.40
C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (Debian 12.2.0-14+deb12u1) 12.2.0")
C++ linker for the host machine: c++ ld.bfd 2.40
Host machine cpu family: aarch64
Host machine cpu: aarch64
Run-time dependency dl found: YES
Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
Run-time dependency libcamera found: YES 0.5.1
Run-time dependency Boost (found: program_options) found: YES 1.74.0 (/usr)
Run-time dependency threads found: YES
Run-time dependency libavcodec found: YES 59.37.100
Found CMake: /usr/bin/cmake (3.25.1)
Run-time dependency libavdevice found: NO (tried pkgconfig and cmake)
Run-time dependency libexif found: YES 0.6.24
Run-time dependency libjpeg found: YES 2.1.5
Run-time dependency libtiff-4 found: YES 4.5.0
Run-time dependency libpng found: YES 1.6.39
Dependency opencv4 skipped: feature enable_opencv disabled
Dependency tensorflow-lite skipped: feature enable_tflite disabled
Run-time dependency hailort found: NO (tried pkgconfig and cmake)
Run-time dependency hailo-tappas-core found: NO (tried pkgconfig and cmake)
Run-time dependency libdrm found: YES 2.4.123
Run-time dependency x11 found: YES 1.8.4
Run-time dependency epoxy found: NO (tried pkgconfig and cmake)
Run-time dependency qt5 (modules: Core, Widgets) found: YES 5.15.8 (pkg-config)
Configuring config.h using configuration
Build targets in project: 10

rpicam-apps 1.8.1

  libcamera
    location             : /usr/lib/aarch64-linux-gnu
    version              : 0.5.1

  Build configuration
    libav encoder        : NO
    drm preview          : YES
    egl preview          : NO
    qt preview           : YES
    OpenCV postprocessing: NO
    TFLite postprocessing: NO
    Hailo postprocessing : NO
    IMX500 postprocessing: NO

Found ninja-1.11.1 at /usr/bin/ninja

*********
Here is the result of:  meson setup build --wipe

fox@fox:~/Documents/projects/rpicam-apps $ ./build/apps/rpicam-hello -t 0 --preview-libs ./build/preview
[9:43:10.676233853] [62322]  INFO Camera camera_manager.cpp:326 libcamera v0.5.1+100-e53bdf1f
[9:43:10.695035870] [62325]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[9:43:10.711267912] [62325]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_D0
[9:43:10.711399765] [62325]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[9:43:10.726458412] [62325]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media3 and ISP device /dev/media2 using PiSP variant BCM2712_D0
Made DRM preview window
Preview window unavailable
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
[... 77 lines cut ...]
#76 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#77 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#78 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#79 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#80 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#81 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#82 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#83 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#84 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#85 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#86 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#87 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#88 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#89 (30.00 fps) exp 33251.00 ag 9.85 dg 1.02
#90 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#91 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#92 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#93 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
^C
fox@fox:~/Documents/projects/rpicam-apps $ ls
apps    core     license.txt        output                  README.md
assets  encoder  meson.build        post_processing_stages  utils
build   image    meson_options.txt  preview
***
here is the result of execution.

naushir, 2025-07-23:
If you are running without installing (as is the case here), you must now use:

```
./build/apps/rpicam-hello -t 0 --preview-libs ./build/preview
```

qq765490023, 2025-07-23:
ah I see，but still 
~~~~
Preview window unavailable
~~~
after using this:
~~~
./build/apps/rpicam-hello -t 0 --preview-libs ./build/preview
~~~

naushir, 2025-07-23:
Can you share the output of the following please:

```
ls -la build/preview
```

```
./build/apps/rpicam-hello --version --preview-libs ./build/preview
```

And finally, can you try
```
./build/apps/rpicam-hello -t 0 --preview-libs ./build/preview --qt-preview
```
please?

qq765490023, 2025-07-23:
ya . solve thank you so much. 
BTW, How can I use drm-preview. 
I tried 
~~~
./build/apps/rpicam-hello -t 0 --preview-libs ./build/preview --drm-preview
~~ ~
which is not fine.

~~~
fox@fox:~/Documents/projects/rpicam-apps $ ls -la build/preview
total 120
drwxr-xr-x  4 fox fox  4096 Jul 23 18:55 .
drwxr-xr-x 15 fox fox  4096 Jul 23 18:55 ..
-rwxr-xr-x  1 fox fox 78432 Jul 23 18:55 drm-preview.so
drwxr-xr-x  2 fox fox  4096 Jul 23 18:55 drm-preview.so.p
-rwxr-xr-x  1 fox fox 84224 Jul 23 18:55 qt-preview.so
drwxr-xr-x  2 fox fox  4096 Jul 23 18:55 qt-preview.so.p
fox@fox:~/Documents/projects/rpicam-apps $ ./build/apps/rpicam-hello --version --preview-libs ./build/preview
rpicam-apps build: v1.8.1 eb293b055248 23-07-2025 (18:55:07)
rpicam-apps capabilites: egl:0 qt:1 drm:1 libav:0
libcamera build: v0.5.1+100-e53bdf1f
fox@fox:~/Documents/projects/rpicam-apps $ ./build/apps/rpicam-hello -t 0 --preview-libs ./build/preview --qt-preview
[9:55:21.723110556] [62489]  INFO Camera camera_manager.cpp:326 libcamera v0.5.1+100-e53bdf1f
[9:55:21.739199932] [62492]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[9:55:21.756423066] [62492]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_D0
[9:55:21.756574195] [62492]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[9:55:21.772287498] [62492]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media3 and ISP device /dev/media2 using PiSP variant BCM2712_D0
Made QT preview window
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[9:55:21.900544769] [62489]  INFO Camera camera.cpp:1205 configuring streams: (0) 1640x1232-YUV420/sYCC (1) 1640x1232-BGGR_PISP_COMP1/RAW
[9:55:21.900671768] [62492]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
^C
fox@fox:~/Documents/projects/rpicam-apps $ 

~~~

qq765490023, 2025-07-23:
Oh, got it. Thank you so much again.

naushir, 2025-07-23:
No problem!  Resolving this now.