raspberrypi/rpicam-apps #845: IMX462 capture takes twice as long as exposure time

harmveer, 2025-07-24:
Using an IMX462 camera sensor (Innomaker CAM-MIPI462RAW-Color) connected to a Raspberry Pi Zero 2 W I'm trying to make long exposure (> 30 s) shots sequentially (~ 1 hour) with high analogue gain (also using HCG mode according to this [thread](https://github.com/raspberrypi/linux/pull/5859)) to detect low intensity light. However, I notice that using e.g. 90 second exposure setting takes 180 sec (i.e. exactly 2 x 90 s) to return an image. When I previously tested the same approach with the Raspberry Pi camera model 3 (wide, IMX707), this wasn't happening (i.e. capturing took basically as long as the set exposure time). To me this issue looks similar to what was described in this [thread](https://github.com/raspberrypi/rpicam-apps/issues/524) where the issue turned out to be in dropping 1 frame for each mode switch with the v3 Pi camera. 

I'm using this picamera2 script:

`import time
from picamera2 import Picamera2
from datetime import datetime, timedelta
from pprint import pprint

#PiCamera setup
picam2 = Picamera2() #instantiates a picamera
modes = picam2.sensor_modes
mode = modes[2] # Use 1280x720-SRGGB12_CSI2P mode
print('mode selected: ', mode)
camera_config = picam2.create_still_configuration(raw={'format': mode['unpacked']}, sensor={'output_size': mode['size'], 'bit_depth': mode['bit_depth']}, buffer_count=2)
picam2.configure(camera_config)
picam2.set_controls({"ExposureTime": 90000000, "ColourGains": (1,1), "AnalogueGain": 64}) # Set HCG mode separately to ON ('Y') through hcg_mode parameter ('echo 1 | sudo tee /sys/module/imx290/parameters/hcg_mode'; verify with 'cat /sys/module/imx290/parameters/hcg_mode') 
# Checking raw configuration
check = picam2.camera_configuration()['raw']
pprint(check)
picam2.start() 

runtime = 60    
endTime = datetime.now() + timedelta(minutes=runtime)

time.sleep(2) # Wait for camera to settle

while datetime.now() <= endTime:
    r = picam2.capture_request()
    r.save("main", datetime.now().strftime("%Y-%m-%d_%H%M%S.jpg"))
    r.save_dng(datetime.now().strftime("%Y-%m-%d_%H%M%S.dng"))
    metadata = r.get_metadata()
    print(metadata)
    r.release()
    
picam2.stop()`

The [imx290 helper](https://git.linuxtv.org/libcamera.git/tree/src/ipa/rpi/cam_helper/cam_helper_imx290.cpp#n59) (which is used for the IMX462 if I understand correctly) also includes a 'hideFramesModeSwitch' function, but as far as I understand there should be no mode-switching happening in the above capturing loop, or am I mistaken? Any thoughts on how I could reduce the lag between capturing pictures? Any fixes/workarounds are very welcome!

Some other threads that I found that seem related: 
- [IMX462 long exposure bug](https://github.com/raspberrypi/rpicam-apps/issues/559)
- [framedrop IMX283](https://github.com/will127534/OneInchEye/issues/37)
- [Capturing long exposure still one after each other without delay](https://github.com/raspberrypi/picamera2/discussions/343)

Some background of my use case, for those interested:  we're using the camera to capture the bioluminescent light signal of different colors coming from our molecular diagnostic test (called LUNAS, for those interested), with different colors indicating presence/absence of the pathogen(s) tested for. I'll be using this set-up within a portable test reader device that I will bring to Uganda next month for a validation study in a rural hospital. Hence, it would really be great to optimize the image capturing, and I would be very grateful if someone of you great developers could help me out soon here, hopefully with a quick fix! (:

davidplowman, 2025-07-25:
So I think you could pretty much do something like this using Picamera2 already.

First, after you configure but before you start the camera, don't set your custom exposure/gain control values. Instead, set the `'FrameRate'` control to something really fast, like 30fps  (which will presumably max out at whatever the camera can actually deliver).

Next, take your list of controls and set them _immediately after_ `picam2.start()`.

Now you'll have to put in a little while loop which will capture requests and check the metadata. Use the first one where the exposure time has shot up to "close" (it won't be exact) to what you wanted.

davidplowman, 2025-07-25:
The behaviour of the pipeline is that you can ask for higher gains than the analogue gain can achieve. Anything above the the maximum analogue gain is applied by the ISP as digital gain. So the final processed output image will reflect that larger gain, but it has no effect on the raw buffer or DNG file which will be limited to exactly what came out of the sensor.

harmveer, 2025-07-26:
Thank you both for the clarification on the gain setting, good to see confirmed that I'm making use of all the analogue gain the sensor has to offer by setting it to `29.51`. 

I tried @davidplowman's suggestion using below script, and it works! I now get pictures separated only by the requested exposure time, perfect! It's a little suprising to me this actually works, could you explain what makes the difference with the original script?

```
import time
from picamera2 import Picamera2
from datetime import datetime, timedelta
from pprint import pprint

#PiCamera setup
picam2 = Picamera2() #instantiates a picamera
modes = picam2.sensor_modes
mode = modes[2] # Use 1280x720-SRGGB12_CSI2P mode
print('mode selected: ', mode)
camera_config = picam2.create_still_configuration(raw={'format': mode['unpacked']}, sensor={'output_size': mode['size'], 'bit_depth': mode['bit_depth']}, buffer_count=2)
picam2.configure(camera_config)
picam2.set_controls({"FrameRate": 30}) # Set high framrate (i.e. minimal exposure time) initially
# Checking raw configuration
check = picam2.camera_configuration()['raw']
pprint(check)
picam2.start()
ExposureTime = 90000000
picam2.set_controls({"FrameDurationLimits": (0,0), "ExposureTime": ExposureTime, "ColourGains": (1,1), "AnalogueGain": 64}) # Directly after starting camera set long exposure. Set HCG mode separately to ON ('Y') through hcg_mode parameter ('echo 1 | sudo tee /sys/module/imx290/parameters/hcg_mode'; verify with 'cat /sys/module/imx290/parameters/hcg_mode')  

runtime = 60    
endTime = datetime.now() + timedelta(minutes=runtime)

time.sleep(2) # Wait for camera to settle
#ActualExposureTime = 0
#while ActualExposureTime < 0.9*ExposureTime:
#    r = picam2.capture_request()
#    metadata = r.get_metadata()
#    print(metadata)
#    ActualExposureTime = metadata["ExposureTime"]
#   r.release()

while datetime.now() <= endTime:
    r = picam2.capture_request()
    r.save("main", datetime.now().strftime("%Y-%m-%d_%H%M%S.jpg"))
    r.save_dng(datetime.now().strftime("%Y-%m-%d_%H%M%S.dng"))
    metadata = r.get_metadata()
    print(metadata)
    r.release()
    
picam2.stop()

```

Sidenote:
I also tried  `rpicam-still -r -t 0 -n --shutter 20000000 --gain 30 --awbgains 1,1 --timelapse 1 --datetime --metadata -`, which does also nicely return pictures spaced 20s apart (same works for 90s exposure time). Based on [this thread](https://github.com/raspberrypi/picamera2/discussions/343) I would actually expect things to be the other way around, since the rpicamera2 based script should not involve mode switching while the rpicam-still method does.

davidplowman, 2025-07-28:
Well, when you set the exposure/framerate _before_ starting the camera then you're guaranteed to get that exposure time immediately. So any frames that the camera system drops, and this includes ones that may simply be corrupt (cameras quite often produce one or two dud frames after powering on) or ones where we are waiting for AGC/AWB to settle, will all happen at that (hopefully fast) framerate.

After starting, you can change to the long exposure and after "a few more frames of delay" (but these are still fast frames), it will change to the long exposure and all should be well.

The original script, IIRC, was setting the slow framerate before starting, so all those corrupt/unwanted frames will be been going by _very_ slowly...

Anyyway, glad to here things are working better for you now!

harmveer, 2025-07-28:
I understand this prevents the slow going by of corrupt frames at the start, but don't really see how this script fixes the delay in between the 2nd and 3rd etc. good frames... Happy it does somehow, but would like to understand why.

davidplowman, 2025-08-04:
Hi again. Not 100% sure what's happening there. There's a couple of things I might try.

1. You could push the buffer count up to 3, just in case the code is short of a buffer and therefore drops a frame.

2. Better still, I wouldn't sleep for 2 seconds. Simply capture every request, check the metadata, and save the first one that has the exposure you asked for. I would regard this as the "proper" way to do capture your frame.

Does either/both of those help at all?

harmveer, 2025-08-17:
Thanks again for your quick response @davidplowman, sorry for my very late reply. Your suggestion works like a charm, I implemented both, but the trick appears to be in nr. 2 (not sleep the camera first). Again I do not understand why this makes the camera not drop the first long exposure frame, but very happy it works!