raspberrypi/rpicam-apps #849: [BUG] `__vb2_queue_cancel` error in kernel logs

goombado, 2025-07-30:
**Describe the bug**
In the latest version of rpicam-apps, we have been getting issues where running rpicam-still repeatedly over time inevitably and consistently causes a kernel crash – the whole Raspberry Pi 5 running the code freezes and becomes completely unresponsive (no networking etc.).

**Bug report**
We call the command
```sh
/usr/bin/rpicam-still --width 4624 --height 3472 --camera 0 -o /dev/shm/tmp.jpg -n -t 3000 --no-raw --encoding jpg --quality 100 --rotation 0
```
once every 60 seconds. After ~10 hours (time varies with each iteration), we observe the following kernel logs. The errors occur 5 seconds after we request a photo to be taken on each of the two cameras plugged into the rpi.

**Error Log 1: `__vb2_queue_cancel`**
```
Jul 30 10:39:15 camera-testing kernel: rp1-cfe 1f00110000.csi: Using a link rate of 720 Mbps
Jul 30 10:39:15 camera-testing kernel: rp1-cfe 1f00128000.csi: Using a link rate of 720 Mbps
Jul 30 10:39:20 camera-testing kernel: Unable to handle kernel paging request at virtual address dead000000000140
Jul 30 10:39:20 camera-testing kernel: Mem abort info:
Jul 30 10:39:20 camera-testing kernel:   ESR = 0x0000000096000004
Jul 30 10:39:20 camera-testing kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Jul 30 10:39:20 camera-testing kernel:   SET = 0, FnV = 0
Jul 30 10:39:20 camera-testing kernel:   EA = 0, S1PTW = 0
Jul 30 10:39:20 camera-testing kernel:   FSC = 0x04: level 0 translation fault
Jul 30 10:39:20 camera-testing kernel: Data abort info:
Jul 30 10:39:20 camera-testing kernel:   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
Jul 30 10:39:20 camera-testing kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Jul 30 10:39:20 camera-testing kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Jul 30 10:39:20 camera-testing kernel: [dead000000000140] address between user and kernel address ranges
Jul 30 10:39:20 camera-testing kernel: Internal error: Oops: 0000000096000004 [#1] PREEMPT SMP
Jul 30 10:39:20 camera-testing kernel: Modules linked in: wireguard libchacha20poly1305 chacha_neon poly1305_neon ip6_udp_tunnel udp_tunnel libcurve25519_generic libchacha 8021q garp stp llc binfmt_misc brcmfmac_wcc vc4 snd_soc_hdmi_codec ov64a40 bu64754 drm_display_helper v4l2_cci regmap_i2c brcmfmac hci_uart cec brcmutil drm_dma_helper btbcm bluetooth snd_soc_core cfg80211 aes_ce_blk aes_ce_cipher ghash_ce snd_compress gf128mul ecdh_generic rpi_hevc_dec snd_pcm_dmaengine pisp_be sha2_ce ecc rp1_cfe snd_pcm sha256_arm64 rfkill sha1_ce v4l2_mem2mem snd_timer libaes v4l2_fwnode v4l2_async videobuf2_dma_contig v3d snd videobuf2_memops sha1_generic w1_gpio videobuf2_v4l2 raspberrypi_hwmon gpu_sched wire videodev drm_shmem_helper cn drm_kms_helper videobuf2_common pwm_fan mc rp1_pio rp1_fw raspberrypi_gpiomem rp1_adc rp1_mailbox nvmem_rmem uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6 spidev spi_bcm2835 i2c_brcmstb gpio_keys i2c_designware_platform i2c_designware_core
Jul 30 10:39:20 camera-testing kernel: CPU: 3 UID: 1006 PID: 97523 Comm: libcamera-still Not tainted 6.12.34+rpt-rpi-2712 #1  Debian 1:6.12.34-1+rpt1~bookworm
Jul 30 10:39:20 camera-testing kernel: Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
Jul 30 10:39:20 camera-testing kernel: pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jul 30 10:39:20 camera-testing kernel: pc : __vb2_queue_cancel+0xe0/0x2d8 [videobuf2_common]
Jul 30 10:39:20 camera-testing kernel: lr : __vb2_queue_cancel+0xa8/0x2d8 [videobuf2_common]
Jul 30 10:39:20 camera-testing kernel: sp : ffffc0008265bb50
Jul 30 10:39:20 camera-testing kernel: x29: ffffc0008265bb50 x28: ffffd06fbd8a8938 x27: ffffd06fbd7b90f8
Jul 30 10:39:20 camera-testing kernel: x26: 0000000000000000 x25: ffff800007987668 x24: ffff800007601a40
Jul 30 10:39:20 camera-testing kernel: x23: 0000000000000000 x22: ffff800007987698 x21: 0000000000000000
Jul 30 10:39:20 camera-testing kernel: x20: 0000000000000009 x19: dead000000000100 x18: 0000000000000000
Jul 30 10:39:20 camera-testing kernel: x17: 0000000000000000 x16: ffffd06fce50a858 x15: 00007ffef8023ea0
Jul 30 10:39:20 camera-testing kernel: x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
Jul 30 10:39:20 camera-testing kernel: x11: 0000000000000000 x10: 0000000000000000 x9 : ffffd06fce50a8b8
Jul 30 10:39:20 camera-testing kernel: x8 : ffffc0008265bd48 x7 : 0000000000000005 x6 : 0000000000000000
Jul 30 10:39:20 camera-testing kernel: x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffff80009f906d80
Jul 30 10:39:20 camera-testing kernel: x2 : 0000000000000020 x1 : ffff80000928f300 x0 : 00000000ffffffff
Jul 30 10:39:20 camera-testing kernel: Call trace:
Jul 30 10:39:20 camera-testing kernel:  __vb2_queue_cancel+0xe0/0x2d8 [videobuf2_common]
Jul 30 10:39:20 camera-testing kernel:  vb2_core_streamoff+0x2c/0xd0 [videobuf2_common]
Jul 30 10:39:20 camera-testing kernel:  vb2_ioctl_streamoff+0x54/0xa0 [videobuf2_v4l2]
Jul 30 10:39:20 camera-testing kernel:  v4l_streamoff+0x2c/0x40 [videodev]
Jul 30 10:39:20 camera-testing kernel:  __video_do_ioctl+0x170/0x3e0 [videodev]
Jul 30 10:39:20 camera-testing kernel:  video_usercopy+0x1c0/0x790 [videodev]
Jul 30 10:39:20 camera-testing kernel:  video_ioctl2+0x20/0x38 [videodev]
Jul 30 10:39:20 camera-testing kernel:  v4l2_ioctl+0x48/0x70 [videodev]
Jul 30 10:39:20 camera-testing kernel:  __arm64_sys_ioctl+0xb4/0x100
Jul 30 10:39:20 camera-testing kernel:  invoke_syscall+0x50/0x120
Jul 30 10:39:20 camera-testing kernel:  el0_svc_common.constprop.0+0x48/0xf0
Jul 30 10:39:20 camera-testing kernel:  do_el0_svc+0x24/0x38
Jul 30 10:39:20 camera-testing kernel:  el0_svc+0x30/0x100
Jul 30 10:39:20 camera-testing kernel:  el0t_64_sync_handler+0x100/0x130
Jul 30 10:39:20 camera-testing kernel:  el0t_64_sync+0x190/0x198
Jul 30 10:39:20 camera-testing kernel: Code: 9ad52400 36000900 f8647833 b40008d3 (f9402274) 
Jul 30 10:39:20 camera-testing kernel: ---[ end trace 0000000000000000 ]---
```

At this point the system has locked up, as we observed by seeing all logging stop at this point for our custom app that invokes rpicam-still. The only additional logging we have is kernel logs from journalctl after this point.

**Error Log 2: `__kmalloc_cache_noprof`**
```
Jul 30 10:41:10 camera-testing kernel: Unable to handle kernel paging request at virtual address 00f7c00000f78480
Jul 30 10:41:10 camera-testing kernel: Mem abort info:
[... 132 lines cut ...]
Jul 30 10:43:20 camera-testing kernel:  w1_search+0x1e0/0x250 [wire]
Jul 30 10:43:20 camera-testing kernel:  w1_search_devices+0x54/0x60 [wire]
Jul 30 10:43:20 camera-testing kernel:  w1_search_process_cb+0x84/0x170 [wire]
Jul 30 10:43:20 camera-testing kernel:  w1_process+0x1b0/0x220 [wire]
Jul 30 10:43:20 camera-testing kernel:  kthread+0x11c/0x128
Jul 30 10:43:20 camera-testing kernel:  ret_from_fork+0x10/0x20
Jul 30 10:43:20 camera-testing kernel: Code: f9002bf9 aa1403e1 aa1303e0 b9402a62 (f8626a99) 
Jul 30 10:43:20 camera-testing kernel: ---[ end trace 0000000000000000 ]---
Jul 30 10:46:18 camera-testing kernel: Unable to handle kernel paging request at virtual address 00f7c00000f78480
```

The above log is the last line of logging we have in the system before it went completely unresponsive.

**Additional context**
At the time of Error Log 1 at 10:39:20, this is the last moment at which our app interacted with the system. Our app's logging ends at 10:39:20, and so we only have these kernel logs after the fact to go off of. Our app shows a Called Process Error with SIGABRT, and no more logging after this.

```
Called Process Error: Command '['/usr/bin/libcamera-still', '--width', '4624', '--height', '3472', '--camera', '0', '-o', '/dev/shm/tmp_0.jpg', '-n', '-t', '3000', '--no-raw', '--encoding', 'jpg', '--quality', '100', '--rotation', '0']' died with <Signals.SIGABRT: 6>.
```

- The rpi's uname is `6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux`.
- We increased the CMA buffer to 512MiB (this used to be the default, but for some reason has changed), but that did not help.
- We increased the page file to 8GB and that also did not help.

@wallarug and I have tried troubleshooting this problem over the past 2-3 weeks but to no avail. This problem has only appeared since we updated to the Linux 2025-06-26 rpi firmware. Any help with this issue would be immensely appreciated, we are happy to provide any further information and welcome any tips for investigation.

naushir, 2025-07-30:
Note that you should be using ``rpicam-still`` in place of ``libcamera-still``. What does the terminal output of ``rpicam-still`` show when the crash occurs?

Can you also share the output of ``rpicam-still --version`` please?

naushir, 2025-07-30:
May I suggest you update rpicam-apps and libcamera to the latest version please?  We have recently fixed a bug (more information at https://github.com/raspberrypi/rpicam-apps/issues/799)  that would cause random lockups, similar to what you might be seeing here.

wallarug, 2025-07-30:
> May I suggest you update rpicam-apps and libcamera to the latest version please? We have recently fixed a bug (more information at [#799](https://github.com/raspberrypi/rpicam-apps/issues/799)) that would cause random lockups, similar to what you might be seeing here.

I just updated one of our production cameras.   We will see how it goes on the new `rpiapps`.   I am hopeful after you pointed us in the direction of that Issue/PR chain that this was a similar or same issue.

We will report back in 24 - 48 hours if we are still seeing issues (it normally takes that long for them to appear).

**Current Version:**
```
rpicam-apps build: v1.8.1 eb293b055248 21-07-2025 (11:50:44)
rpicam-apps capabilites: egl:0 qt:0 drm:1 libav:0
libcamera build: v0.5.1+100-e53bdf1f
```

wallarug, 2025-08-03:
From what I can see in our logging for the software, it appears the camera `rpicam-still` _could have caused_ the lock up of the system.   It doesn't make sense to me though that `rpicam-still` under certain conditions causes this to happen.    More to the point - it seems to happen after ~3 days (consistently).   I was checking the camera periodically, there did not appear to be any 'zombie' or 'dead' processes when I checked it, unlike before.

Interestingly, in all our logs when this happens - it looks like all processes crash (even non-celery ones) and attempt to soft-restart.  They do restart but then very quickly the system stops working.

This one has puzzled us for about a month now.  We even re-wrote our celery stack to be more 'restrictive' to stop a flood of tasks being added.   That did not solve the problem.  So now we think it must be something further down the system stack (such as `rpicam-still`).

See the logs below (at least what we could gather before the system locked up)...

**celery.log** (Timezone: +8 GMT)
```
Aug 02 16:14:25 xx sh[1367]: [2025-08-02 16:14:25,210: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (tasks.timelapse_runtime)
Aug 02 16:14:25 xx kernel: rp1-cfe 1f00128000.csi: Using a link rate of 720 Mbps
Aug 02 16:14:25 xx kernel: rp1-cfe 1f00110000.csi: Using a link rate of 720 Mbps
Aug 02 16:14:30 xx sh[1367]: [2025-08-02 16:14:30,211: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (tasks.timelapse_runtime)
Aug 02 16:14:30 xx kernel: rp1-cfe 1f00128000.csi: Using a link rate of 720 Mbps
Aug 02 16:14:30 xx kernel: rp1-cfe 1f00110000.csi: Using a link rate of 720 Mbps
EOF
```

**beat.log** (Timezone: +8 GMT)
```
[2025-08-02 16:14:22,466: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task Get CPU Temperature Reading (environment.tasks.record_temp)
[2025-08-02 16:14:25,210: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (picam.tasks.timelapse_runtime)
[2025-08-02 16:14:30,211: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (picam.tasks.timelapse_runtime)
[2025-08-02 16:14:35,211: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (picam.tasks.timelapse_runtime)
[2025-08-02 16:14:37,467: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task Get CPU Temperature Reading (environment.tasks.record_temp)
[2025-08-02 16:14:40,211: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (picam.tasks.timelapse_runtime)
[2025-08-02 16:14:34,089: INFO/MainProcess] celery.beat.start[beat] beat: Starting...
[2025-08-02 16:14:35,281: INFO/MainProcess] DatabaseScheduler: Schedule changed.
[2025-08-02 16:14:35,289: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task Get CPU Temperature Reading (environment.tasks.record_temp)
[2025-08-02 16:14:35,483: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task upload-pending-images (management.tasks.upload_pending_images)
[2025-08-02 16:14:35,486: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (picam.tasks.timelapse_runtime)
[2025-08-02 16:14:35,492: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task internet-health-check (management.tasks.check_internet_health_task)
[2025-08-02 16:14:40,487: INFO/MainProcess] celery.beat.apply_entry[beat] Scheduler: Sending due task camera-takes-images (picam.tasks.timelapse_runtime)
```

**Database Logging System (not celery driven)**
```
[2025-08-02 16:13:09,627: INFO] Database has 11 active connections
[2025-08-02 16:14:09,926: INFO] Database has 10 active connections
[2025-08-02 16:14:23,710: INFO] Starting database connection monitor
[2025-08-02 16:14:24,521: INFO] Database has 6 active connections
[2025-08-03 09:20:53,377: INFO] Database has 8 active connections
[2025-08-03 09:21:53,782: INFO] Database has 7 active connections
[2025-08-03 09:22:54,176: INFO] Database has 7 active connections
[2025-08-03 09:23:54,477: INFO] Database has 7 active connections
```

**Django Server log (not celery driven)**
```
[2025-07-30 23:14:56,024: WARNING/MainProcess] django.request.log_response[log] Not Found: /favicon.ico
[2025-08-02 16:14:35,633: INFO/MainProcess] django.utils.autoreload.run_with_reloader[autoreload] Watching for file changes with StatReloader
```

I have slightly more verbose logging from celery, I am happy to share that via email / private message. 

This one has stumped us.  It seems it could be something camera software related, but we are at a dead end as to why all the system processes "soft restart" because of it.

Interested in your thoughts and opinions here,

naushir, 2025-08-04:
To help debug this issue, let's try and reproduce it with minimal software overhead.  Can you run the ``rpicam-still`` command on it's own in a loop, i.e. without any additional software running on top.  And if possible, ideally on a fresh sdcard image with all the updates applied.

wallarug, 2025-08-04:
> To help debug this issue, let's try and reproduce it with minimal software overhead. Can you run the `rpicam-still` command on it's own in a loop, i.e. without any additional software running on top. And if possible, ideally on a fresh sdcard image with all the updates applied.

I can try this later today (it's 1.15 AM right now so I'm about to knock off).

Just FYI - we are still getting the orphaned processes that were reported in #799 

[image]

goombado, 2025-08-05:
Hi all,

Further looking into this issue after updating to the latest version of rpicam-apps, we have received another kernel panic directly related to rpicam-still:

```
Aug 04 17:42:27 rpi kernel: Unable to handle kernel paging request at virtual address 0000003f7fc00a08
Aug 04 17:42:27 rpi kernel: Mem abort info:
Aug 04 17:42:28 rpi kernel:   ESR = 0x0000000096000005
Aug 04 17:42:28 rpi kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Aug 04 17:42:28 rpi kernel:   SET = 0, FnV = 0
Aug 04 17:42:28 rpi kernel:   EA = 0, S1PTW = 0
Aug 04 17:42:28 rpi kernel:   FSC = 0x05: level 1 translation fault
Aug 04 17:42:28 rpi kernel: Data abort info:
Aug 04 17:42:28 rpi kernel:   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
Aug 04 17:42:28 rpi kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Aug 04 17:42:28 rpi kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Aug 04 17:42:28 rpi kernel: user pgtable: 16k pages, 47-bit VAs, pgdp=0000000100d10000
Aug 04 17:42:28 rpi kernel: [0000003f7fc00a08] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Aug 04 17:42:28 rpi kernel: Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
Aug 04 17:42:28 rpi kernel: Modules linked in: nf_tables nfnetlink wireguard libchacha20poly1305 chacha_neon poly1305_neon ip6_udp_tunnel udp_tunnel libcurve25519_generic libchacha binfmt_misc brcmfmac_wcc vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper brcmfmac snd_soc_core ov64a40 brcmutil bu64754 cfg80211 v4l2_cci regmap_i2c snd_compress hci_uart aes_ce_blk snd_pcm_dmaengine btbcm aes_ce_cipher bluetooth snd_pcm ghash_ce rpi_hevc_dec gf128mul rp1_cfe pisp_be ecdh_generic sha2_ce snd_timer ecc sha256_arm64 snd rfkill v4l2_fwnode v4l2_mem2mem sha1_ce libaes v4l2_async videobuf2_dma_contig videobuf2_memops v3d videobuf2_v4l2 sha1_generic videodev gpu_sched raspberrypi_hwmon videobuf2_common drm_shmem_helper mc drm_kms_helper pwm_fan rp1_pio rp1_fw raspberrypi_gpiomem rp1_mailbox rp1_adc nvmem_rmem uio_pdrv_genirq uio drm drm_panel_orientation_quirks fuse backlight dm_mod ip_tables x_tables ipv6 spidev i2c_brcmstb spi_bcm2835 gpio_keys i2c_designware_platform i2c_designware_core
Aug 04 17:42:28 rpi kernel: CPU: 1 UID: 1000 PID: 177037 Comm: rpicam-still Not tainted 6.12.34+rpt-rpi-2712 #1  Debian 1:6.12.34-1+rpt1~bookworm
Aug 04 17:42:28 rpi kernel: Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
Aug 04 17:42:28 rpi kernel: pstate: a0400009 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 04 17:42:28 rpi kernel: pc : kfree+0x60/0x2d0
Aug 04 17:42:28 rpi kernel: lr : pispbe_prepare_job+0x2e8/0x540 [pisp_be]
Aug 04 17:42:28 rpi kernel: sp : ffffc0008414b9c0
Aug 04 17:42:28 rpi kernel: x29: ffffc0008414b9d0 x28: ffffd06fa7950938 x27: ffffd06fa7855078
Aug 04 17:42:28 rpi kernel: x26: 0000000000000000 x25: ffff8000c0f44000 x24: ffff8000c0f44200
Aug 04 17:42:28 rpi kernel: x23: ffff800006440098 x22: 00000000000000f7 x21: ffffd06fa79195b8
Aug 04 17:42:28 rpi kernel: x20: ffffffbfc00a0100 x19: 0000003f7fc00a00 x18: 0000000000000000
Aug 04 17:42:28 rpi kernel: x17: 0000000000000000 x16: ffffd06fce7626a0 x15: 00007fff5b6cc7c8
Aug 04 17:42:28 rpi kernel: x14: 0000000400000000 x13: 0000000000000000 x12: 0000000000000000
Aug 04 17:42:28 rpi kernel: x11: 0000000000000000 x10: 0000000000000000 x9 : ffffd06fa79195b8
Aug 04 17:42:28 rpi kernel: x8 : ffffffbfc00a0200 x7 : 0000000000000000 x6 : 000000000000003f
Aug 04 17:42:28 rpi kernel: x5 : 0000000000000040 x4 : ffff800003d22690 x3 : 0000000000000004
Aug 04 17:42:28 rpi kernel: x2 : 0000000000000000 x1 : ffff800006443cf8 x0 : ffffffbfc0000000
Aug 04 17:42:28 rpi kernel: Call trace:
Aug 04 17:42:28 rpi kernel:  kfree+0x60/0x2d0
Aug 04 17:42:28 rpi kernel:  pispbe_prepare_job+0x2e8/0x540 [pisp_be]
Aug 04 17:42:28 rpi kernel:  pispbe_node_buffer_queue+0x50/0x90 [pisp_be]
Aug 04 17:42:28 rpi kernel:  __enqueue_in_driver+0x54/0xe8 [videobuf2_common]
Aug 04 17:42:28 rpi kernel:  vb2_core_qbuf+0x3c8/0x5c8 [videobuf2_common]
Aug 04 17:42:28 rpi kernel:  vb2_qbuf+0xb4/0x198 [videobuf2_v4l2]
Aug 04 17:42:28 rpi kernel:  vb2_ioctl_qbuf+0x68/0x1660 [videobuf2_v4l2]
Aug 04 17:42:28 rpi kernel:  v4l_qbuf+0x50/0x70 [videodev]
Aug 04 17:42:28 rpi kernel:  __video_do_ioctl+0x170/0x3e0 [videodev]
Aug 04 17:42:28 rpi kernel:  video_usercopy+0x1c0/0x790 [videodev]
Aug 04 17:42:28 rpi kernel:  video_ioctl2+0x20/0x38 [videodev]
Aug 04 17:42:28 rpi kernel:  v4l2_ioctl+0x48/0x70 [videodev]
Aug 04 17:42:28 rpi kernel:  __arm64_sys_ioctl+0xb4/0x100
Aug 04 17:42:28 rpi kernel:  invoke_syscall+0x50/0x120
Aug 04 17:42:28 rpi kernel:  el0_svc_common.constprop.0+0x48/0xf0
Aug 04 17:42:28 rpi kernel:  do_el0_svc+0x24/0x38
Aug 04 17:42:28 rpi kernel:  el0_svc+0x30/0x100
Aug 04 17:42:28 rpi kernel:  el0t_64_sync_handler+0x100/0x130
Aug 04 17:42:28 rpi kernel:  el0t_64_sync+0x190/0x198
Aug 04 17:42:28 rpi kernel: Code: b26287e0 d34efe73 f2dff7e0 8b131813 (f9400660) 
Aug 04 17:42:28 rpi kernel: ---[ end trace 0000000000000000 ]---
```

We are currently running the following script to simulate a full day of capture (~1000 images). It will take about 2.5 hours to run, so we will report back once it's done with any issues if they've arisen.

```sh
while true; do
    echo "$(date '+%Y-%m-%d %H:%M:%S') - Starting camera capture 0" >> /var/log/camera_capture.log
    /usr/bin/rpicam-still --width 4626 --height 3472 --camera 0 -o /dev/shm/tmp0.jpg -n -t 3000 --no-raw --encoding jpg --quality 100 --rotation 0 >> /var/log/camera_capture_cam0.log 2>&1 &
[... cut ...]
    /usr/bin/rpicam-still --width 4626 --height 3472 --camera 1 -o /dev/shm/tmp1.jpg -n -t 3000 --no-raw --encoding jpg --quality 100 --rotation 0  >> /var/log/camera_capture_cam1.log 2>&1 &

    wait
    echo "$(date '+%Y-%m-%d %H:%M:%S') - Camera captures completed" >> /var/log/camera_capture.log

done
```

I will note as well that we are constantly monitoring the voltage level of the RPi using the internal voltage monitor, and we haven't been able identify this as a potential cause for the issues.

[image]

goombado, 2025-08-05:
Sure, it's

```
6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
```

As a final note of all the dumped logs, after the above occurred I tried sending SIGKILL to the two processes at the top of the `htop` log above, 182351 and 182353. This gave me the following in htop

```
182351 root        20   0     0     0     0 Z   0.0  0.0  0:00.60 rpicam-still
182353 root        20   0     0     0     0 Z   0.0  0.0  0:01.46 rpicam-still
182365 root        20   0     0     0     0 D   0.0  0.0  0:00.00 rpicam-still
182372 root        20   0     0     0     0 X   0.0  0.0  0:00.95 rpicam-still
```

I then ran `rpicam-still` one final time in my shell, and saw the following:
```
[4:50:26.583731686] [319374]  INFO Camera camera_manager.cpp:326 libcamera v0.5.1+100-e53bdf1f
[4:50:26.598997310] [319377]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[4:50:26.599951959] [319377] ERROR V4L2 v4l2_device.cpp:390 'ov64a40 10-0036': Unable to set controls: Device or resource busy
[4:50:26.610384857] [319377]  WARN RPiAf af.cpp:77 Missing parameter "retrigger_ratio"
[4:50:26.610425338] [319377]  WARN RPiAf af.cpp:77 Missing parameter "retrigger_delay"
[4:50:26.610437690] [319377]  WARN RPiAf af.cpp:77 Missing parameter "check_for_ir"
[4:50:26.611549876] [319377]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@88000/ov64a40@36 to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:50:26.611617284] [319377]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[4:50:26.621197051] [319377]  WARN RPiAf af.cpp:77 Missing parameter "retrigger_ratio"
[4:50:26.621235329] [319377]  WARN RPiAf af.cpp:77 Missing parameter "retrigger_delay"
[4:50:26.621246495] [319377]  WARN RPiAf af.cpp:77 Missing parameter "check_for_ir"
[4:50:26.622359552] [319377]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@80000/ov64a40@36 to CFE device /dev/media1 and ISP device /dev/media3 using PiSP variant BCM2712_C0
Made X/EGL preview window
Made DRM preview window
Preview window unavailable
Mode selection for 4624:3472:12:P
    SRGGB10_CSI2P,1920x1080/0 - Score: 11359.2
    SRGGB10_CSI2P,2312x1736/0 - Score: 9096
    SRGGB10_CSI2P,3840x2160/0 - Score: 5359.24
    SRGGB10_CSI2P,4624x3472/0 - Score: 1000
    SRGGB10_CSI2P,8000x6000/0 - Score: 2476.58
    SRGGB10_CSI2P,9248x6944/0 - Score: 3024
Stream configuration adjusted
[4:50:26.644870867] [319374]  INFO Camera camera.cpp:1205 configuring streams: (0) 4624x3472-YUV420/sYCC (1) 4624x3472-BGGR_PISP_COMP1/RAW
[4:50:26.644931756] [319377] ERROR V4L2 v4l2_device.cpp:390 'ov64a40 10-0036': Unable to set controls: Device or resource busy
```

The system is now almost entirely unresponsive. I could not `CTRL+C` to kill this process, and SSH after 2 minutes completely stopped working.

naushir, 2025-08-05:
This is quite strange.  I'm running a test myself right now to try and reproduce this.  I do have a later kernel (6.12.41) installed though.  Would you be able to update to this kernel with ``sudo rpi-update`` and test once more please?

naushir, 2025-08-05:
Also, would it be possible to run this with an official Raspberry Pi camera, like Camera v3 (IMX708) if you have one available?

naushir, 2025-08-05:
> Happy to try it, but for the effort, I would like to know what we are looking to achieve in the testing process

I want to make sure we are both testing with the same hardware setup. I don't have access to the Arducam camera, so the Camera V3 was my first suggestion.

goombado, 2025-08-06:
We've updated from `6.12.34+rpt-rpi-2712` to `6.12.41-v8-16k+` and are running the script now. On `.34` it lasted about 15 minutes, but on `.41` it's been going for the past 1.5 hours, which is definitely a good sign.

I wanted to clarify, I'm seeing a bit of information out there saying things to the effect of "don't use rpi-update". Is staying on `6.12.41-v8-16k+` going to be okay for us in the long run, or should we be using the kernel version that's installed by something like `apt install raspi-firmware`?

goombado, 2025-08-06:
Nevermind, I rebooted the rpi and after 5 minutes ran our script again, and almost immediately got this kernel error:

```
Aug 06 22:56:24 camera-testing kernel: Unable to handle kernel paging request at virtual address 0000003f7fc09708
Aug 06 22:56:24 camera-testing kernel: Mem abort info:
Aug 06 22:56:24 camera-testing kernel:   ESR = 0x0000000096000005
Aug 06 22:56:24 camera-testing kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Aug 06 22:56:24 camera-testing kernel:   SET = 0, FnV = 0
Aug 06 22:56:24 camera-testing kernel:   EA = 0, S1PTW = 0
Aug 06 22:56:24 camera-testing kernel:   FSC = 0x05: level 1 translation fault
Aug 06 22:56:24 camera-testing kernel: Data abort info:
Aug 06 22:56:24 camera-testing kernel:   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
Aug 06 22:56:25 camera-testing kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Aug 06 22:56:25 camera-testing kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Aug 06 22:56:25 camera-testing kernel: user pgtable: 16k pages, 47-bit VAs, pgdp=0000000087c48000
Aug 06 22:56:25 camera-testing kernel: [0000003f7fc09708] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Aug 06 22:56:25 camera-testing kernel: Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
Aug 06 22:56:25 camera-testing kernel: Modules linked in: wireguard libchacha20poly1305 chacha_neon poly1305_neon ip6_udp_tunnel udp_tunnel libcurve25519_generic libchacha 8021q garp stp llc binfmt_misc ov64a40 bu64754 v4l2_cci regmap_i2c brcmfmac_wcc spidev vc4 snd_soc_hdmi_codec drm_display_helper brcmfmac cec drm_dma_helper snd_soc_core brcmutil snd_compress cfg80211 hci_uart snd_pcm_dmaengine rpi_hevc_dec pisp_be rp1_cfe btbcm snd_pcm bluetooth raspberrypi_hwmon snd_timer v4l2_mem2mem aes_ce_blk v4l2_fwnode videobuf2_dma_contig v3d ecdh_generic aes_ce_cipher snd v4l2_async videobuf2_memops ecc ghash_ce rfkill videobuf2_v4l2 gf128mul libaes gpu_sched spi_bcm2835 videodev drm_shmem_helper sha2_ce drm_kms_helper i2c_brcmstb sha256_arm64 videobuf2_common sha1_ce mc w1_gpio i2c_designware_platform wire rp1_pio rp1_mailbox i2c_designware_core cn raspberrypi_gpiomem sha1_generic rp1_adc uio_pdrv_genirq gpio_keys pwm_fan rp1_fw uio nvmem_rmem i2c_dev drm drm_panel_orientation_quirks backlight fuse dm_mod ip_tables x_tables ipv6
Aug 06 22:56:25 camera-testing kernel: CPU: 2 UID: 0 PID: 5128 Comm: rpicam-still Not tainted 6.12.41-v8-16k+ #1897
Aug 06 22:56:25 camera-testing kernel: Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
Aug 06 22:56:25 camera-testing kernel: pstate: a0400009 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 06 22:56:25 camera-testing kernel: pc : kfree+0x5c/0x2d0
Aug 06 22:56:25 camera-testing kernel: lr : pispbe_prepare_job+0x32c/0x508 [pisp_be]
Aug 06 22:56:25 camera-testing kernel: sp : ffffc000822139b0
Aug 06 22:56:25 camera-testing kernel: x29: ffffc000822139b0 x28: ffffd06f9bfb0e40 x27: ffff8000747fe640
Aug 06 22:56:25 camera-testing kernel: x26: ffff800006a94ae8 x25: ffff8000773ba800 x24: 00000000c058560f
Aug 06 22:56:25 camera-testing kernel: x23: 00000000000000f7 x22: ffff800006a90080 x21: ffffd06f9c2d5424
Aug 06 22:56:25 camera-testing kernel: x20: ffffffbfc0970000 x19: 0000003f7fc09700 x18: 0000000000000000
Aug 06 22:56:25 camera-testing kernel: x17: 0000000000000000 x16: ffffd06fce75a350 x15: 00007ffee16987c8
Aug 06 22:56:25 camera-testing kernel: x14: 0000000400000000 x13: 0000000000000000 x12: 0000000000000000
Aug 06 22:56:25 camera-testing kernel: x11: 000000000008cd57 x10: 0000000000008248 x9 : ffffd06f9c2d5424
Aug 06 22:56:25 camera-testing kernel: x8 : ffffffbfc0970100 x7 : 0000000000000000 x6 : 000000000000003f
Aug 06 22:56:25 camera-testing kernel: x5 : 0000000000000040 x4 : ffff800008a42690 x3 : 0000000000000004
Aug 06 22:56:25 camera-testing kernel: x2 : 0000000000000000 x1 : ffff800006a97dc8 x0 : ffffffbfc0000000
Aug 06 22:56:25 camera-testing kernel: Call trace:
Aug 06 22:56:25 camera-testing kernel:  kfree+0x5c/0x2d0
Aug 06 22:56:25 camera-testing kernel:  pispbe_prepare_job+0x32c/0x508 [pisp_be]
Aug 06 22:56:25 camera-testing kernel:  pispbe_node_buffer_queue+0x50/0x90 [pisp_be]
Aug 06 22:56:25 camera-testing kernel:  __enqueue_in_driver+0x54/0xe8 [videobuf2_common]
Aug 06 22:56:25 camera-testing kernel:  vb2_core_qbuf+0x444/0x680 [videobuf2_common]
Aug 06 22:56:25 camera-testing kernel:  vb2_qbuf+0xb0/0x190 [videobuf2_v4l2]
Aug 06 22:56:25 camera-testing kernel:  vb2_ioctl_qbuf+0x64/0x80 [videobuf2_v4l2]
Aug 06 22:56:25 camera-testing kernel:  v4l_qbuf+0x50/0x70 [videodev]
Aug 06 22:56:25 camera-testing kernel:  __video_do_ioctl+0x194/0x400 [videodev]
Aug 06 22:56:25 camera-testing kernel:  video_usercopy+0x31c/0x758 [videodev]
Aug 06 22:56:25 camera-testing kernel:  video_ioctl2+0x20/0x38 [videodev]
Aug 06 22:56:25 camera-testing kernel:  v4l2_ioctl+0x48/0x70 [videodev]
Aug 06 22:56:25 camera-testing kernel:  __arm64_sys_ioctl+0xb0/0x100
Aug 06 22:56:25 camera-testing kernel:  invoke_syscall+0x50/0x120
Aug 06 22:56:25 camera-testing kernel:  el0_svc_common.constprop.0+0x48/0xf8
Aug 06 22:56:25 camera-testing kernel:  do_el0_svc+0x28/0x40
Aug 06 22:56:25 camera-testing kernel:  el0_svc+0x30/0x100
Aug 06 22:56:25 camera-testing kernel:  el0t_64_sync_handler+0x13c/0x158
Aug 06 22:56:25 camera-testing kernel:  el0t_64_sync+0x190/0x198
Aug 06 22:56:25 camera-testing kernel: Code: b26287e0 d34efe73 f2dff7e0 8b131813 (f9400660) 
Aug 06 22:56:25 camera-testing kernel: ---[ end trace 0000000000000000 ]---
```

After killing the shell and rpicam-still processes, I'm still left with these two lingering processes in htop

```
   5121 root        20   0     0     0     0 Z   0.0  0.0  0:00.59 rpicam-still
   5129 root        20   0  486M 28720 20928 D   0.0  0.7  0:00.29 /usr/bin/rpicam-still --width 4626 --height 3472 --camera 0 -o /dev/shm/tmp0.jpg -n -t 3000 --no-raw --encoding jpg --quality 100 --rotation 0
```

naushir, 2025-08-06:
Thanks for the update.  Is this with the same Arducam camera, or the IMX219/IMX708?

naushir, 2025-08-06:
Camera v2 should be fine for this test - no need to purchase a v3 to begin with.

goombado, 2025-08-06:
Sure thing. Once we have our hands on the v2 we'll run the same tests and see how we go. Do you have any suspicions about anything in the Arducams that you think may be causing these consistent `kfree` / `kmalloc_noprof` kernel errors?

naushir, 2025-08-06:
> Do you have any suspicions about anything in the Arducams that you think may be causing these consistent kfree / kmalloc_noprof kernel errors?

Nothing comes to mind unfortunately.  I will continue trying to reproduce this and that should hopefully give me a better way to debug this issue.

wallarug, 2025-08-06:
> ov64a40 producing 16MPix frames (4624x3472) for preview are quite a bit larger than most of the sensors.
> 
> I don't have one of those Omnivision modules (thepihut stock them), but I do have one of Arducam's predecessor Sony IMX682 64MPix modules that might be worth a test. It'll need a little faffing as the camera_helper and tuning files were never upstreamed because nor was the driver.

That would be really helpful.   Appreciate the effort.  I have two of those cameras as well - I am well aware of the faffing around required.

> My only (weak) suspicion is we may be running out of space in the IOMMU mapping table with 2x 64Mpix cameras running simultaneously.

We are running them at 16 MP.   I suspect you may be onto something here - but as we mentioned, we have on started seeing the issues since we updated in 6 July 2025.

wallarug, 2025-08-08:
> Any updates with the single camera test?

Not yet - I've been out for the last 48 hours.   I'll run the single camera test tomorrow (Saturday) and report back.  

I've got the two V2 Cameras on my desk ready to go as well.

Sorry for inactivity on this one, it's been flat chat this week.

wallarug, 2025-08-10:
Ohh - one other development.  On the camera we are not running the tests on, it failed again over the weekend.  It has a slightly different stack trace.  I've attached below.

It hints at something inside the HEVC Encoder is causing the issue.  However, this could just be where it stopped at this point in time.   Thought I would add it here for you all to review anyway.  

This is way beyond my current understanding of the camera code.  So I'll leave it up to your expertise to provide thoughts and analysis.

[message (6) (1).txt](https://github.com/user-attachments/files/21705915/message.6.1.txt)

```bash
Aug 09 15:08:50 hostname kernel: rp1-cfe 1f00128000.csi: Using a link rate of 720 Mbps
Aug 09 15:08:50 hostname kernel: rp1-cfe 1f00110000.csi: Using a link rate of 720 Mbps
Aug 09 15:09:45 hostname kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000081
Aug 09 15:09:45 hostname kernel: Mem abort info:
Aug 09 15:09:45 hostname kernel:   ESR = 0x0000000096000005
Aug 09 15:09:45 hostname kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Aug 09 15:09:45 hostname kernel:   SET = 0, FnV = 0
Aug 09 15:09:46 hostname kernel:   EA = 0, S1PTW = 0
Aug 09 15:09:46 hostname kernel:   FSC = 0x05: level 1 translation fault
Aug 09 15:09:46 hostname kernel: Data abort info:
Aug 09 15:09:46 hostname kernel:   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
Aug 09 15:09:46 hostname kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Aug 09 15:09:46 hostname kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Aug 09 15:09:46 hostname kernel: user pgtable: 16k pages, 47-bit VAs, pgdp=0000000103c9c000
Aug 09 15:09:46 hostname kernel: [0000000000000081] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Aug 09 15:09:46 hostname kernel: Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
Aug 09 15:09:46 hostname kernel: Modules linked in: nf_tables nfnetlink wireguard libchacha20poly1305 chacha_neon poly1305_neon ip6_udp_tunnel udp_tunnel li                                                                                                                                                                bcurve25519_generic libchacha binfmt_misc vc4 snd_soc_hdmi_codec drm_display_helper ov64a40 bu64754 cec brcmfmac_wcc brcmfmac v4l2_cci regmap_i2c drm_dma_hel                                                                                                                                                                per brcmutil hci_uart snd_soc_core btbcm cfg80211 bluetooth aes_ce_blk snd_compress aes_ce_cipher pisp_be rp1_cfe rpi_hevc_dec ecdh_generic ghash_ce snd_pcm_                                                                                                                                                                dmaengine snd_pcm ecc gf128mul sha2_ce v4l2_mem2mem v4l2_fwnode rfkill snd_timer sha256_arm64 libaes videobuf2_dma_contig v4l2_async snd sha1_ce videobuf2_me                                                                                                                                                                mops videobuf2_v4l2 v3d videodev sha1_generic raspberrypi_hwmon gpu_sched drm_shmem_helper videobuf2_common drm_kms_helper mc rp1_pio pwm_fan raspberrypi_gpi                                                                                                                                                                omem rp1_fw rp1_mailbox rp1_adc nvmem_rmem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6 spidev spi_bcm2                                                                                                                                                                835 i2c_brcmstb gpio_keys i2c_designware_platform i2c_designware_core
Aug 09 15:09:46 hostname kernel: CPU: 2 UID: 1000 PID: 3213443 Comm: rpicam-still Not tainted 6.12.34+rpt-rpi-2712 #1  Debian 1:6.12.34-1+rpt1~bookworm
Aug 09 15:09:46 hostname kernel: Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
Aug 09 15:09:46 hostname kernel: pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 09 15:09:46 hostname kernel: pc : __kmalloc_node_noprof+0x1cc/0x3d0
Aug 09 15:09:46 hostname kernel: lr : __kvmalloc_node_noprof+0x24/0x128
Aug 09 15:09:46 hostname kernel: sp : ffffc000846436a0
Aug 09 15:09:46 hostname kernel: x29: ffffc000846436b0 x28: 0000000000000000 x27: 0000000000a40001
Aug 09 15:09:46 hostname kernel: x26: ffffd06fbf3aefc0 x25: 0000000000000000 x24: 00000000ffffffff
Aug 09 15:09:46 hostname kernel: x23: 0000000000000001 x22: ffffd06fce70586c x21: 0000000000000dc0
Aug 09 15:09:46 hostname kernel: x20: ffff800080004400 x19: 00000000000000e8 x18: 0000000000000000
Aug 09 15:09:46 hostname kernel: x17: 0000000000000000 x16: ffffd06fce705848 x15: 0000000000000000
Aug 09 15:09:46 hostname kernel: x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
Aug 09 15:09:46 hostname kernel: x11: 0000000000000000 x10: 0000000000000000 x9 : ffffd06fce70586c
Aug 09 15:09:46 hostname kernel: x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
Aug 09 15:09:46 hostname kernel: x5 : 0000000000000006 x4 : 0000000000000000 x3 : 00000000004966ba
Aug 09 15:09:46 hostname kernel: x2 : 0000000000000080 x1 : 0000000000000001 x0 : ffff800080004400
Aug 09 15:09:46 hostname kernel: Call trace:
Aug 09 15:09:46 hostname kernel:  __kmalloc_node_noprof+0x1cc/0x3d0
[... cut ...]
Aug 09 15:10:04 hostname kernel: Code: f9002bf9 aa1403e1 aa1303e0 b9402a62 (f8626a99)
Aug 09 15:10:04 hostname kernel: ---[ end trace 0000000000000000 ]---
Aug 09 15:10:28 hostname kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000081
Aug 09 15:10:28 hostname kernel: Mem abort info:
Aug 09 15:10:28 hostname kernel:   ESR = 0x0000000096000005
Aug 09 15:10:28 hostname kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Aug 09 15:10:28 hostname kernel:   SET = 0, FnV = 0
Aug 09 15:10:29 hostname kernel:   EA = 0, S1PTW = 0
```

naushir, 2025-08-11:
If it's a memory allocation/fragmentation issue to do with the MMU, I won't be surprised if you see callstacks unrelated to camera/imaging drivers.

naushir, 2025-08-11:
Just curious - is your usecase actually using the HEVC decoder while the camera is running?

wallarug, 2025-08-11:
Hey @naushir

> Just curious - is your usecase actually using the HEVC decoder while the camera is running?

Nope.  We are just capturing a jpeg and then saving it.  None of our camera stuff uses the HEVC encoder - which is why this was odd.

> If it's a memory allocation/fragmentation issue to do with the MMU, I won't be surprised if you see callstacks unrelated to camera/imaging drivers.

Yes.

> I've been running the single camera test for over an hour now and nothing of note has happened yet. Exact same script as before, but only with camera 0.
> 
> I'll leave it running over night as well.

It has not crashed yet.   So a single camera seems ok.  It's been running for more than 24 hours.

What do you think the next steps should be on this one?   Test with the V2 camera?  Wait for it to crash again with 2 OV64A40s?

naushir, 2025-08-11:
Yes, it would be a useful data point to run the single 64MPix camera in a loop as well!

wallarug, 2025-08-12:
Hey @naushir  - No crashes with just the 64 MP camera going for more than 16 hours now.

njhollinghurst, 2025-08-12:
This kernel change https://github.com/raspberrypi/linux/pull/6997 allows IOMMU aperture sizes to be set using DTS. It will increase both the ISP-BE and ISP-FE's IOVA spaces from 2GB (default) to 4GB (which is currently the maximum limit).

This comes at a cost of increasing allocated memory for IOMMU tables (at a ratio of 1024:1, so the change as it stands will consume another 4MB), but it might still be interesting to test against.

goombado, 2025-08-13:
I ran this script again after restarting the raspberry pi, and after ~2 minutes the same issue occurred. I tried again to run `rpicam-still` after the kernel logs appeared in `journalctl`, which as before caused the entire system to become completely unresponsive within about a minute.

naushir, 2025-08-13:
@goombado thanks for the update.  I will concentrate my efforts on testing dual cameras since it seems that's the problematic case.

Can you provide the output of ``uname -a`` and ``rpicam-hello --version`` please?

wallarug, 2025-08-13:
> [@goombado](https://github.com/goombado) thanks for the update. I will concentrate my efforts on testing dual cameras since it seems that's the problematic case.
> 
> Can you provide the output of `uname -a` and `rpicam-hello --version` please?

**production camera**
```bash
Linux XXXXXX 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
rpicam-apps build: v1.8.1 eb293b055248 21-07-2025 (11:50:44)
rpicam-apps capabilites: egl:0 qt:0 drm:1 libav:0
libcamera build: v0.5.1+100-e53bdf1f
```

**testing camera**
```bash
Linux camera-testing 6.12.41-v8-16k+ #1897 SMP PREEMPT Fri Aug  1 12:23:30 BST 2025 aarch64 GNU/Linux
rpicam-apps build: v1.8.1 eb293b055248 21-07-2025 (11:50:44)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.5.1+100-e53bdf1f
```

njhollinghurst, 2025-08-13:
Actually, the IOMMU aperture change is probably not relevant here (although I think we've had past reports that IOVA could be exhausted in some scenarios with two 64 Mpix cameras and many buffers).

wallarug, 2025-08-14:
> [raspberrypi/linux#6999](https://github.com/raspberrypi/linux/pull/6999) ought to fix this issue now. When the CI has completed all the builds, you can get the kernel for testing with `sudo rpi-update pulls/6999`.

Hi @naushir ,

Thanks for the help.  You mentioned in your PR that the changes were not enough to fix this issue.  Were there additional changes made after that comment?

goombado, 2025-08-15:
Hi @naushir,

We've been testing our two-camera scripts at lots of different resolutions, both with V2 and arducam, and so far haven't had any kernel issues crop up. We'll keep testing for the rest of today, but from the looks of things for now everything seems to be going well. Thank you again for the help on fixing this issue!

wallarug, 2025-08-16:
> Yes, there was an additional change needed which I added to the PR.

Hey @naushir - we have been testing this for 2 days and all seems much better.  Appreciate the support getting this resolved.

When / how do we know this has gone into the main kernel update or do we just keep running on the 6999 PR one?