raspberrypi/rpicam-apps #856: [BUG] --immediate flag seems to be in rpicam-still but not in picamera2

almowery, 2025-08-28:
naushir, I posted this to comments from the post, but that seemed wrong. So I started a new issue to reach you

https://github.com/raspberrypi/rpicam-apps/issues/524

I hope you can help me with this.  I am trying to use a picamera2 app called indi_pylibcamera.  I have been using a the rpicam-still command with the
 --immediate option with great success.  I wanted to try using this app which works with picamera2 and I am seeing different results.  
 
 When I run picamera with script 5sec works fine 5sec result, raw mode.
 When i run picamera with a loop repeated the 5sec jumps to 10sec.
 
 The indi_pylibcamera app is worse it always takes double the time and often even longer.  I am confused what is the difference between rpicam-still and picamera2
 and then indi_pylibcamera which is supposed to be built on picamera2  why aren't they the same?
 
I do not know if the driver(.dtb)from the kernal, something in libcamera, or if it is just a setting in indi_pylibacamera is not making the --immediate mode.

The driver/camera I am using is imx283 which went from a developer created driver to incorporation into the latest kernal build.

I thought the fix you put in from this post was to make the switch --immediate stop preview frames.  Is this correct?

You seem to have a real handle on this problem with some cameras I hope you cna help me solve this.

I have included the py script that repeats the exposures in a loop in case this helps.

#!/usr/bin/python3

"""
test program for taking raw still images the same was as it is done by indi_pylibcamera
"""

import time
import datetime
import pprint
from picamera2 import Picamera2
from libcamera import controls
from playhouse.sqlite_udf import total_seconds

picam2 = Picamera2()  # TODO: select correct camera if you have more than one connected!

# Print camera information. The "config" and "controls" settings below must fit to the capabilities
# of the camera!
print('Camera properties:')
pprint.pprint(picam2.camera_properties)
print()
print("Raw sensor modes:")
pprint.pprint(picam2.sensor_modes)
print()
print('Camera controls:')
pprint.pprint(picam2.camera_controls)
print()

# We want to make still images.
config = picam2.create_still_configuration(queue=False, buffer_count=2)

# We are interested in raw images only. But the driver can not provide raw images without a
# processed ("main") one. To avoid wasting time and memory for calculating the "main" image we 
# configure it here with a pretty small size.
config["main"]["size"] = (240, 190)

# The raw image is the important one. Its configuration mut fit to the capabilities of the camera,
# which can be determined with "picam2.sensor_modes". The following settings are for IMX477.
config["raw"] = {"size": (4576, 3648), "format": 'SRGGB12'}

# Set configuration
picam2.align_configuration(config)
picam2.configure(config)

# Set the exposure controls. This must fit t the capabilities of your camera! Possible controls and
# their value ranges can be determined with "picam2.camera_controls"
# The following is for IMX477.
controls = {
    # Exposure time and analog gain are the most important controls for raw images!
    "ExposureTime": int(5 * 1e6),  # unit: microsecond
    "AnalogueGain": 16.0,
    # Disable all control loops which take multiple exposures to settle.
    "AeEnable": False,  # disable auto-exposure control loop which would need many exposures for one image!
    "AwbEnable": False, # disable the auto-widebalance control loop
    # "AfMode": "MANUAL", # For cameras with auto-focus we also need to disable the auto-focus control loop.
    # The following is just to be sure.
    "ExposureValue": 0.0, # not sure if this has an effect on raw images
    "NoiseReductionMode": controls.draft.NoiseReductionModeEnum.Off, # disable "star-eater" - should not have an effect on raw images but who knows
}
picam2.set_controls(controls)

# Take a number of images in a loop.
for n in range(5):
    TimeStamp_Start = datetime.datetime.now()

    # Start camera.
    picam2.start()

    # Start an exposure. We use the asynchronous capture functions because the "indi_pylibcamera" needs to
    # stay responsive during exposure.
    job = picam2.capture_arrays(
        ["raw"],
        wait=False, # don't block
        # a "signal_function" can be given here which gets called at exposure end
    )

    # Wait until end of exposure and read image.
    (array, ), metadata = picam2.wait(job)

    # "array" is a numpy array with raw image data. "metadata" describes this image.
    # You can examine the meta data with "pprint.pprint(metadata)"

    # Stop camera.
    picam2.stop()

    TimeStamp_Stop = datetime.datetime.now()
    TimeNeeded = (TimeStamp_Stop - TimeStamp_Start).total_seconds()

    # Report CPU time needed for exposure. This should not differ too much from image exposure time!
    print(f'Exposure {n+1} took {TimeNeeded:.3f} sec, {array.mean()=:.3f}, {array.std()=:.3f}.')

    # If needed: wait some time between exposures to simulate frame processing.
    time.sleep(2.0)  # unit: seconds

print("Finished.")

Thank You.

naushir, 2025-08-29:
Adding @davidplowman 

Can you provide the exact output generated by this script please?

davidplowman, 2025-08-29:
Is this the same problem as https://github.com/raspberrypi/picamera2/issues/1312 ?

Which issue are we replying to, that one, or this one? Generally it is helpful to report problems just once, so that the discussion doesn't get spread around the place and everyone who is trying to help gets lost.

If you're seeing 10 seconds go by in the loop, it sounds a bit like something is waiting for another 5 second frame. Maybe:

* Try it with an official camera like the imx477. We don't have imx283 here, so you need to rule that out as the source of the problem.
* Sample the time after the image capture but before the stop(). It would help to know where your extra 5 seconds is going.
* Presumably the captured images are still reporting 5s exposure, right?
* Please confirm what kind of Pi this is, I couldn't see that you had said this anywhere.

Thanks.

davidplowman, 2025-08-29:
Am I right that this question has now popped up [here](https://github.com/raspberrypi/rpicam-apps/issues/856)?

I'm going to close this one so that the discussion all stays in one place.

almowery, 2025-08-30:
Here are the results.

I am using a RaspberryPi 5
with a imx283 sensor made by Will Wang.  Arducam was selling them for awhile and now they are selling a system.
imx2283.c for kernel is in the newest build .y  I was originally using Will Wangs own driver/libcamera/rpicam-app build
But for this testing I upgraded everything to the newest builds from repositories  and it works the same as Will's drivers.

I am using the takeStill.py file I was asked in #1312

Imx477

0:00:28.210705055] [1774]  INFO Camera camera.cpp:1205 configuring streams: (0) 192x190-BGR888/sRGB (1) 4056x3040-SBGGR16/RAW
[0:00:28.210811629] [1780]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@80000/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected CFE format: 4056x3040-BYR2
Exposure 1 took 5.650 sec, array.mean()=69.302, array.std()=72.673.
Exposure 2 took 5.609 sec, array.mean()=69.316, array.std()=72.665.
Exposure 3 took 5.614 sec, array.mean()=69.317, array.std()=72.679.
Exposure 4 took 5.613 sec, array.mean()=69.316, array.std()=72.655.
Exposure 5 took 5.606 sec, array.mean()=69.344, array.std()=72.651.
Finished.

I added prints for each stage of loop. printout here

[27:09:39.271986476] [6009]  INFO Camera camera.cpp:1205 configuring streams: (0) 240x190-BGR888/sRGB (1) 5472x3648-SRGGB16/RAW
[27:09:39.272106735] [6015]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@80000/imx283@1a - Selected sensor format: 5472x3648-SRGGB12_1X12 - Selected CFE format: 5472x3648-RG16
[27:09:39.272219494] [6015]  INFO IPARPI ipa_base.cpp:610 Adjusting mode minimum line length from 12.32us to 14.40us because of ISP constraints.
Exposure 1 took 5.638 sec beforejob 0.062 sec before array 0.062 sec before stop 5.635 sec, array.mean()=81.901, array.std()=65.591.
Exposure 2 took 10.023 sec beforejob 0.001 sec before array 0.001 sec before stop 10.021 sec, array.mean()=81.851, array.std()=65.613.
Exposure 3 took 10.011 sec beforejob 0.001 sec before array 0.001 sec before stop 10.008 sec, array.mean()=81.786, array.std()=65.634.
Exposure 4 took 10.020 sec beforejob 0.001 sec before array 0.001 sec before stop 10.017 sec, array.mean()=81.755, array.std()=65.631.
Exposure 5 took 10.019 sec beforejob 0.001 sec before array 0.001 sec before stop 10.016 sec, array.mean()=81.727, array.std()=65.643.
Finished.

loop portion of code>>

for n in range(5):
 TimeStamp_Start = datetime.datetime.now()
 picam2.start()
 TimeStamp_Beforejob = datetime.datetime.now()
 job = picam2.capture_arrays(
        ["raw"],
        wait=False, # don't block
        # a "signal_function" can be given here which gets called at exposure end
 )
 # Wait until end of exposure and read image.
 TimeStamp_Beforearray = datetime.datetime.now()
 (array, ), metadata = picam2.wait(job)
 TimeStamp_Before = datetime.datetime.now()
 # Stop camera.
 picam2.stop()
 #picam2.capture_file("test.jpg")
 TimeStamp_Stop = datetime.datetime.now()
 TimeNeeded = (TimeStamp_Stop - TimeStamp_Start).total_seconds()
 TimeNeededBefore = (TimeStamp_Before - TimeStamp_Start).total_seconds()
 TimeNeededBeforejob = (TimeStamp_Beforejob - TimeStamp_Start).total_seconds()
 TimeNeededBeforearray = (TimeStamp_Beforearray - TimeStamp_Start).total_seconds()
    # Report CPU time needed for exposure. This should not differ too much from image exposure time!
 print(f'Exposure {n+1} took {TimeNeeded:.3f} sec beforejob {TimeNeededBeforejob:.3f} sec before array {TimeNeededBeforearray:.3f} sec before stop {TimeNeededBefore:.3f} sec, {array.mean()=:.3f}, {array.std()=:.3f}.')

print("Finished.")

It appears that stopping causes the extra delay.  I had been mistakenly not stopping each exposure in the code I was originally working with and thus I was not seeing the extra delay. when I realized this the delay after the first exposure started.

Thank You for the testing suggestions.

almowery, 2025-09-02:
When i studied this problem months ago I saw the line in cam_helper_imx283.cpp in the libcamera build.

In order to do a test on a pi 5 to see if this is the problem would I:

>>change the line in

unsigned int CamHelperImx283::hideFramesModeSwitch() const
{
	/* After a mode switch, we seem to get 1 bad frame. */
	return 1;
}

>>to return 0; to turn off the call for extra frame.

>>Then rebuild the libcamera from local directory.
> >Then rebuild picamera2 from local directory.

I rebuild and installed libcamera from libcamera directory I downloaded.
Then to install picamera2 from picamera2 directory I downloaded I had to go to my venv then install picamera2.

should this work?
didn't seem to.

Thank you for staying with me.

naushir, 2025-09-04:
Thanks for the update! So is your issue resolved now?