raspberrypi/rpicam-apps #859: How to support RISC-V devices

happydog6666, 2025-09-10:
When I tried to run it on a RISC-V device, I got a prompt saying it can only be used on Raspberry Pi devices.
`roma@roma ~/rpicam-apps (main)> rpicam-hello
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps`

happydog6666, 2025-09-10:
> When I tried to run it on a RISC-V device, I got a prompt saying it can only be used on Raspberry Pi devices.当我尝试在 RISC-V 设备上运行它时，我收到提示说它只能在 Raspberry Pi 设备上使用。 `roma@roma ~/rpicam-apps (main)> rpicam-hello ERROR: rpicam-apps currently only supports the Raspberry Pi platforms. Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps`

closed 2025-09-10 by happydog6666

naushir, 2025-09-10:
> Can I modify the code to make it support RISC-V?

If you add support for other platforms we will be happy to take a look at a PR and consider merging it.